negan07/ancistrus #1: Free NAND space

negan07, 2017-04-20:
```
# df
Filesystem           1024-blocks    Used Available Use% Mounted on
mtd:rootfs               36864     34824      2040  94%
mtd:factory               1024       456       568  45% /config/factory
mtd:scnvram               4096       448      3648  11% /config/nvram
mtd:xxx                   1024       388       636  38% /config/xxx
mtd:language_ENU          1024       476       548  46% /config/language_ENU
mtd:language_CHS          1024       496       528  48% /config/language_CHS
mtd:language_RUS          1024       516       508  50% /config/language_RUS
mtd:language_DEU          1024       496       528  48% /config/language_DEU
mtd:language_FRA          1024       500       524  49% /config/language_FRA
mtd:language_NLD          1024       496       528  48% /config/language_NLD
mtd:language_SVE          1024       492       532  48% /config/language_SVE
mtd:language_ITA          1024       488       536  48% /config/language_ITA
mtd:language_KOR          1024       492       532  48% /config/language_KOR
mtd:language_ESP          1024       492       532  48% /config/language_ESP
mtd:data                  2048       388      1660  19% /data

mtd:rootfs               36864     34824      2040  94%
```
need more space to install new packages
ideas & strategies are welcome

negan07, 2017-04-20:
- rebuilding part table may bring to failure and compatibility issue with upgrades
- removing heavy builtin apps (mediaserver, time machine ...) is not a clean way

Activating opkg overlap & moving thirdparty packages to a stick usb/hdd losing 1/2 usb ports ?

muscettola, 2017-04-21:
Remove languages except English?

negan07, 2017-04-21:
can be done but with low gain: each lang is 1mb only with the risk of bricking all non english web gui setup

there's something unused, because the summatory of df gives:
37 + 12 + 2 + 4 = 55 Mb
NAND is 128Mb

the others ?

muscettola, 2017-04-21:
#cat /proc/mtd
dev:    size   erasesize  name
mtd0: 02400000 00020000 "rootfs"
mtd1: 00160000 00020000 "bootloader"
mtd2: 00100000 00020000 "language_DEU"
mtd3: 00400000 00020000 "scnvram"
mtd4: 00100000 00020000 "factory"
mtd5: 03000000 00020000 "upgrade"
mtd6: 00200000 00020000 "flag"
mtd7: 00120000 00020000 "pcbasn"
mtd8: 00100000 00020000 "xxx"
mtd9: 00100000 00020000 "language_ITA"
mtd10: 00100000 00020000 "language_RUS"
mtd11: 00100000 00020000 "language_CHS"
mtd12: 00100000 00020000 "language_ENU"
mtd13: 00100000 00020000 "language_dev"
mtd14: 00100000 00020000 "language_NLD"
mtd15: 00100000 00020000 "language_KOR"
mtd16: 00100000 00020000 "language_FRA"
mtd17: 00100000 00020000 "language_SVE"
mtd18: 00100000 00020000 "language_ESP"
mtd19: 00100000 00020000 "language_CSY"
mtd20: 00100000 00020000 "language_TUR"
mtd21: 00200000 00020000 "data"
mtd22: 000a0000 00020000 "part-map"
mtd23: 02400000 00020000 "rootfs2"

negan07, 2017-04-22:
no match against img build part table (D7000.par)

:D7000_boot.bin                      &00000000
:rootfs.img                             &00200000
:langpkg_ENU.img                        &02600000
:langpkg_CHS.img                        &02700000
:langpkg_ITA.img                        &02800000
:langpkg_DEU.img                        &02900000
:langpkg_NLD.img                        &02a00000
:langpkg_KOR.img                        &02b00000
:langpkg_FRA.img                        &02c00000
:langpkg_SVE.img                        &02d00000
:langpkg_RUS.img                        &02e00000
:langpkg_ESP.img                        &02f00000
:langpkg_CSY.img                        &03000000
:langpkg_TUR.img                        &03100000
%D7000.pid		                &00003fff
$size				        &10000

some parts of the NAND are not touched during upgrade
rootfs2 & upgrade should be reused because the largest: what contained needs a dump...

a suicide way could be to renew the part map D7000.par enlarging rootfs & using all the 128Mb
rootfs would have a size of about 115Mb

:D7000_boot.bin                      &00000000
:rootfs.img                             &00200000
:langpkg_ENU.img                        &07400000
:langpkg_CHS.img                        &07500000
:langpkg_ITA.img                        &07600000
:langpkg_DEU.img                        &07700000
:langpkg_NLD.img                        &07800000
:langpkg_KOR.img                        &07900000
:langpkg_FRA.img                        &07a00000
:langpkg_SVE.img                        &07b00000
:langpkg_RUS.img                        &07c00000
:langpkg_ESP.img                        &07d00000
:langpkg_CSY.img                        &07e00000
:langpkg_TUR.img                        &07f00000
%D7000.pid		                &00003fff
$size				        &20000

but all the partition mtd table should be rebuild because not contiguous (rootfs will run out and cover booloader) and so

negan07, 2017-04-24:
At the end the only undestructive way is to remove unused language: poor gain poor pain...

The annexb language_CSY language_TUR are not mounted and could be dropped: 2Mb more space and should not bring damage remaining under 0x4000000h position

:D7000_boot.bin                      &00000000
:rootfs.img                             &00200000
:langpkg_ENU.img                        &02800000
:langpkg_CHS.img                        &02900000
:langpkg_ITA.img                        &02a00000
:langpkg_DEU.img                        &02b00000
:langpkg_NLD.img                        &02c00000
:langpkg_KOR.img                        &02d00000
:langpkg_FRA.img                        &02e00000
:langpkg_SVE.img                        &02f00000
:langpkg_RUS.img                        &03000000
:langpkg_ESP.img                        &03100000
%D7000.pid		                &00003fff
$size				        &10000

the same can do annex b against annex a region langs

Leaving only english ENU and the ownlanguage the gain is 10Mb for rootfs: better than nothing

negan07, 2017-04-24:
Made a dump of the /dev/mtd partition available
They're mtd0 to mtd14 (mtd15 to mtd23 are missed)

mtd0 = bootloader (D7000_boot.bin)
mtd1 = the rootfs only (rootfs.img)
mtd2 = lang (DEU ?)
mtd3 = scnvram (but data appears not readable..)
mtd4 = factory (99% void)
mtd5 = upgrade: compound of mtd1(rootfs)+all languages in the order of D7000.par (eng chs ita deu ...). 
Substantially, the D7000.bin firmware file minus the first 2Mb (Bootloader) 3000000 is the last pos.

flag, pcbasn, xxx are 99% void like factory.
Others are languages.

128 Mb NAND: half is about useless

negan07, 2017-05-22:
target includes also useless .gz or .svn files: removing them helps saving some more MTD blocks.
/www.eng includes some N600 jpg logo files not used, can be removed.
One image file target/www.eng/image/apmodelink.jpg is a bmp: converting into real jpg with mogrify will save about 700k

negan07, 2017-05-26:
the startup.sh script has been updated with the garbage removal stuff

on 1.0.1.48 orig firmware
used space before garbage removal: 98%
used space after garbage removal and opkg installation: 93%

negan07, 2017-12-10:
Issue reopened

many small files creates overhead more than a single big one

opkg ipk's installation references are the examples
**info** dir & **status** file with **50 files** and a total amount of space of **~8Kb** only...

now they are located in **/usr/local/opkg** and this is located into rootfs part **mtd0**
the suggestion is: _why not pointing them safely into another unused partition_ ?

there are 2:
**language_CSY (1Mb)**
**language_TUR (1Mb)**

hoping this to be enough...

closed 2017-12-12 by negan07