negan07/ancistrus #6: Mobile Broadband 3G dongle / NUT for UPS Monitoring

ComputaBloke, 2017-08-23:
Thanks @negan07! Like others, I am a bit frustrated at the pace of firmware development from Netgear for what seems to be very capable hardware. Yet, I am pleased to find their code released under GPL and you already created a GitHub. Also very happy to follow your wiki to find I can telnet to the D7000 through 'debug' mode to explore and tinker around.

I am very interested to see if we could do more with the USB ports:
1. 3G/4G USB Dongle for Mobile Broadband Internet failover if VDSL drops. Perhaps some partial OpenWRT functions?
1. Integrated UPS monitoring/alerting through NUT

D7000 uses CPU: Broadcom BCM[9?]63138 @1GHz (DualCore ARM Cortex A9)
CPUInfo Features        : swp half thumb fastmult edsp tls

Further reading:
https://wiki.openwrt.org/doc/hardware/soc/soc.broadcom.bcm63xx
https://github.com/networkupstools/nut/

I am no expert and don't have much time, but have some systems/programming background so will try to follow your lead to share what I can.

Regards,
  David

negan07, 2017-11-12:
a brief support seems to be done from the housemother
dongle usb 4g with unix v3.x kernel driver compatibles are needed

negan07, 2017-12-09:
have a look to **/usr/etc/prepare_3g** script file to be added at startup
and then to these web pages

http://192.168.0.1/3g_cfg_bak.htm
http://192.168.0.1/3g_cfg_connecting.htm
http://192.168.0.1/3g_cfg_connect_result.htm
http://192.168.0.1/3g_cfg_h.htm
http://192.168.0.1/3g_cfg.htm
http://192.168.0.1/3g_SMS_h.htm
http://192.168.0.1/3g_SMS_outbox.htm
http://192.168.0.1/3g_SMS_read.htm
http://192.168.0.1/3g_SMS_send.htm
http://192.168.0.1/3g_wiz_upgrade_h.htm
http://192.168.0.1/3g_wiz_upgrade.htm
http://192.168.0.1/RST_st_3g.htm
http://192.168.0.1/st_3g_adsl_button.htm
http://192.168.0.1/st_3g_adsl.htm
http://192.168.0.1/st_3g_button.htm
http://192.168.0.1/st_3g.htm
http://192.168.0.1/test_3g.htm
http://192.168.0.1/upgrade3g.htm
http://192.168.0.1/wiz_3g_abort_conn.htm
http://192.168.0.1/wiz_3g_abort.htm
http://192.168.0.1/wiz_3g_bak.htm
http://192.168.0.1/wiz_3g_connecting.htm
http://192.168.0.1/wiz_3g_detecting.htm
http://192.168.0.1/wiz_3g_detect_modem.htm
http://192.168.0.1/wiz_3g_fail.htm
http://192.168.0.1/wiz_3g_h.htm
http://192.168.0.1/wiz_3g.htm
http://192.168.0.1/wiz_3g_identifying.htm
http://192.168.0.1/wiz_3g_initializing.htm
http://192.168.0.1/wiz_3g_nonetwork.htm
http://192.168.0.1/wiz_3g_nosim.htm
http://192.168.0.1/wiz_3g_nousb.htm
http://192.168.0.1/wiz_3g_ok.htm
http://192.168.0.1/wiz_3g_other_isp.htm
http://192.168.0.1/wiz_3g_pin_fail.htm
http://192.168.0.1/wiz_3g_prepare_detect.htm
http://192.168.0.1/wiz_3g_scanning.htm
http://192.168.0.1/wiz_3g_sim_checking.htm
http://192.168.0.1/wiz_3g_unsupmodem.htm
http://192.168.0.1/wiz_3g_updating.htm
http://192.168.0.1/wiz_3gwelc.htm
http://192.168.0.1/wiz_3gwelc_mbr1000v2.htm

plug a 3g dongle and see what happen: maybe some support is there already.

negan07, 2018-01-04:
telnet the router and:

`vi /www.eng/adv_index.htm`

press 'i'

scroll down finding:
`var dongle_3g_flag=@3g_is_visiable#;`

change the line as follows:
`var dongle_3g_flag=1;`

to save changes press ESC button & type:
:wq

now in the gui web interface on advanced/setup you will see:
`3G Mobile Setup`

with the connection settings

on advanced/administration you will find:
`Dongle Driver Update`

with the 3g support driver upload:

If someone has a 3g dongle with source driver module support can try to plug and see what happen.

ComputaBloke, 2018-01-10:
Will try thanks! 👍 
Now getting latest Ancistrus mono img (build 195) to begin the experiment, will love to try the full source build too, but this will save time for now.

FYI I have a [Huawei R205/E586 3G Mifi](https://bc.whirlpool.net.au/bc/hardware/?action=h_view&model_id=1381) device to try.. not a classic USB stick.
Only seeking an emergency failover I chose this device on cost and the integrated battery and don't need 4G. Will prefer to use it via USB dongle and it's 'hilink' capability, and expecting to be able to get a usb over ethernet (cdc-ether) - eth1/eth2 (may need a usb_modeswitch) - ref [1](http://www.monblocnotes.com/node/1943),[2](https://www.hackster.io/hologram/working-with-hilink-mode-on-the-huawei-e303-353-58ab21).
But in case of any challenges it seems reasonably flashable back to basic TTY serial modem, and the WiFi option is also available.
Fingers crossed.

negan07, 2018-01-10:
I've looked quickly:  don't know if there are full compiling sources for drivers.

the best way could be finding generic drivers valid for a chipset family group to use at once activated just in case of emergency

otherwise the obstacle is finding a compatible driver for arm cortex a9 for every kind of common usb stick...

in the `Dongle Driver Update` could be downloaded but producer has to furnish them...

ComputaBloke, 2018-01-11:
Enjoying browsing around terminal. :grin:
I agree. The features enabled through the `var dongle_3g_flag=1;` and others you helpfully listed are in a bare state. It seems the out of box config would require a PPP serial modem taking AT commands, and necessary drivers.

I am doing some more reading on USB devices/modems on ARM, ref: [1_embeddlinux](http://embeddlinux.blogspot.com.au/2013/01/how-to-interface-usb-modem-to-arm-board.html), [2_usb_modeswitch](http://www.draisberghof.de/usb_modeswitch), but can't locate a recent pre-compiled arm binary for usb_modeswitch at this time, may have to setup the build environment!
D7000 has a `/etc/udev/rules.d/` and that `prepare_3g` script looks partially useful.
I'll keep playing with it, and plug the 3G device in to do some tracing as to the effect.

negan07, 2018-01-11:
the openwrt mirrors may comes in help
https://wiki.openwrt.org/about/mirrors
only the 1rst one seems to be updated
http://ba.mirror.garr.it/mirrors/openwrt/releases/17.01.4/packages/arm_cortex-a9/packages/
there seems to be many usb drivers already compiled for this cpu and also for nut ...

ComputaBloke, 2018-01-12:
Getting closer and will resume tomorrow.

Inserting the Huawei 3G USB dongle into the D7000. I see this result in `dmesg`:
```
usb 3-1: new high-speed USB device number 2 using ehci-platform
INFO008C: Add device intf c703fa00, dev c6977800
INFO0C02:  filter pure storage dev
scsi1 : usb-storage 3-1:1.0
scsi 1:0:0:0: CD-ROM            Vodafone CD ROM (Huawei)  2.31 PQ: 0 ANSI: 2
```

This confirms I will need `usb-modeswitch` to make this modem work in 'hilink' mode (break out of the default 'USB mass storage device').

To `/etc/opkg.conf` I added the OpenWRT sources you recommended:
```
arch arm_cortex-a9 4
src/gz openwrt_arm_base http://ba.mirror.garr.it/mirrors/openwrt/releases/17.01.4/packages/arm_cortex-a9/base
src/gz openwrt_arm_packages http://ba.mirror.garr.it/mirrors/openwrt/releases/17.01.4/packages/arm_cortex-a9/packages
```
And ran `opkg update`
Then `opkg install usb-modeswitch`
```
Package: usb-modeswitch
Version: 2017-05-24-453da8e5-1
Depends: libc, libubox, libblobmsg-json, libusb-1.0
Status: unknown ok not-installed
Section: utils
Architecture: arm_cortex-a9
Size: 12812
Filename: usb-modeswitch_2017-05-24-453da8e5-1_arm_cortex-a9.ipk
Description: USB mode switching utility
Installed-Size: 12066
```
But unfortunately there are unresolved base dependencies. Any clues on where I might find these?:
```
Collected errors:
 * calculate_dependencies_for: Cannot satisfy the following dependencies for usb-modeswitch:
 *      libc *  libc *  libc *  libc *  libc *  libpthread *    librt *
 * opkg_solver_install: Cannot install package usb-modeswitch.
```

PS: The 'nut' packages also look very promising. I have a CyberPower PowerBric needing `usbhid-ups` drivers. So planning on getting these from the OpenWRT mirror too! 😄 
```
nut-driver-usbhid-ups_2.7.4-2_arm_cortex-a9.ipk
nut-web-cgi_2.7.4-2_arm_cortex-a9.ipk
nut_2.7.4-2_arm_cortex-a9.ipk
```

negan07, 2018-01-12:
no probably the drivers must be recompiled against our uclibc & kernel (libc)

so if there are the sources of those related packages they must be recompiled

ComputaBloke, 2018-01-13:
Went for a last-resort just to confirm. I ran the opkg install with `--force-depends`. But then the issue is that it has installed files in places not directly compatible with D7000 (e.g. `/etc/init.d`, `/etc/rc.common`) and worse, the main binary `/sbin/usbmode` doen't run probably due to missing libc (shell produces 'not found')

So you're right. Will need to custom compile and build a `usb-modeswitch` package from source to suit armD7000/anc. Is that something you have time to look into @negan07 ? Otherwise will try to follow your lead on Wiki to set up a [Debian build VM](https://github.com/negan07/ancistrus/wiki/Prepare-&-compile-sources).

I guess it's best to do this anyway since even if we can get the 3G modem on USB operational as an `eth5` interface, we'd still need to look at ethernet bridging (`/usr/sbin/brctl`) and eth fail-over monitoring config, and ideally some nvram/Startup/WebUI for it to make it easy for other users...

 On that front, I should further explore the existing bare 3G functions like usages of the nvram `hsdpa_*` and `wan_mode_3g=hsdpa`. Can you recommend any potentially similar firmware sources which may support 3G USB Dongles which we can look into to fast-track some development (like the R7000 perhaps)?

With this roadblock I will divert to *nut* project for a short while, hope to have less dependencies there :persevere:

ComputaBloke, 2018-01-13:
Unfortunately 'nut' also a no-go through those OpenWRT packages anyway. Need that underlying `libc, librt, libpthread, libgd`. So like `usb-modeswitch`, will most likely need to look at independent source/compilation for `nut`.

Just reading Linux kernel for D7000 is: `Linux version 3.4.11-rt19 (root@Debian) (gcc version 4.6.2 (Buildroot 2011.11-g00639bc-dirty) ) #2 SMP PREEMPT Wed Jan 3 09:48:00 CET 2018`. So *7 years old*! Apparently this is common for embedded-linux devices due to using Broadcom SDK .

Also OpenWRT 15.05 still used uclibc, but then moved to musl-libc, so definitely the latest 17.x streams won't be a compatible source, but there is still a slim chance to find compatible binary packages from older generations (will still need `--force-depends`).

Cracking this would be very useful since OpenWRT offers so much. Alternative being to cross-compile many packages to suit D7000 will not be much fun unless that can be automated in some fashion.

More to explore tomorrow... 👌

negan07, 2018-01-13:
use [this](https://github.com/negan07/ancistrus/issues/10) issue to talk about NUT to separate the questions (better move the above post too)

I can give a look to modeswitch for configuring and compiling
the kernel version maybe needs linking: there should be not problems with 3g support as they included a sort of setup

>  guess it's best to do this anyway since even if we can get the 3G modem on USB operational as an eth5 interface, we'd still need to look at ethernet bridging (/usr/sbin/brctl) and eth fail-over monitoring config, and ideally some nvram/Startup/WebUI for it to make it easy for other users... 

making it run first: the user friendly, later...

> On that front, I should further explore the existing bare 3G functions like usages of the nvram hsdpa_* and wan_mode_3g=hsdpa. Can you recommend any potentially similar firmware sources which may support 3G USB Dongles which we can look into to fast-track some development (like the R7000 perhaps)?

I know of some like zyxel and dlink broadcom supporting 3g fail-over but they are closed source

maybe finding the general broadcom developing package may help a bit (4.02L1 or something similar)

negan07, 2018-11-08:
anyone still interested to the 3g failsafe functionality ?

As it takes hardware for testing and a good amount of work-load external help is needed

If someone interested helping, post here: otherwise, this job will be freezed and issue closed in a while

negan07, 2018-11-13:
Now closed.

Open another if needed.

closed 2018-11-13 by negan07