negan07/ancistrus #7: State of the art

Jack12FR, 2017-11-12:
Hi negan07,

Great job :)
What is the state of the art ?
I've just bought the D7000 to replace my R7000 because I wanted to have 1 box instead of my ISP modem and my R7000 router behind.
The problem is that I didn't see that no open firmware was compatible with it.
So, I would like to know if your work works and if it is stable.
In fact, I just want an other firmware than netgear's because I need to have OpenVPN on it.

Thanks in advance for your help.

Cheers,

Kévin

negan07, 2017-11-12:
packets are stable as they run since february

Jack12FR, 2017-11-13:
And do you think I can use an OpenVPN client with it ?

negan07, 2017-11-13:
vpn has not been treated yet here

negan07, 2017-12-04:
openvpn package now up

closed 2017-12-04 by negan07