negan07/ancistrus #10: UPS monitoring/alerting through NUT

negan07, 2017-11-30:
http://networkupstools.org/

hardware needed for testing

ComputaBloke, 2018-01-14:
I have a CyberPower PowerBric needing `usbhid-ups` drivers. So planning on getting these from the OpenWRT mirror:

```
nut-driver-usbhid-ups_2.7.4-2_arm_cortex-a9.ipk
nut-web-cgi_2.7.4-2_arm_cortex-a9.ipk
nut_2.7.4-2_arm_cortex-a9.ipk
```

Currently blocked by OpenWRT compatibility, so opened #18 to look into it.
Otherwise will have to compile/build our own NUT packages specifically for anc/D7000.

negan07, 2018-01-14:
> Unfortunately 'nut' also a no-go through those OpenWRT packages anyway. Need that underlying `libc, librt, libpthread, libgd`. So like `usb-modeswitch`, will most likely need to look at independent source/compilation for `nut`.

> Just reading Linux kernel for D7000 is: `Linux version 3.4.11-rt19 (root@Debian) (gcc version 4.6.2 (Buildroot 2011.11-g00639bc-dirty) ) #2 SMP PREEMPT Wed Jan 3 09:48:00 CET 2018`. So *7 years old*! Apparently this is common for embedded-linux devices due to using Broadcom SDK .

> Also OpenWRT 15.05 still used uclibc, but then moved to musl-libc, so definitely the latest 17.x streams won't be a compatible source, but there is still a slim chance to find compatible binary packages from older generations (will still need `--force-depends`).

> Cracking this would be very useful since OpenWRT offers so much. Alternative being to cross-compile many packages to suit D7000 will not be much fun unless that can be automated in some fashion.

> More to explore tomorrow... 👌 

back from other issue

closed 2020-12-05 by negan07