negan07/ancistrus #12: Project page does not explain what it does

troffasky, 2017-12-20:
I have a Netgear D7000. I am interested in replacing the ghastly stock firmware with something better. I have found this project and this site too: 

https://negan07.github.io/ancistrus/

Once I've run these scripts, what will my D7000 do? Will it be running OpenWRT? Something similar? There is no explanation of what this project actually is/does.

negan07, 2017-12-20:
For a better knowledge, read the wiki board:
[https://github.com/negan07/ancistrus/wiki](url)

Openwrt and similar projects works with open gpl drivers like ethernet routers.

This is a **modem router** with broadcom/sercomm/netgear owned close source code driver and binaries: moving to an open firmware means losing the connection capatibility.

troffasky, 2017-12-20:
So just userspace is touched, not the kernel? Is the existing userspace replaced entirely or is it just tweaked?
Is it possible undo everything and revert to stock firmware?
Is there a feature comparison of stock firmware and ancistrus?
Why the name ancistrus?
Is it a fork of something else or is it a new "distro"?

negan07, 2017-12-21:
Read the wiki board again now updated with all the answers,

closed 2017-12-23 by negan07