negan07/ancistrus #13: 1.0.1.50 to 1.0.1.54

negan07, 2017-12-27:
[diff-D7000_V50_V54.zip](https://github.com/negan07/ancistrus/files/1589993/diff-D7000_V50_V54.zip)

- added ntp.htm www page
- user-defined ntp server in addition to netgear time servers
- renewed http pem cert
- web gui cookie support (genie ?)
- some login safety fixes (gui, ftp, console)
- some www .js fixes

infabius, 2017-12-28:
Hi man you can create a build 196?

infabius, 2017-12-28:
The build 195 it's so great!

phoenix1975, 2017-12-28:
You can create firmware 1.0.1.54 with the ancistrus changes

negan07, 2017-12-28:
the build number is the commit number

working on a qos-sqm upgrade:
it's better to include it on a new build

infabius, 2017-12-28:
Thank you so much man I wait for your updates, but a web gui made on purpose is not possible?

negan07, 2017-12-28:
it is in the project plan

infabius, 2017-12-28:
Well done i'll with your updates

closed 2018-01-06 by negan07