negan07/ancistrus #14: Restart stuck (build 195)

BoostOver, 2018-01-01:
When push on web GUI restart system or start the system for the first time ,router stuck with 1 and 2 LED ETH on for more and more minute.
Some time the restart go OK, but it is long (with 1 and 2 LED ETH ON)
I'm not only, in the forum (hwupgrade.it) there are others member .
And if the power fails and we are not at home?

Anyway, you did a great job.
Many thanks

negan07, 2018-01-02:
In fact i've noticed this stuck but I didn't take too care of it because of many other modifications in act.

Kernel is the key without doubt.

To analize the problem deeply serial debug could help but it require opening the router and pinning the board with warranty issues.

Safer and quicker to find which of the patches inserted gives troubles.

negan07, 2018-01-03:
both kernel & uclibc fpu neon created the problem maybe
now removed
reverted to soft float

the release zip file has been bumped in the morning
after modifications 24 hours of hot&cold reboots never stucked

master committed too

BoostOver, 2018-01-04:
Thanks, master ;-)

closed 2018-01-04 by BoostOver