negan07/ancistrus #15: Build 195 don't build

superpippo82xxx, 2018-01-04:
Hi, i'm trying to make build 195 by my self but compilation don't finish.
here is where it stop:
```
Downloading file:///home/user/ancistrus/build/Packages.gz.
Updated source 'local'.
Package nvram (0.1.0) installed in root is up to date.
Package utelnetd (0.1.11) installed in root is up to date.
Package iproute2 (4.10.0) installed in root is up to date.
Package hdparm (9.52) installed in root is up to date.
Package openvpn (2.4.4) installed in root is up to date.
Package ancistrus-core (0.1.0) installed in root is up to date.
Package netperf (2.7.0) installed in root is up to date.
Package network-test (0.1.0) installed in root is up to date.
Package qos-sqm (0.1.0) installed in root is up to date.
Package opkg (0.3.4) installed in root is up to date.
Package smartmontools (6.5) installed in root is up to date.
mv: cannot stat '/home/user/ancistrus/D7000_V1.0.1.50_WW_src/Source/target/usr/lib/opkg/*': No such file or directory
Makefile:96: recipe for target 'monolithic' failed
make[2]: *** [monolithic] Error 1
Makefile:87: recipe for target 'third-party' failed
make[1]: *** [third-party] Error 2
Makefile:44: recipe for target 'source' failed
make: *** [source] Error 2
Makefile:32: recipe for target 'sources' failed
make: *** [sources] Error 2
```

i haven't done any modification yet
Another thig i've changed shell from dash to bash to solve other compilation problem
Thanks

negan07, 2018-01-04:
the packages cannot be up to date
It should be the first time installing

use [this package](https://github.com/negan07/ancistrus/releases/download/195/ancistrus-195_sources.zip) the others self created when creating the release refers to the boot stuck one.

or maybe git clone the last commit master

which operative system ? arch ? 
dash cannot be used for compiling and debian is recommended

negan07, 2018-01-04:
the error comes out because you didn't install anything in /home/user/ancistrus/D7000_V1.0.1.50_WW_src/Source/target/usr/lib/opkg/

the installed files has gone somewhere out

superpippo82xxx, 2018-01-05:
sorry but probably i was not clear.
i've installed fresh debian 9 as wiki
I followed all the steps of the wiki
i run " git checkout 701d94c993822fb1fa8d459044536ab2e931d69c" to be at 195
after i run "make" and after it stop
if i run 'make sources_orig' ( it create orig image)

about dash is default in debian since DebianSqueeze [https://wiki.debian.org/Shell](url)

negan07, 2018-01-05:
the source pack can be compiled with bash only as reported [here](https://github.com/negan07/ancistrus/wiki/Prepare-&-compile-sources) and [here](https://github.com/negan07/ancistrus/blob/master/scripts/hostprepare.sh)
```
# debian has dash posix compliant only default shell, faster than bash but with tiny command set
# e.g. += support is missing creating problem on making and many more
# switching from dash to bash (action is reversible)
# answer no to next question
# sudo dpkg-reconfigure dash
sudo dpkg-reconfigure -p critical dash

```
to compile #195 use [this](https://github.com/negan07/ancistrus/releases/download/195/ancistrus-195_sources.zip) package because the build has been modified after moving to .54 so the master commit can't match

> moving to commit "988f457b905598eb76007ba187b70cdbd471125c" i can create ancitrus image based on 1.0.1.54 but image is not working, VDSL2 sync pppoe connect but can't ping any host. revert to original image fix the problem<

you started compiling from [this](https://github.com/negan07/ancistrus/commit/988f457b905598eb76007ba187b70cdbd471125c) commit

that point of the code is older than [this](https://github.com/negan07/ancistrus/commit/1be5af0041f7f59be909d534736bc89c6e4c1ca5), the first one fixing the kernel & uclibc code compilation.

the build you obtained not working is still affected by NEON instructions collateral side effects applied on the non compliant proprietary code (eg the drivers)

as suggested, clone the master snapshot

superpippo82xxx, 2018-01-06:
Ok, i understand now.
now cloned master and  `make dist_clean && make dist_clean_toolchain`

this is the new problem
`...
Making all in doc
Making all in vendor
Making all in tests
Making all in unit_tests
work: make ancistrus-core-0.1.0
work: make netperf-2.7.0
Making all in src
work: make network-test-0.1.0
Nothing to do: run 'make install' to install.
work: make qos-sqm-1.2.0-13.020
Nothing to do: run 'make install' to install.
work: make opkg-utils-0.3.4
work: make opkg-0.3.4
Making all in libopkg
Making all in src
work: make smartmontools-6.5
work: prepare package nvram-0.1.0
work: prepare package lzo-2.10
Nothing to install...
work: prepare package zlib-1.2.11
Nothing to install...
work: prepare package libarchive-3.3.1
Nothing to install...
work: prepare package utelnetd-0.1.11
work: prepare package iproute2-4.10.0
work: prepare package hdparm-9.52
work: prepare package openvpn-2.4.4
work: prepare package ancistrus-core-0.1.0
work: prepare package netperf-2.7.0
work: prepare package network-test-0.1.0
work: prepare package qos-sqm-1.2.0-13.020
install: cannot stat 'rc_qos': No such file or directory
Makefile:10: recipe for target 'install' failed
make[3]: *** [install] Error 1
Makefile:56: recipe for target 'prepare_package' failed
make[2]: *** [prepare_package] Error 1
Makefile:94: recipe for target 'third-party' failed
make[1]: *** [third-party] Error 2
Makefile:44: recipe for target 'source' failed
make: *** [source] Error 2
Makefile:32: recipe for target 'sources' failed
make: *** [sources] Error 2
user@debian9:~/ancistrus$`

negan07, 2018-01-06:
you are right
the rc_qos files are all missed...

...because wrongly included into .gitignore

git pull the master and do `make` you should skip the error now...

superpippo82xxx, 2018-01-07:
good job, now i can create image.
One thing, package imagemagick  is not istalled by your script but is needed to compile image

SOURCE: make FS
Input File Name /home/user/ancistrus/D7000_V1.0.1.54_WW_src/Source//target/vmlinux.lz, Output File Name /home/lorenzo/ancitarget/kernel_cksum
checksum 345e2abc
Done!
SOURCE: FS garbage orphans cleanup
make[1]: mogrify: Command not found
Makefile:153: recipe for target 'fs' failed
make[1]: *** [fs] Error 127
Makefile:44: recipe for target 'source' failed
make: *** [source] Error 2
Makefile:32: recipe for target 'sources' failed
make: *** [sources] Error 2

negan07, 2018-01-07:
strange
mogrify was built in also on ubuntu with unity and here with gnome too

...
negan07@Debian:~/Desktop$ dpkg -l | grep imagemagick
ii  imagemagick                           8:6.9.7.4+dfsg-11+deb9u4                    amd64        image manipulation programs -- binaries
ii  imagemagick-6-common                  8:6.9.7.4+dfsg-11+deb9u4                    all          image manipulation programs -- infrastructure
ii  imagemagick-6.q16                     8:6.9.7.4+dfsg-11+deb9u4                    amd64        image manipulation programs -- quantum depth Q16
...

which graphic environment did u install if any ?
hostprepare.sh tries to obtain the principal development packages needed but cannot follow all the os variations

superpippo82xxx, 2018-01-07:
i've installed any graphic enviroment, only basic tools and ssh

negan07, 2018-01-07:
no gui interface ? console only ?
this is the reason maybe

now added imagemagick to hostprepare.sh

superpippo82xxx, 2018-01-07:
yes no gui, console only.

closed 2018-01-07 by superpippo82xxx