negan07/ancistrus #16: 35b support?

infabius, 2018-01-11:
Hi u can try to make a new update with 35b on vdsl settings?

negan07, 2018-01-11:
The profiles available depend on adsl broadcom driver strictly (providers too).
Broadcom must provide the support if any: nothing else to do.

Actually there is: 
ADSL PHY: AnnexA version - `A2pvbH042n.d26b`
Binary file is: `adsl_phy.bin`
Executable is: `xdslctl`
Profiles are: `8a 8b 8c 8d 12a 12b 17a 30a`

Drivers are usually low level monolithic code.
Sometimes with a little bit of luck, it should be possible to try to find out an updated `adsl_phy.bin` broadcom driver and the related `xdslctl` (ARM compatible) from other source codes of other brands with v35b support.

Actually `adsl_phy.bin` version is `A2pvbH042n.d26b`
In any case an updated one is appreciated.

superpippo82xxx, 2018-01-12:
i've a zyxel router with this firmware:
ADSL PHY: AnnexA version - A2pvfbH043e.d26n
$ adsl profile --show

Modulations:
        G.Dmt   Enabled
        G.lite  Enabled
        T1.413  Enabled
        ADSL2   Enabled
        AnnexL  Enabled
        ADSL2+  Enabled
        AnnexM  Enabled
        VDSL2   Enabled
VDSL2 profiles:
        8a      Enabled
        8b      Enabled
        8c      Enabled
        8d      Enabled
        12a     Enabled
        12b     Enabled
        17a     Enabled
        30a     Disabled
        35b     Enabled
        US0     Enabled
Phone line pair:
        Inner pair
Capability:
        bitswap         On
        sra             On
        trellis         On
        sesdrop         Off
        CoMinMgn        Off
        24k             On
        phyReXmt(Us/Ds) Off/On
        Ginp(Us/Ds)     On/On
        TpsTc           AvPvAaPa
        monitorTone:    On
        dynamicD:       On
        dynamicF:       Off
        SOS:            On
        Training Margin(Q4 in dB):      -1(DEFAULT)

Which files you need? i've source code provided by Zyxel

negan07, 2018-01-12:
good job

which zyxel model exactly ?

superpippo82xxx, 2018-01-12:
zyxel VMG8823-B50B

infabius, 2018-01-12:
https://mega.nz/#F!Pd8TgAJA!t-Yj3SktM8dNqvoIwiWbmg New driver adsl and more

negan07, 2018-01-12:
bad news and good news
the xdslctl seems compatible with the arch
but it needs some shared libs:
`libbcm_crc.so`
`libbcm_flashutil.so`

more i have looked the adsl drivers deeply and i've seen that `adsl_sdram` is versioned too
so it should be better to have `adsl_sdram` of the same version ...043e.d26n too

superpippo82xxx, 2018-01-13:
i can't find adsl_sdram... 
Here missing library
[lib.zip](https://github.com/negan07/ancistrus/files/1628267/lib.zip)

negan07, 2018-01-18:
bad news and good news

the bad one is that trying to install the adsl driver above with 35mhz vectoring not working
here's the dmesg log output:

```
*** dslThread dslPid=2288
BcmAdsl_Initialize=0xBF152028, g_pFnNotifyCallback=0xBF187E4C
AdslCoreSetSDRAMBaseAddr: pAddr=0x0FE00000 sdramPageAddr=0xCFE00000
BcmAdslCoreCalibrate: (cnt1-cnt)=0x039388d7, adslCoreCyclesPerMs=1000007, BCMOS_MSEC_PER_TICK =1
AdslCoreSetSdramImageAddr: lmem2=0x10500000, sdramSize=0x19d460
pSdramPHY=0xCFFFFFF8, 0x1B7749 0xDEADBEEF
AdslCoreSetSdramImageAddr: sdramPageAddr=0xcfe00000, sdramImageAddr=0xcff00000, sdramPhyImageAddr=0x10500000
*** AdslCoreSetXfaceOffset: data[0]=0xFFFFFE6F data[1]=0x190 ***
Failed to read SDRAM image from '/etc/adsl/adsl_phy.bin'.
AdslCoreHwReset:  Failed to load ADSL PHY image (0)
AdslCoreSetSdramImageAddr: lmem2=0x10500000, sdramSize=0x19d460
pSdramPHY=0xCFFFFFF8, 0xB0AF 0xF8FFBD27
AdslCoreSetSdramImageAddr: sdramPageAddr=0xcfe00000, sdramImageAddr=0xcff00000, sdramPhyImageAddr=0x10500000
*** AdslCoreSetXfaceOffset: data[0]=0xFFFFFE6F data[1]=0x190 ***
Failed to read SDRAM image from '/etc/adsl/adsl_phy.bin'.
AdslCoreHwReset:  Failed to load ADSL PHY image (0)
AdslCoreSetSdramImageAddr: lmem2=0x10500000, sdramSize=0x19d460
pSdramPHY=0xCFFFFFF8, 0xB0AF 0xF8FFBD27
AdslCoreSetSdramImageAddr: sdramPageAddr=0xcfe00000, sdramImageAddr=0xcff00000, sdramPhyImageAddr=0x10500000
*** AdslCoreSetXfaceOffset: data[0]=0xFFFFFE6F data[1]=0x190 ***
Failed to read SDRAM image from '/etc/adsl/adsl_phy.bin'.
AdslCoreHwReset:  Failed to load ADSL PHY image (0)
dgasp: kerSysRegisterDyingGaspHandler: dsl0 registered 
ADSL led state change to <2>
bcmxtmcfg: DS xDSL G.inp Mode = DISABLED 
bcmxtmcfg: xDSL G.Fast Mode = DISABLED 
XTM Init: Ch:0 - 1553 rx BDs at 0xffdbc000
XTM Init: Ch:1 - 16 rx BDs at 0xffde6000
bcmxtmrt: PTM/ATM Non-Bonding Mode configured in system
bcmxtmcfg: Out of sequence call to XTM_ASM_HANDLER::Uninitialize().  Recovering.
bcmxtmcfg: ATM Bonding configured in system. Fallback mode = Enabled 
bcmxtmcfg: Bonding State is DATA_IDLE 
bcmxtmcfg: SID MODE SET to 12 BIT MODE 
bcmxtmcfg: ATM Bonding Mgmt Log Area = c715dd08 
```
the main reason is:
`Failed to read SDRAM image from '/etc/adsl/adsl_phy.bin'.`
so _adsl_sdram is mandatory_ and must be founded out to make further experiments

the good news is that searching in the [link proposed above](https://github.com/negan07/ancistrus/issues/16#issuecomment-357249230) there are other adsl driver versions, of the **A2pvbH** series and of the **A2pvfbH** series (35b ?):
A2pvbH - **042r 042u**
A2pvfbH - **043f 043g 043k**
the one proposed above is the **A2pvfbH043f**
the original built in driver is the **A2pvbH042n**

Trying out the most recent built **A2pvbH042u** looking good.

```
# xdslctl profile --version
xdslctl version 1.2
ADSL PHY: AnnexA version - A2pvbH042u.d26b
```

Can't say what performance differences are there in brief, but working.

negan07, 2018-03-04:
bumping this issue because while working on dsl tweaks & fixes some interesting came out

`xdslctl` not revealing all its stuffs on --help like --vectoring option

typing:

`xdslctl info --vectoring`
You can see the vectoring state
i have response:
```
Vectoring state: 5
VCE MAC Address: 0:0:0:0:0:0
Total error samples Ethernet pkts sent: 0
Total error samples Ethernet pkts discarded: 0
Total error samples statuses sent: 0
Total error samples statuses discarded: 0
```
according to this definitions:
```
#define VECT_WAIT_FOR_CONFIG  0
#define VECT_FULL             1
#define VECT_WAIT_FOR_TRIGGER 2
#define VECT_RUNNING          3
#define VECT_DISABLED         4
#define VECT_UNCONFIGURED     5
```
it looks unconfigured but there's nothing to do for configuring

`xdslctl.bin configure --vectoring on`
(xdslctl.bin to skip the configure parser)

option unable

0d1n0, 2018-03-27:
the main reason is:
Failed to read SDRAM image from '/etc/adsl/adsl_phy.bin'.
so adsl_sdram is mandatory and must be founded out to make further experiments

---------

adsl_sdram.bin  is generated when you build kernel image .... so the base of generation is the binary drivers probably. i'm investigating about it.

0d1n0, 2018-03-27:
x

0d1n0, 2018-03-27:
Some News,
I have just loaded without problem  adsl driver A2pvbH042j2.d26b of a tplink VR2800 V1 and it seem to work without problem on my D7000 with original xdsctl. you don't need sdram.bin but only '/etc/adsl/adsl_phy.bin and to be binary version aligned adsldd.ko  module of same firmware version.however  i didn't change adsldd on my d7000.

root@D7000:$ xdslctl --version
xdslctl version 1.2
ADSL PHY: AnnexA version - A2pvbH042j2.d26b
root@D7000:$ www
root@D7000:$ xdslctl info
xdslctl: ADSL driver and PHY status
Status: Showtime
Last Retrain Reason:    8000
Last initialization procedure status:   0
Max:    Upstream rate = 969 Kbps, Downstream rate = 18420 Kbps

I cannot try vdsl link because i have adsl+ here but probably it will work without problem with 35b too.

to follow a list with version and some model that should support 35b from where you can get adsl_phy.bin and adsldd.ko

ASUS DSL-AC88U_FW_V1_10.05_Build503  version of broadcom adsl_phy. bin A2pvbH042j2 built 01/13/2016 14:08:30 
TPLINK VR2800V1 *iBroadcom DSL Version  support 35b  A2pvbH042j2  built 01/13/2016 14:08:30  
zyxel  VMG8823-B50B_V5.13(ABEJ.2)C1 Broadcom DSL Version A2pvfbH043i1  Built on 05/07/2017 01:28:20

0d1n0, 2018-03-27:
Ok now i just loaded firmware B2pvfbH043g.d26p on my D7000 . The problem was adsldd.ko

dgasp: kerSysRegisterDyingGaspHandler: dsl0 registered
ADSL led state change to <2>
***dslThreadTerminate: Unlocking dslThread dslPid=31683
***dslThread: Exiting ***
***dslThreadTerminate: Exiting
kerSysDeregisterDyingGaspHandler: dsl0 is deregistering
kerSysDeregisterDyingGaspHandler: dsl0 is deregistered
*** dslThread dslPid=1682
BcmAdsl_Initialize=0xBFB11028, g_pFnNotifyCallback=0xBFB49150
AdslCoreSetSDRAMBaseAddr: pAddr=0x0FE00000 sdramPageAddr=0xCF800000
AdslCoreSetSdramImageAddr: lmem2=0x10500000, pgSize=0x800000 sdramSize=0x19E628
AdslCoreSetSdramImageAddr: lmem2(0x500000) vs ADSL_PHY_SDRAM_BIAS(0x500000); origAddr=0xCFD00000 newAddr=0xCFD00000
pSdramPHY=0xCFFFFFF8, 0x1B782B 0xDEADBEEF
AdslCoreSetSdramImageAddr: sdramPageAddr=0xcf800000, sdramImageAddr=0xcfd00000, sdramPhyImageAddr=0x10500000
*** AdslCoreSetXfaceOffset: data[0]=0xFFFFFE6F data[1]=0x190 ***
*** XfaceOffset: 0x190 => 0x190 ***
*** AdslCoreSetSdramTrueSize: data[0]=0x9730DDFF data[1]=0x68CF2200 ***
*** AdslCoreSetSdramTrueSize: data[0]=0xFFDD3097 data[1]=0x22CF68 ***
*** PhySdramSize got adjusted: 0x19E628 => 0x22CF68 ***
AdslCoreSharedMemInit: shareMemSize=864368(864368)
__AdslCoreHwReset: pAdslX=0xfc700190
__AdslCoreHwReset: pAdslX->sdramBaseAddr=0x0000d0af, pAdslX->gfcTable[]=0x0000d0cf, adslCorePhyDesc.sdramImageAddr=0xcfd00000
AdslCoreHwReset:  pLocSbSta=c1768000 bkupThreshold=3072
AdslCoreHwReset:  AdslOemDataAddr = 0xCFE5FDD0, time=12 ms
VersionInfo: B2pvfbH043g.d26p
***BcmDiagsMgrRegisterClient: 0 ***
dgasp: kerSysRegisterDyingGaspHandler: dsl0 registered
ADSL led state change to <2>
BcmAdsl_Initialize=0xBFB11028, g_pFnNotifyCallback=0xBFB49150
dgasp: kerSysRegisterDyingGaspHandler: dsl0 registered
ADSL led state change to <2>
BcmAdsl_Initialize=0xBFB11028, g_pFnNotifyCallback=0xBFB49150
dgasp: kerSysRegisterDyingGaspHandler: dsl0 registered
ADSL led state change to <2>

oot@D7000:$ cd /tmp/x
root@D7000:$ ./xdslctl profile --show

Modulations:
        G.Dmt   Enabled
        G.lite  Enabled
        T1.413  Enabled
        ADSL2   Enabled
        AnnexL  Enabled
        ADSL2+  Enabled
        AnnexM  Disabled
        VDSL2   Enabled
VDSL2 profiles:
        8a      Enabled
        8b      Enabled
        8c      Enabled
        8d      Enabled
        12a     Enabled
        12b     Enabled
        17a     Enabled
        30a     Enabled
        35b     Enabled
        US0     Enabled
Phone line pair:
        Inner pair
Capability:
        bitswap         On
        sra             Off
        trellis         On
        sesdrop         Off
        CoMinMgn        Off
        24k             On
        phyReXmt(Us/Ds) Off/On
        Ginp(Us/Ds)     On/On
        TpsTc           AvPvAa
        monitorTone:    On
        dynamicD:       On
        dynamicF:       Off
        SOS:            On
        Training Margin(Q4 in dB):      -1(DEFAULT)
	
	
root@D7000:$ ./xdslctl --version
./xdslctl version 1.2
ADSL PHY: AnnexA version - A2pvfbH043i1.d26p

negan07, 2018-03-27:
good work
post here the driver binary files you obtained to make some tries (adsldd.ko & phy.bin)

compiling kernel the driver is simply copied from a dir to another because it comes already as a binary standalone

but the driver version is not the real problem because also 042u can work with vectoring

the problem is maybe as you found out the adsldd.ko against a compatible kernel

http://forum.kitz.co.uk/index.php/topic,20343.255.html
this router DGA4130 (and also many similar brands all around with different names) has 35b vect support and uses the same broadcom kernel 3.4 of D7000 and includes the already tested 042u

it should be useful to extract from DGA4130 firmware the driver binaries (better the entire dump filesystem)

these routers are branded but the can be unlocked: they should include ssh so it should be easy to dump the binary files on local machine using scp

If someone is able to find out / dump / receive the binary adsl driver files from DGA4130 or similar post them here they could result useful.

At least some of the files involved should be located at `/lib/modules` and they are:

```
bdmf.ko
rdpa_gpl.ko
rdpa.ko
rdpa_mw.ko
chipinfo.ko
bcmxtmrtdrv.ko
bcmxtmcfg.ko
pktflow.ko
pktrunner.ko
adsldd.ko
bcm_enet.ko
nciTMSkmod.ko
bcmvlan.ko
pwrmngtd.ko
rdpa_cmd_tm.ko
rdpa_cmd_spdsvc.ko
rdpa_cmd_ds_wan_udp_filter.ko
rdpa_cmd_dr.ko
/usr/sbin/xdslctl
```

with:
**bcmxtmrtdrv.ko
bcmxtmcfg.ko
adsldd.ko
xdslctl**

mostly involved

0d1n0, 2018-03-27:
Follow a partial list of some models that have 35b support at least in latest firmaware images:
Asus DSL_AC87VG
Asus  DSL-AC88U
Tplink Archer VR2800V1
Zyxel VMG8823-B50B
TIM Technicolor DGA4130

0d1n0, 2018-03-27:
You are sure that 042u with  dga4130 can support 35b? It is tested?

0d1n0, 2018-03-27:
http://anonfile.com/L7QfVcdebf/zz.rar adslphy, adsldd.ko and xdsclt used in test

0d1n0, 2018-03-27:
I think that 042u cannot support profile 35b but can support vectoring with profile 17a and 30a. (max 100 or 200 mbits).I think that support for 35mhz  band 35b  AnnexQ band is present only with 43 version of broadcom drivers.

0d1n0, 2018-03-28:
Ok should be solved 042u support 35b probably. The problem is that you should  update xdslctl with  the adsl_phy.bin to manage new settings. I have one that seem compatible but should be better  to recover original xdslclt that came with  adsl_phy.bin to be version aligned.
here the link to binary xdslclt
http://anonfile.com/Ded9Wad2b3/xdslctl
 xdsctl profile --show 

OLD COMMAND:
        VDSL2   Enabled
VDSL2 profiles:
        8a      Enabled
        8b      Enabled
        8c      Enabled
        8d      Enabled
        12a     Enabled
        12b     Enabled
        17a     Enabled
        30a     Enabled
        US0     Enabled
Phone line pair:
        Inner pair

--------------------------------------------------------------------------------------------------------------------------------

NEW COMMAND

        VDSL2   Enabled
        Gfast   Disabled
VDSL2 profiles:
        8a      Enabled
        8b      Enabled
        8c      Enabled
        8d      Enabled
        12a     Enabled
        12b     Enabled
        17a     Enabled
        30a     Enabled
        35b     Enabled
        US0     Enabled
Phone line pair:
        Inner pair

negan07, 2018-03-30:
tested the above driverset with newer 023i1 and different adsldd.ko xdslctl

the good news is that there is a valid carry
but no valid connection
both xdslctl give same result

reverting back to old original xdslctl with new adsldd.ko module gives segfault
and the newer driver without adsldd.ko can't handle carry

there is (only) to find out the right driverset: at least adsldd.ko, adsl_phy.bin & xdslctl
DGA4130 looks to be compiled with the same sdk

0d1n0, 2018-04-05:
http://anonfile.com/K1K8aee2b9/AGTEF1.0.4.rar 
in the file binaries (kernel and userspace) of dga4130 version 1.0.4 that use A2pvbH042u.
Last version of firmware is 1.10 but it use A2pvfbH043k and you should substitute every kernel dependent modules to work properly .
In the help of xdslctl there is no 35b profile but probably you can activate it with 
xdslctl configure --profile BrcmPriv1 command line.
Adsl part of driver work properly on my d7000 tested with only substitution of adsl_phy.bin and xdslctl but i cannot test vdsl.

0d1n0, 2018-04-07:
Yes BrcmPriv1 is 35b  old names probably .I can extract 1.10 without problem but i suggest to  test this set of module to see if they work .

negan07, 2018-04-08:
Very Good. Doing some tests of it then will see if packaging all the driverset or at least adsldd.ko & bins

negan07, 2018-04-17:
full driverset 042u.d26o uploaded to repository: if vectoring & 35mhz is available there should be nothing more to do

negan07, 2018-04-17:
already tested 043k.d26r but no carry: and a msg error appears at every xdslctl invocation
maybe adsldd.ko is the issue other modules and library seems to load successfull

did agev bumped kernel in the meanwhile ?

didn't try 043k.d26o but shouldn't make so practical difference

Coliuto, 2018-04-20:
Hi negan, I've tried the latest firmare 322 with a VDSL 35b (using the package xdsl-driver-full-new) but unfortunately it does not work.
As you can see from these screens:  https://i.imgur.com/eH2b6o8.png - https://i.imgur.com/rabrREL.png
There's something else I could try to make it work? Thank you.

negan07, 2018-04-20:
no carry and no link or carry and no link ?
from the image seems connected

tried to force profile manually with 

`xdslctl.bin --profile BrcmPriv1` 

?

post also `dmesg` output after cmd execution

negan07, 2018-04-21:
this package must be installed:
`opkg update && opkg install xdsl-driver-full-new`

the `--profile` option with or without .bin should be the same

any `dmesg` output after 

`xdslctl.bin --profile BrcmPriv1` 
?

Coliuto, 2018-04-21:
The package is already installed but using the command "xdslctl.bin" same result: https://i.imgur.com/ZeG8wE8.png

negan07, 2018-04-22:
type `dmesg` after cmd execution
and post here display content

negan07, 2018-04-22:
yes profile not including configure
if running  `xdslctl configure`
 after setting the gui should be the same: the profile is not actived
the `dmesg` output should be useful to see if it reports an error message maybe

0d1n0, 2018-04-22:
Yes could be that vdsl slam support only 17a (177.28).

Coliuto, 2018-04-22:
> type dmesg after cmd execution
> and post here display content

Now the command seems to be working but no synchronization at all. This is the output with dmesg:

D7000 login: admin
Password:
admin@D7000:$ xdslctl.bin configure --profile BrcmPriv1
admin@D7000:$ dmesg
t: MAC address: a0 63 91 0d 7e 98
[DoCreateDeviceReq.775]: register_netdev
[DoCreateDeviceReq.780]: register_netdev done
bcmxtmcfg: Reserve TxQueueIdx=0 for vcid 0
bcmxtmrt: Traffic Type is 2.  TxPAF control-0.  Disable RDP-US-Bonding.
bcmxtmrt: Egress TM Q 0 Setup. Buffering/Q - 128
bcmxtmcfg: Connection UP, LinkActiveStatus=0x1, US=21600000, DS=106622000
ptm0.1 MAC address set to A0:63:91:0D:7E:95
netdev path : ptm0.1 -> ptm0
BCMVLAN : ptm0 mode was set to RG
wan interface=[ptm0.1], index=21
wan interface=[ppp1], index=22
lan interface=[group1], index=15
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
netlink: 20 bytes leftover after parsing attributes.
eth3 (Ext switch port: 2) (Logical Port: 10) Link DOWN.
group1: port 3(eth3) entered disabled state
eth3 (Ext switch port: 2) (Logical Port: 10) Link UP 10 mbps full duplex
[... 232 lines cut ...]
ADSL led state change to <2>
message received before monitor task is initialized kerSysSendtoMonitorTask
message received before monitor task is initialized kerSysSendtoMonitorTask
Line 0: xDSL link down
ADSL led state change to <2>
message received before monitor task is initialized kerSysSendtoMonitorTask
Line 0: xDSL G.994 training
ADSL led state change to <2>
message received before monitor task is initialized kerSysSendtoMonitorTask
message received before monitor task is initialized kerSysSendtoMonitorTask
Line 0: xDSL link down
ADSL led state change to <2>
message received before monitor task is initialized kerSysSendtoMonitorTask
Line 0: xDSL G.994 training
ADSL led state change to <2>
message received before monitor task is initialized kerSysSendtoMonitorTask
message received before monitor task is initialized kerSysSendtoMonitorTask
Line 0: xDSL link down
ADSL led state change to <2>
message received before monitor task is initialized kerSysSendtoMonitorTask
Line 0: xDSL G.994 training
ADSL led state change to <2>
message received before monitor task is initialized kerSysSendtoMonitorTask
message received before monitor task is initialized kerSysSendtoMonitorTask
admin@D7000:$

Coliuto, 2018-04-22:
> Yes could be that dslam have some problem if it 177.28 with 35b.
> 177.152 seem to be without problem with 35b.

No, The issue concerning the DSLAM version is relating only with Lantiq chipset. Using broadcom chipset every DSLAM version is ok for 35b. This is my line on the Asus DSL-AC88U (broadcom):
https://i.imgur.com/CDlxyle.png
https://i.imgur.com/90gTX3l.png

0d1n0, 2018-04-22:
What modem is the second screenshot?

negan07, 2018-04-22:
it seems to enter into a training loop...

try activating the 30a profile and see what happen

0d1n0, 2018-04-22:
i found this  interesting screenshot from a forum about dga4130
[image]
http://forum.kitz.co.uk/index.php?topic=20343.195 original post 347. version A2pvbH042u.d26o

Coliuto, 2018-04-23:
> which firmware version in these screenshot?

This is the Asus DSL-AC88U with the latest firmware.

Coliuto, 2018-04-23:
> bcmxtmcfg: Connection UP, LinkActiveStatus=0x1, US=21600000, DS=106622000 in initial handshake is at 100... not 200... could be before command?

Yes, those are the values of the previous synchronization, before the "xdslctl.bin configure --profile BrcmPriv1" command

Coliuto, 2018-04-23:
On july 2017 you mean? I think yes. This is a screenshot by an italian user on June 2017 and the 35b was already supported by the DSL-AC88U: https://www.hwupgrade.it/forum/showpost.php?p=44848276&postcount=135

Coliuto, 2018-04-23:
According to the Asus changelog the driver 43g have been upgraded on March 2017 (with the second firmware 1.10.00_build350). So I dont know if the initial firmware is compatible with the 35b profile, because I've bought the router on April 2017.

0d1n0, 2018-04-23:
ok thank you

negan07, 2018-04-23:
driver version looks not to be crucial
there seems to be more

a) the dga4130 1.0.4 not full 35b compatible ?
b) kernel level stage configurations ?

1) assure dga4130 1.0.4 works with 35b
2) driverset not enough: need all the dumped filesystem, as much as possible
vmlinux, kernel modules, drivers, binaries, boot files & scripts, all can help

(trying to decrypt the binary cobalt .rbi with no success)

Coliuto, 2018-04-24:
I can assure that the DGA4130 with 1.0.4 firmware (but even with the previous 1.0.3) works perfectly with 35b profile.

0d1n0, 2018-04-24:
@Coliuto you can check firmware broadcom version of dga4130 1.0.4 that work with 35b profile?

0d1n0, 2018-04-24:
@negan07  https://expirebox.com/download/1cf499963fc28f832eefc54ef5b5a507.html agtef 1.0.4 squashfs

0d1n0, 2018-04-24:
this is more interesting from dva-5592 20170908

https://expirebox.com/download/ea784895227571f9307a5e955432c4fc.html

xdsctls output with 
./xdslctl version 1.2
ADSL PHY: AnnexA version - A2pvbH042u.d26o

       ./xdslctl --help
root@D7000:$ ./xdslctl  profile --show

Modulations:
        G.Dmt   Disabled
        G.lite  Disabled
        T1.413  Disabled
        ADSL2   Disabled
        AnnexL  Disabled
        ADSL2+  Enabled
        AnnexM  Disabled
        VDSL2   Enabled
        Gfast   Disabled
VDSL2 profiles:
        8a      Enabled
        8b      Enabled
        8c      Enabled
        8d      Enabled
        12a     Enabled
        12b     Enabled
        17a     Enabled
        30a     Enabled
        35b     Enabled
        US0     Enabled

negan07, 2018-04-24:
link expired

Found a mismatched library in the full driverset
solib: **libxdslctl.so** not included but different from original one

bumped xdsl-full-new package including  libxdslctl.so from dumped binary

0d1n0, 2018-04-24:
which link expired?

negan07, 2018-04-25:
still trying to extract .rbi from agev
asus extract good
agev not good (no extract): how did u extract .rbi ?

asus AC88U is arm 3.4.11rt19 too so should be also good for tests
think Dlink 5592 should be similar too...

0d1n0, 2018-04-26:
rbi is an encrypted firmware format you need private key to decrypt it.
dlink seem better than ac88u because for sure have 35b support on 42 version of firmware and seem more better xdslctl command. i'm not sure that ac88 have 35b support on 42 version of firmware.

negan07, 2018-04-26:
did all the tries possible with above driversets: nothing good

the only one working is the 042u.d26o 

but there's a strange library mismatch with its libxdslctl.so

using the library from the same 1.0.4 xdslctl gives a ctype error: drivers not so monolithic as they seem

replacing with original library, `xdslctl.bin configure --profile BrcmPriv1` starts looping in retraining mode

replaced with asus library `xdslctl.bin configure --profile BrcmPriv1` gives a correct link but must be tested with real 35b active

driver is up named `xdsl-driver-agtef_042u.026o`

0d1n0, 2018-04-26:
We need Coliuto to test  042u.026o.

Coliuto, 2018-05-04:
I've just tried with the new package: 
https://i.imgur.com/ALo5Hl9.png
https://i.imgur.com/GrBdI2Y.png

Unfortunately it does not work. Auto-mode only 17a, using the command "xdslctl.bin configure --profile BrcmPriv1" no link:

admin@D7000:$ dmesg
ontroller
ohci-platform ohci-platform.0: new USB bus registered, assigned bus number 4
ohci-platform ohci-platform.0: irq 104, io mem 0x8000c400
hub 4-0:1.0: USB hub found
hub 4-0:1.0: 2 ports detected
Ethernet Auto Power Down and Sleep: Enabled
Energy Efficient Ethernet: Enabled
eth3 (Ext switch port: 2) (Logical Port: 10) Link UP 1000 mbps full duplex
kcg 333 :GPL NetUSB up!
kc   90 : run_telnetDBGDServer start
kc  227 : init_DebugD end
INFO187A: NetUSB 1.02.64, 0003040B : Apr 10 2015 15:50:48
INFO187C: 7423: D7000 :D7000
INFO187D:  AUTH ISOC
INFO187E:  MOVE_HEAD emb_lp filterAudio
INFO0715:  0 : tunnelID emusblp 0
INFO0715:  1 : tunnelID emusblp 1
INFO0715:  2 : tunnelID emusblp 2
INFO0715:  3 : tunnelID emusblp 3
INFO0715:  4 : tunnelID emusblp 4
INFO0715:  5 : tunnelID emusblp 5
INFO0715:  6 : tunnelID emusblp 6
INFO0715:  7 : tunnelID emusblp 7
INFO0715:  8 : tunnelID emusblp 8
INFO0715:  9 : tunnelID emusblp 9
INFO0ADC:  --- checkSBusTimeoutThread start ---
usbcore: registered new interface driver KC NetUSB General Driver
INFO18D6: Reset device when disconnect
INFO0145:  init proc : PAGE_SIZE 4096
INFO174F:  infomap bf93ce10
INFO1752:  sleep to wait group1 to wake up
INFO02C5:  use dev Name group1 : fail
INFO02C5:  use dev Name group1 : fail
sc_do_led_ctrl_ioctl:138 set ctrl to 1
INFO02CE:  use dev Name group1
INFO175B:  sleep to wait group1 end.
INFO1774:  UDP_BROAD 7423 tcpPort:20005
INFO02CE:  use dev Name group1
INFO1623: tcpConnector() started...
ADDRCONF(NETDEV_UP): eth1: link is not ready
device eth1 entered promiscuous mode
ADDRCONF(NETDEV_UP): eth2: link is not ready
device eth2 entered promiscuous mode
device eth3 entered promiscuous mode
group1: port 3(eth3) entered forwarding state
group1: port 3(eth3) entered forwarding state
ADDRCONF(NETDEV_UP): eth4: link is not ready
device eth4 entered promiscuous mode
device wl0 entered promiscuous mode
group1: port 5(wl0) entered forwarding state
group1: port 5(wl0) entered forwarding state
device wl1 entered promiscuous mode
group1: port 6(wl1) entered forwarding state
group1: port 6(wl1) entered forwarding state
group1: port 6(wl1) entered disabled state
group1: port 5(wl0) entered disabled state
group1: port 3(eth3) entered disabled state
group1: port 6(wl1) entered forwarding state
group1: port 6(wl1) entered forwarding state
group1: port 5(wl0) entered forwarding state
group1: port 5(wl0) entered forwarding state
group1: port 3(eth3) entered forwarding state
group1: port 3(eth3) entered forwarding state
[... 234 lines cut ...]
ADSL led state change to <2>
message received before monitor task is initialized kerSysSendtoMonitorTask
message received before monitor task is initialized kerSysSendtoMonitorTask
Line 0: xDSL link down
ADSL led state change to <2>
message received before monitor task is initialized kerSysSendtoMonitorTask
Line 0: xDSL G.994 training
ADSL led state change to <2>
message received before monitor task is initialized kerSysSendtoMonitorTask
message received before monitor task is initialized kerSysSendtoMonitorTask
Line 0: xDSL link down
ADSL led state change to <2>
message received before monitor task is initialized kerSysSendtoMonitorTask
Line 0: xDSL G.994 training
ADSL led state change to <2>
message received before monitor task is initialized kerSysSendtoMonitorTask
message received before monitor task is initialized kerSysSendtoMonitorTask
Line 0: xDSL link down
ADSL led state change to <2>
message received before monitor task is initialized kerSysSendtoMonitorTask
Line 0: xDSL G.994 training
ADSL led state change to <2>
message received before monitor task is initialized kerSysSendtoMonitorTask
message received before monitor task is initialized kerSysSendtoMonitorTask
admin@D7000:$

Ansuel, 2018-05-08:
@negan07 i mean firmware of the modem... we can't decrypt the adsl_phy blob....

negan07, 2018-05-08:
i mean the source building tool of the firmware .rbi
the tool generally includes a closed source part broadcom/private where driver are simply copied at make install

the driver itself looks not to be an issue, but xdslctl and modules maybe

Ansuel, 2018-05-08:
we have part of the source but i think we don't have xdsl source... 

we have something to decrypt .rbi file (of some modem) and then with binwalk you can extract the firmware and have a squashfs 

https://repository.ilpuntotecnicoeadsl.com/

i think you need to fist sign in to the forum to access it

negan07, 2018-05-08:
xdsl drivers can't be compiled just copied from a device to another hoping to find a good compatibility

considering that devices shares about the same kernel, arch, chipset, and maybe board, the aim of this issue is trying to port 35b driverset to this device if possible

Ansuel, 2018-05-08:
anyway if you have more driver than i have pls share them...

also what bin contain the profile? ko modules? xdslctl?

negan07, 2018-05-08:
original driver is 042n.d26b
updated is 042u.d26o

haven't seen any 042 driver version more than you got: they should not differs so much each other

just a note:
https://repository.ilpuntotecnicoeadsl.com/index.php?dir=Ansuel/AGTEF/adsl_driver/
the physical driver `adsl_phy.bin` is `A2pvbH042j2`
the `adsldd.ko` is `d26r`
the dir contains the first not the second

I think xdslctl is the main responsible of profile training but with the support of at least adsldd.ko
xdslctl is shared with libxdslctl.so and or some bcm libs

[106 more replies]

closed 2019-04-26 by negan07