negan07/ancistrus #18: Compatibility with OpenWRT packages

ComputaBloke, 2018-01-14:
As determined for #6 and #10, some base dependencies are required if we are to use any OpenWRT `opkg` package binaries on the anc/D7000.

Tried this:
To /etc/opkg.conf I added the OpenWRT sources:
```
arch arm_cortex-a9 4
src/gz openwrt_arm_base http://ba.mirror.garr.it/mirrors/openwrt/releases/17.01.4/packages/arm_cortex-a9/base
src/gz openwrt_arm_packages http://ba.mirror.garr.it/mirrors/openwrt/releases/17.01.4/packages/arm_cortex-a9/packages
```
And ran `opkg update`
Then `opkg install usb-modeswitch`

Produced errors:
```
 * calculate_dependencies_for: Cannot satisfy the following dependencies for usb-modeswitch:
 *      libc *  libc *  libc *  libc *  libc *  libpthread *    librt *
 * opkg_solver_install: Cannot install package usb-modeswitch.
```

Additionally, `--force-depends` will ignore this and install the package, but the binaries don't run due to incompatible libc.

Linux kernel for D7000 is: Linux version 3.4.11-rt19 (root@Debian) (gcc version 4.6.2 (Buildroot 2011.11-g00639bc-dirty) ) #2 SMP PREEMPT Wed Jan 3 09:48:00 CET 2018. So 7 years old! Apparently this is common for embedded-linux devices due to using Broadcom SDK .

OpenWRT 15.05 still used uclibc, but then moved to musl-libc, so definitely the latest 17.x streams won't be a compatible source, but there is still a chance to find compatible binary packages from older generations (will still need `--force-depends`).

Cracking this would be very useful since OpenWRT offers so much. Alternative being to cross-compile many packages from source to suit D7000 will not be much fun unless that can be automated in some fashion.

ComputaBloke, 2018-01-14:
OpenWRT Attitude Adjustment (*AA*) used Kernel 3.3 [final release in April 2013](https://forum.openwrt.org/viewtopic.php?pid=199599), beyond this (*BB*) they switched to Kernel 3.10.

The bcm63138 chipset currently has no compatible OpenWRT build, but is listed on [bcm63xx SoC variant list](https://wiki.openwrt.org/doc/hardware/soc/soc.broadcom.bcm63xx), with it's hardware specs.

So still not sure if this will work, but it looks like best compatible source for next attempt will be: [https://archive.openwrt.org/attitude_adjustment/12.09/brcm63xx](https://archive.openwrt.org/attitude_adjustment/12.09/brcm63xx/generic/)

negan07, 2018-01-14:
the building won't take so much advantage as it doesn't commit arm processors compatibility and moreover ther will be problems with broadcom closed drivers

so it should be compiled with this 3.4.11 kernel and maybe this 0.9.32 uclibc and then the rootfs with all the adaptations needed for arm and moreover the tuning of packages, many older than those furnished by netgear source package...

anyway they need to be recompiled and maybe it's the quickest and most reliable way:
the packages may be a useful guide for configuration tunings and code patches as i used to do for the package created here

if the drivers nut/dongle has been compiled for many architectures on openwrt the sources of them should be available (downloaded and patched as usual)

negan07, 2018-01-14:
so this is the only tryout possible:
https://archive.openwrt.org/attitude_adjustment/12.09/brcm63xx/generic/packages/
there are very old versions but some standalone ones:
https://archive.openwrt.org/attitude_adjustment/12.09/brcm63xx/generic/packages/openssh-keygen_6.1p1-1_brcm63xx.ipk
this has deps libc (uclibc) openssl (good) zlib (not needed for keygen)
so forcing undep install:

`/bin/ssh-keygen: line 6: syntax error: unexpected ")"`

as supposed, not working

ComputaBloke, 2018-01-15:
Same: `/sbin/usb_modeswitch: line 1: syntax error: unexpected "("` 😢 
Would have been nice to get 3500+ packages for zero effort, but this direct method won't work.

OpenWRT package sources can be found here: https://wiki.openwrt.org/doc/packages#package_sources
Will need the OpenWRT build root: https://wiki.openwrt.org/doc/howto/build etc etc.

Perhaps this can be parked until we can work out easiest way to get package sources patched and compiled for D7000, or you can kill this issue if you like. Ultimately this is an Ancistrus project, not an OpenWRT one!

For now I'll stick to setting up my Anc build VM.

negan07, 2018-01-16:
some sources are available, like script and non binary tools

the configs & patches have the price of the ticket: compiling is simple if there are no issues

closed 2018-01-18 by negan07