negan07/ancistrus #22: snmp

adrianhewitt, 2018-02-18:
any chance we could get an snmp daemon on this firmware?

negan07, 2018-02-19:
yes possible
time to time: there's some work to do

adrianhewitt, 2018-02-20:
yay! very much looking forward to the future of this project

negan07, 2018-02-20:
there are many smnp daemons all around: any preference ?

adrianhewitt, 2018-02-21:
no preference here. i had a look around earlier and noticed some have 32bit counters (mini snmpd) and others 64bit. i don't really know if the d7000 supports 64bit counters or if that really is necessary... my use case will be basically reading data transfer readouts from the router and write them to an influxdb database every 30secs or so. pretty basic. unsure if anyone wants it to be much more complex?

negan07, 2018-04-17:
mini-snmpd package is up with gui panel
looks tiny but give a try and tell

adrianhewitt, 2018-04-18:
hrmm cant seem to get it completely working, but likely just missing something. i installed/updated mini-snmpd and dependencies as the wiki said, and manually setting minisnmpd_enable and running it from telnet gets it up and running... only issue i find is it's not returning network port usage (does return mem/disk/cpu data though). also if i go to look into the webui to see if theres anything i can switch there i find loading the Ancistrus panel collects a whole heap of 404 not founds while trying to access http://192.168.0.1/opkgstatus / http://192.168.0.1/opkglistupgr. oh, and can't find mention of the snmp daemon there :) theres a list of things under Core Setup, but Management / Networking / Storage / General are empty. Maybe there is something i need to update?

adrianhewitt, 2018-04-18:
(more background: pretty sure router is running the latest monolithic firmware, haven't done much to it since installing that)

negan07, 2018-04-18:
which client did u use for snmp ?

the web gui needs a complete package upgrade before: run 
`opkg update && opkg upgrade`

the minisnmpd doesn't have config  but all options are at invocation level
the void menu are filled dynamically when installing some are void because no packages

adrianhewitt, 2018-04-18:
i was mainly testing via snmpwalk, but also tried PeakHour 3 to double check and that couldn't find any interfaces. heres the output i get from snmpwalk:

```
$ snmpwalk -v2c -c public 192.168.0.1 .
SNMPv2-MIB::sysDescr.0 = STRING: router
SNMPv2-MIB::sysObjectID.0 = OID: SNMPv2-SMI::enterprises
DISMAN-EVENT-MIB::sysUpTimeInstance = Timeticks: (7200167) 20:00:01.67
SNMPv2-MIB::sysContact.0 = STRING: -i
SNMPv2-MIB::sysName.0 = STRING: D7000
SNMPv2-MIB::sysLocation.0 = STRING: here
HOST-RESOURCES-MIB::hrSystemUptime.0 = Timeticks: (7205721) 20:00:57.21
UCD-SNMP-MIB::memTotalReal.0 = INTEGER: 227428 kB
UCD-SNMP-MIB::memAvailReal.0 = INTEGER: 87880 kB
UCD-SNMP-MIB::memShared.0 = INTEGER: 232673 kB
UCD-SNMP-MIB::memBuffer.0 = INTEGER: 0 kB
UCD-SNMP-MIB::memCached.0 = INTEGER: 26984 kB
UCD-SNMP-MIB::dskIndex.1 = INTEGER: 1
UCD-SNMP-MIB::dskPath.1 = STRING: /mnt/shares
UCD-SNMP-MIB::dskTotal.1 = INTEGER: 0
UCD-SNMP-MIB::dskAvail.1 = INTEGER: 0
UCD-SNMP-MIB::dskUsed.1 = INTEGER: 0
UCD-SNMP-MIB::dskPercent.1 = INTEGER: 0
UCD-SNMP-MIB::dskPercentNode.1 = INTEGER: 0
UCD-SNMP-MIB::laIndex.1 = INTEGER: 1
UCD-SNMP-MIB::laIndex.2 = INTEGER: 2
UCD-SNMP-MIB::laIndex.3 = INTEGER: 3
UCD-SNMP-MIB::laNames.1 = STRING: Load-1
UCD-SNMP-MIB::laNames.2 = STRING: Load-5
UCD-SNMP-MIB::laNames.3 = STRING: Load-15
UCD-SNMP-MIB::laLoad.1 = STRING: 0.20
UCD-SNMP-MIB::laLoad.2 = STRING: 0.09
UCD-SNMP-MIB::laLoad.3 = STRING: 0.07
UCD-SNMP-MIB::laConfig.1 = STRING: 1
UCD-SNMP-MIB::laConfig.2 = STRING: 5
UCD-SNMP-MIB::laConfig.3 = STRING: 15
UCD-SNMP-MIB::laLoadInt.1 = INTEGER: 20
UCD-SNMP-MIB::laLoadInt.2 = INTEGER: 9
UCD-SNMP-MIB::laLoadInt.3 = INTEGER: 7
UCD-SNMP-MIB::ssCpuRawUser.0 = Counter32: 20132
UCD-SNMP-MIB::ssCpuRawNice.0 = Counter32: 0
UCD-SNMP-MIB::ssCpuRawSystem.0 = Counter32: 232695
UCD-SNMP-MIB::ssCpuRawIdle.0 = Counter32: 4292342470
UCD-SNMP-MIB::ssRawInterrupts.0 = Counter32: 0
UCD-SNMP-MIB::ssRawContexts.0 = Counter32: 3204311252
UCD-SNMP-MIB::ssRawContexts.0 = No more variables left in this MIB View (It is past the end of the MIB tree)
```

pretty sure when i was playing previously i tried `opkg update && opkg upgrade` (also just ran it again to be sure)

aaaand curiously when i run `opkg install --force-reinstall opkg`, it returns:

```
Collected errors:                                                    
 * pkg_run_script: Internal error: opkg has a NULL tmp_unpack_dir.   
 * prerm_upgrade_old_pkg: prerm script for package "opkg" failed     
 * opkg_solver_install: Cannot install package opkg.    
```             

sorry to be a pain :) and i appreciate your work!

adrianhewitt, 2018-04-19:
(also if it's going to be easier just to wait for the next monolithic build, i'm happy to wait)

negan07, 2018-04-19:
no need:
simply
remove opkg

`opkg remove opkg`

`rm -r /usr/lib/opkg/*`

and start again with startup.sh:
https://negan07.github.io/ancistrus/

adrianhewitt, 2018-04-19:
had an issue with one of the removes and reinstall, so i restarted from the latest monolithic then ran startup.sh and the web interface is fine now :) installed minisnmpd again but seems to be reporting the same data, details about everything but the network interfaces. i had a quick look to see if there was a minismtpd config file anywhere to attempt to manually load them but i wasn't find anything.

negan07, 2018-04-19:
disable mini-snmpd and run the cmd manually with parameters

mini_snmpd -c ${minisnmpd_community} -D ${minisnmpd_description} -V ${minisnmpd_vendor} -L ${minisnmpd_location} -C ${minisnmpd_contact} -i ${minisnmpd_interfaces} -d ${minisnmpd_disks}

here you can include the interfaces on the -i option
included are ppp1 & group1 (wan & lan) supposed to be enough: maybe you need others present on
`ifconfig -a`

adrianhewitt, 2018-04-22:
Maybe it's group1 that caused the issue. I have this router hooked up to fibre through eth3 without DSL, and when i run `ip link show` it has one of the group1 listings status as unknown (having a hard time reading output, using a bad third party telnet client since apple removed the in-build telnet client ugh). once i disabled and ran `mini_snmpd -i eth0,eth1,eth2,eth3` (yeah just realised i forgot the wifi interfaces) everything came up! :)

negan07, 2018-04-23:
no you are probably right: the group1 is a setting variable `lan_if` while ppp1 is `wan_ifname`
maybe your particular kind of cable (ethernet) connection has some different behaviour

will fix the rc adding all the available network interfaces

does ppp1 (or the one you have as nvram `wan_ifname`) have same behaviour?

adrianhewitt, 2018-04-24:
bare with me, i'm not super familiar with linux routing software, and only passable with linux ;) plus i think i'm not answering your question, but maybe you will find something in all this information

wan_ifname details:

```
admin@D7000:$ nvram get wan_ifname
wan_ifname=eth0.1
```

and for curiosity's sake, here is my complete `ip link show` readout

```
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 16436 qdisc noqueue state UNKNOWN mode DEFAULT group default 
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: ifb0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 32
    link/ether 66:4e:b4:be:90:a0 brd ff:ff:ff:ff:ff:ff
3: ifb1: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 32
    link/ether 8e:d0:65:2b:07:8b brd ff:ff:ff:ff:ff:ff
4: sit0: <NOARP> mtu 1480 qdisc noop state DOWN mode DEFAULT group default 
    link/sit 0.0.0.0 brd 0.0.0.0
5: ip6tnl0: <NOARP> mtu 1452 qdisc noop state DOWN mode DEFAULT group default 
    link/tunnel6 :: brd ::
6: dsl0: <> mtu 0 qdisc noop state DOWN mode DEFAULT group default 
    link/[29] 
7: bcmsw: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 2048 qdisc noop state UNKNOWN mode DEFAULT group default qlen 1000
    link/ether 08:02:8e:e6:b2:4e brd ff:ff:ff:ff:ff:ff
8: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 2048 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether 08:02:8e:e6:b2:52 brd ff:ff:ff:ff:ff:ff
9: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 2048 qdisc pfifo_fast master group1 state UP mode DEFAULT group default qlen 1000
    link/ether 08:02:8e:e6:b2:4e brd ff:ff:ff:ff:ff:ff
10: eth2: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast master group1 state DOWN mode DEFAULT group default qlen 1000
    link/ether 08:02:8e:e6:b2:4e brd ff:ff:ff:ff:ff:ff
11: eth3: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast master group1 state DOWN mode DEFAULT group default qlen 1000
    link/ether 08:02:8e:e6:b2:4e brd ff:ff:ff:ff:ff:ff
12: eth4: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast master group1 state DOWN mode DEFAULT group default qlen 1000
    link/ether 08:02:8e:e6:b2:4e brd ff:ff:ff:ff:ff:ff
13: wl0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master group1 state UNKNOWN mode DEFAULT group default qlen 1000
    link/ether 08:02:8e:e6:b2:4e brd ff:ff:ff:ff:ff:ff
14: wl1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master group1 state UNKNOWN mode DEFAULT group default qlen 1000
    link/ether 08:02:8e:e6:b2:50 brd ff:ff:ff:ff:ff:ff
15: group1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default 
    link/ether 08:02:8e:e6:b2:4e brd ff:ff:ff:ff:ff:ff
16: group2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UNKNOWN mode DEFAULT group default 
    link/ether 76:e8:cf:31:66:01 brd ff:ff:ff:ff:ff:ff
17: wl0.1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master group1 state UNKNOWN mode DEFAULT group default qlen 1000
    link/ether 62:02:8e:e6:b2:4f brd ff:ff:ff:ff:ff:ff
18: wl0.2: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 08:02:8e:e6:b2:4e brd ff:ff:ff:ff:ff:ff
19: wl0.3: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 08:02:8e:e6:b2:4e brd ff:ff:ff:ff:ff:ff
20: wl1.1: <BROADCAST,MULTICAST> mtu 1500 qdisc pfifo_fast master group1 state DOWN mode DEFAULT group default qlen 1000
    link/ether 62:02:8e:e6:b2:51 brd ff:ff:ff:ff:ff:ff
21: wl1.2: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 08:02:8e:e6:b2:50 brd ff:ff:ff:ff:ff:ff
22: wl1.3: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
    link/ether 08:02:8e:e6:b2:50 brd ff:ff:ff:ff:ff:ff
23: eth0.1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default 
    link/ether 08:02:8e:e6:b2:4f brd ff:ff:ff:ff:ff:ff
```

aaaaand here is a fresh snmpwalk with eth0.1 enabled

```
SNMPv2-MIB::sysDescr.0 = STRING: 
SNMPv2-MIB::sysObjectID.0 = OID: SNMPv2-SMI::enterprises
DISMAN-EVENT-MIB::sysUpTimeInstance = Timeticks: (534) 0:00:05.34
SNMPv2-MIB::sysContact.0 = STRING: 
SNMPv2-MIB::sysName.0 = STRING: D7000
SNMPv2-MIB::sysLocation.0 = STRING: 
IF-MIB::ifNumber.0 = INTEGER: 8
[... 149 lines cut ...]
UCD-SNMP-MIB::laIndex.1 = INTEGER: 1
UCD-SNMP-MIB::laIndex.2 = INTEGER: 2
UCD-SNMP-MIB::laIndex.3 = INTEGER: 3
UCD-SNMP-MIB::laNames.1 = STRING: Load-1
UCD-SNMP-MIB::laNames.2 = STRING: Load-5
UCD-SNMP-MIB::laNames.3 = STRING: Load-15
UCD-SNMP-MIB::laLoad.1 = STRING: 0.07
UCD-SNMP-MIB::laLoad.2 = STRING: 0.05
UCD-SNMP-MIB::laLoad.3 = STRING: 0.05
UCD-SNMP-MIB::laConfig.1 = STRING: 1
UCD-SNMP-MIB::laConfig.2 = STRING: 5
UCD-SNMP-MIB::laConfig.3 = STRING: 15
UCD-SNMP-MIB::laLoadInt.1 = INTEGER: 7
UCD-SNMP-MIB::laLoadInt.2 = INTEGER: 5
UCD-SNMP-MIB::laLoadInt.3 = INTEGER: 5
UCD-SNMP-MIB::ssCpuRawUser.0 = Counter32: 158847
UCD-SNMP-MIB::ssCpuRawNice.0 = Counter32: 65
UCD-SNMP-MIB::ssCpuRawSystem.0 = Counter32: 1587083
UCD-SNMP-MIB::ssCpuRawIdle.0 = Counter32: 83814132
UCD-SNMP-MIB::ssRawInterrupts.0 = Counter32: 4052
UCD-SNMP-MIB::ssRawContexts.0 = Counter32: 4
UCD-SNMP-MIB::ssRawContexts.0 = No more variables left in this MIB View (It is past the end of the MIB tree)
```

background on the internet connection which likely doesn't matter, it's on the ISP iiNet via the NBN in Australia. been a while, but pretty sure the modem is automatically configured through the ISP on initial setup.

negan07, 2018-04-24:
bumped mini-snmpd with eth0,eth1,eth2,eth3,eth4,ip6tnl0,${wan_ifname},ptm0,wl0,wl1