negan07/ancistrus #23: xdslctl profile

negan07, 2018-02-25:
Looking at the dsl parameters tweaking.

Should be useful to have boundary conditions for both **VDSL(PTM)/ADSL(ATM)** connection modes.

The console output of the command:

`xdslctl profile --show`

**VDSL(PTM)** got.

Still needed **ADSL(ATM)**

Someone still under **ADSL(ATM)** can please write the result of the command above.

BoostOver, 2018-02-25:
FW 1.64

Modulations:
        G.Dmt   Enabled
        G.lite  Enabled
        T1.413  Enabled
        ADSL2   Enabled
        AnnexL  Enabled
        ADSL2+  Enabled
        AnnexM  Enabled
        VDSL2   Enabled
VDSL2 profiles:
        8a      Enabled
        8b      Enabled
        8c      Enabled
        8d      Enabled
        12a     Enabled
        12b     Enabled
        17a     Enabled
        30a     Enabled
        US0     Enabled
Phone line pair:
        Inner pair
Capability:
        bitswap         On
        sra             On
        trellis         On
        sesdrop         Off
        CoMinMgn        Off
        24k             On
        phyReXmt(Us/Ds) Off/On
        Ginp(Us/Ds)     On/On
        TpsTc           AvPvAa
        monitorTone:    On
        dynamicD:       On
        dynamicF:       Off
        SOS:            On
        Training Margin(Q4 in dB):      -1(DEFAULT)

Maiux92, 2018-02-25:
I'm on FW V1.0.0.38_1.0.1_lanip_reserve with ancistrus installed.

Modulations:
        G.Dmt   Enabled
        G.lite  Enabled
        T1.413  Enabled
        ADSL2   Enabled
        AnnexL  Enabled
        ADSL2+  Enabled
        AnnexM  Enabled
        VDSL2   Enabled
VDSL2 profiles:
        8a      Enabled
        8b      Enabled
        8c      Enabled
        8d      Enabled
        12a     Enabled
        12b     Enabled
        17a     Enabled
        30a     Enabled
        US0     Enabled
Phone line pair:
        Inner pair
Capability:
        bitswap         On
        sra             On
        trellis         On
        sesdrop         Off
        CoMinMgn        Off
        24k             On
        phyReXmt(Us/Ds) Off/On
        Ginp(Us/Ds)     On/On
        TpsTc           AvPvAa
        monitorTone:    On
        dynamicD:       On
        dynamicF:       Off
        SOS:            On
        Training Margin(Q4 in dB):      -1(DEFAULT)

negan07, 2018-02-25:
**FW V1 .0.0.38_1.0.1_lanip_reserve**
which fw is it ?

j27h, 2018-02-25:
Here is mine (with your build 211 based on FW V1.0.1.54):

Modulations:
        G.Dmt   Enabled
        G.lite  Enabled
        T1.413  Enabled
        ADSL2   Enabled
        AnnexL  Enabled
        ADSL2+  Enabled
        AnnexM  Disabled
        VDSL2   Enabled

Phone line pair:
        Inner pair

Capability:
        bitswap         On
        sra             Off
        trellis         On
        sesdrop         Off
        CoMinMgn        Off
        24k             On
        phyReXmt(Us/Ds) Off/On
        Ginp(Us/Ds)     On/On
        TpsTc           AvPvAa
        monitorTone:    On
        dynamicD:       On
        dynamicF:       Off
        SOS:            On
        Training Margin(Q4 in dB):      65480

negan07, 2018-02-25:
it seems drivers makes all defaults not dslam: I had this doubt now fixed

Unfortunately wifi drivers are closed so no way to substitute them but you can create a package and install them into newer releases.

ekelon, 2018-02-26:
FW 1.54
Modulations:
        G.Dmt   Enabled
        G.lite  Enabled
        T1.413  Enabled
        ADSL2   Enabled
        AnnexL  Enabled
        ADSL2+  Enabled
        AnnexM  Disabled
        VDSL2   Enabled
Phone line pair:
        Inner pair
Capability:
        bitswap         On
        sra             Off
        trellis         On
        sesdrop         Off
        CoMinMgn        Off
        24k             On
        phyReXmt(Us/Ds) Off/On
        Ginp(Us/Ds)     On/On
        TpsTc           AvPvAa
        monitorTone:    On
        dynamicD:       On
        dynamicF:       Off
        SOS:            On
        Training Margin(Q4 in dB):      25

closed 2018-02-28 by negan07