negan07/ancistrus #24: Build 265 is wrong

BoostOver, 2018-03-01:
Any attempt to update a package with OPKG corrupts the firmware.
Example, try updating the xdsl driver .....

negan07, 2018-03-02:
yes...

ther is a problem with opkg syncing packages

for some reason it removes xdslctl making connection impossible at next reboot

blocking the release until investigations

BoostOver, 2018-03-02:
And gui stuck/crash

negan07, 2018-03-02:
package installed into monolithic cannot be sync with rep because coming from an offline installation: so the status file is different and it cause package upgrading the first time installing a package like xdsl driver

there was an error into core postinstall creating a symlink overwriting xdslctl.bin

now core package bumped and problem solved

the gui stuck noticed was certainly related to xdslctl missing

closed 2018-03-02 by BoostOver