negan07/ancistrus #25: Dsl-Tweaking (Save or not Save config?)

BoostOver, 2018-03-02:
In according with https://github.com/negan07/ancistrus/wiki/Dsl-Tweaking:

admin@D7000:$ anc nvram get bitswap sra trellis sesdrop cominmgn sos dynamicd dy
namicf i24k monitortone phyrexmt ginp
bitswap=on
sra=off
trellis=on
sesdrop=on            <<<<<<<<<<<
cominmgn=off
sos=on
dynamicd=off  <<<<<<<<<<<
dynamicf=off
i24k=on
monitortone=on
phyrexmt=3 <<<<<<<<<<<
ginp=0 <<<<<<<<<<<

But:
admin@D7000:$ xdslctl profile --show

        bitswap         On
        sra             Off
        trellis         On
        sesdrop         Off        <<<<<<<<<<<
        CoMinMgn        Off
        24k             On
        phyReXmt(Us/Ds) Off/On      <<<<<<<<<<
        Ginp(Us/Ds)     On/On       <<<<<<<<<<<<
        TpsTc           AvPvAa
        monitorTone:    On
        dynamicD:       On             <<<<<<<<<<<<
        dynamicF:       Off
        SOS:            On
        Training Margin(Q4 in dB):      -1(DEFAULT)

negan07, 2018-03-02:
you didn't enable [tweaking flag](https://github.com/negan07/ancistrus/wiki/Dsl-Tweaking#additional-tweak-settings) and/or you didn't apply setting properly with `dslreconfig.sh`

the [hidden webpage](http://192.168.0.1/DSL_PRO_Config.html) set main settings only but it doesn't retrain

BoostOver, 2018-03-02:
The hidden page is empty .... is it right?
Repeated the procedure but there are no changes.

negan07, 2018-03-02:
fixed [hidden webpage](http://192.168.0.1/DSL_PRO_config.html)

for some reason config was written with uppercase in the wiki page...

now update & upgrade all the packages bumping to 0.9.3 and do your tests again

keep in mind [this]https://github.com/negan07/ancistrus/wiki/Dsl-Tweaking#additional-tweak-settingsl) part of the wiki

BoostOver, 2018-03-02:
The hidden page now work but...........I can not or will not work.
Example:
Show me how to set up PhyReXmt.
Thank you.

negan07, 2018-03-02:
Found the problem:
settings are always applied in `driver default mode`
What mislead is the `sra `setting: while testing I was sra=off and the driver configure too...

The problem resides probably in the parameters passed to the driver, that will be fixed in the next version of `ancistrus-core`: connection reup is useless too so maybe reconfig script will be removed as configure is enough.

So now it's known that:
* Broadcom driver calls 2 different things with the same name: `profile`: `profile` is the union of the `configure` settings (mod snr sra sesdrop....) but also the VDSL `profile` connection so it's better to call `profile settings` or `profile configure` the above values and `VDSL profile` the VDSL connection mode
* Driver default `profile settings` is different from netgear's default `profile settings`

* XDSLCTL PROFILE --SHOW <NETGEAR'S DEFAULT>
```
Modulations:
	G.Dmt	Enabled
	G.lite	Enabled
	T1.413	Enabled
	ADSL2	Enabled
	AnnexL	Enabled
	ADSL2+	Enabled
	AnnexM	Enabled
	VDSL2	Enabled
VDSL2 profiles:
	8a	Enabled
	8b	Enabled
	8c	Enabled
	8d	Enabled
	12a	Enabled
	12b	Enabled
	17a	Enabled
	30a	Enabled
	US0	Enabled
Phone line pair:
	Inner pair
Capability:
	bitswap		On
	sra		On
	trellis		On
	sesdrop		Off
	CoMinMgn	Off
	24k		On
	phyReXmt(Us/Ds)	Off/On (1)
	Ginp(Us/Ds)	On/On (3)
	TpsTc		AvPvAa (14)
	monitorTone:	On
	dynamicD:	On
	dynamicF:	Off
	SOS:		On

```

* XDSLCTL PROFILE --SHOW <DRIVER'S DEFAULT> ( 'xdslctl --configure' )
```
Modulations:
	G.Dmt	Enabled
	G.lite	Enabled
	T1.413	Enabled
	ADSL2	Enabled
	AnnexL	Enabled
	ADSL2+	Enabled
	AnnexM	Disabled
	VDSL2	Enabled
VDSL2 profiles:
	8a	Enabled
	8b	Enabled
	8c	Enabled
	8d	Enabled
	12a	Enabled
	12b	Enabled
	17a	Enabled
	30a	Enabled
	US0	Enabled
Phone line pair:
	Inner pair
Capability:
	bitswap		On
	sra		Off
	trellis		On
	sesdrop		Off
	CoMinMgn	Off
	24k		On
	phyReXmt(Us/Ds)	Off/On
	Ginp(Us/Ds)	On/On
	TpsTc		AvPvAa
	monitorTone:	On
	dynamicD:	On
	dynamicF:	Off
	SOS:		On
	Training Margin(Q4 in dB):	-1(DEFAULT)
```
Apart AnnexM, the difference is just `sra` ...

For this reason now It' preferable to change the tweak behaviour and also `snr` option hide setting.

negan07, 2018-03-04:
Core is bumped to 0.9.4 and now exec() should be run properly and settings applied.

Settings revised on dsl tweak so better to unset previous and store the new settings following the [updated wiki page](https://github.com/negan07/ancistrus/wiki/Dsl-Tweaking)

BoostOver, 2018-03-05:
Many thanks.

closed 2018-03-05 by BoostOver