negan07/ancistrus #26: snr not change

ekelon, 2018-03-05:
hello the commands for editing snr via telnet do not work typing:

nvram set anc_snrtweak_enable = 1
nvram set snr = xx (example 25)
but any result and router restart with old snr;
with command xdslctl configure --snr = xx (example 25)
same result, router restart but snr does not change.
Any idea? Thank

negan07, 2018-03-05:
usage is:
```
nvram set anc_dslconfprofile=own
nvram set anc_snrtweak_enable=1
nvram set snr=25
xdslctl configure
```
or:
```
anc nvram set anc_dslconfprofile own anc_snrtweak_enable 1 snr 25
nvram commit
xdslctl configure
```
forgot to set `anc_dslconfprofile` setting to `own`

ekelon, 2018-03-06:
Thank so much, I solved the problem. With your firmware, the router is an arrow :-)

closed 2018-03-06 by ekelon