negan07/ancistrus #28: Error log on update on fresh monolithic bug build #272

MrZeD81, 2018-04-28:


negan07, 2018-04-29:
same as:
https://github.com/negan07/ancistrus/issues/27

package has been mostly updated after #272 monolithic
in particular lock removal has been cleared while old packages don't

In general, when troubling with opkg, do this step:

Remove all installation references with:
`rm -rf /usr/lib/opkg/* /tmp/opkg/`
`sync`
Now no packages result installed but no package files have been removed. Proceed reinstalling:
`opkg update`
`opkg install opkg`
`opkg install ancistrus-core`
`opkg install zlib`
`opkg install openssh`
...and so on

BoostOver, 2018-04-29:
At the end with all command "install or upgrade" i get this error:
Collected errors:
 * pkg_get_installed_files: Failed to open ///usr/lib/opkg/info/zlib.list: No such file or directory.

negan07, 2018-04-29:
do the suggested above

BoostOver, 2018-04-30:
Yes.
Now there is this error:
Downloading https://raw.githubusercontent.com/negan07/ancistrus/gh-pages/ancistrus-arm-D7000/openvpn_2.4.5_armD7000.ipk.killall: openvpn: no process killed
Mon Apr 30 07:50:27 2018 ERROR: Cannot open TUN/TAP dev /dev/net/tun: No such device (errno=19)
Mon Apr 30 07:50:27 2018 Exiting due to fatal error
rmmod: can't unload 'tun': unknown symbol in module, or unknown parameter

And when try:
admin@D7000:$ opkg update
Downloading https://raw.githubusercontent.com/negan07/ancistrus/gh-pages/ancistrus-arm-D7000/Packages.gz.
Updated source 'ancistrus-arm-D7000'.
admin@D7000:$ opkg upgrade
Collected errors:
 * pkg_get_installed_files: Failed to open ///usr/lib/opkg/info/zlib.list: No such file or directory.

I'm on your last build 272

Thanks

negan07, 2018-04-30:
if `/usr/lib/opkg/*` files and `/tmp/opkg/` dir all removed (status file too) no installation references remain

rerun the startup.sh

```
rm /usr/sbin/opkg /usr/lib/opkg/*
rm -rf /tmp/opkg
cd /etc
curl -k -O https://raw.githubusercontent.com/negan07/ancistrus/master/scripts/startup.sh
chmod 755 startup.sh
./startup.sh
```

BoostOver, 2018-05-01:
Thanks, now is fine.

negan07, 2018-05-01:
added opkg troubleshooting solutions to wiki
https://github.com/negan07/ancistrus/wiki/opkg-package#erase-all-package-reference-infos--status

> I made all negan07 said, and screens are fixed, but unfortunally, after a physical reboot, my d7000 is stuck with power red led on, and doesnt boot ;(
> How can I fix this problem ?

maybe deleted wrong files

if problem persists recovery with tftp
https://community.netgear.com/t5/DSL-Modems-Routers/D7000-R7000-bricked-after-SW-reboot/m-p/1138950#M12699

MrZeD81, 2018-05-01:
No one of recovery system works, even the 30-30-30.
No luck at all.

negan07, 2018-05-01:
so not bricked but locked

[nftp.zip](https://github.com/negan07/ancistrus/files/1965380/nftp.zip)

this tool is generic for netgear

power down
keep tiny reset rear button pressed with a stick
keep pressed and power on
maintain pressed until top left leds blinks
release button
run the tool (unix machine or maybe w10 shell)
answer yes to all conditions
when rebooted follow the tftp procedure above: https://github.com/negan07/ancistrus/issues/28#issuecomment-385806687

MrZeD81, 2018-05-02:
No luck again: after nftp command on unix, old firmware was successiful uploaded, but power led still stuck on blinking white led. All firmwares tryed, but 192.168.1.1 was un-pingable, and tft2 cant reach host.

negan07, 2018-05-02:
white blinking led means router stopped at bootloader stage

now tftp has to work
tftp server is at 192.168.1.1
no canonic ping is visible
setup local machine netcard to 192.168.1.10 255.255.255.0 gw 192.168.1.1
upload the monolithic #211 not #272 as it's too big for tftp
`tftp -i 192.168.1.1 put D7000-V1.0.1.54_1.0.1-ancistrus_negan07-build-211.img`

MrZeD81, 2018-05-02:
Seems all works fine now! Tnx for your support and I hope to see soon new monolithic release.
Have a nice day.
P.S.: if you need a little Paypal donation, let me know :)

negan07, 2018-05-02:
> Seems all works fine now!

will add a wiki failsafe recovery page about

> I hope to see soon new monolithic release.

monolithic 's just a snapshot: actually no differences from: original firmware+package installations

> P.S.: if you need a little Paypal donation, let me know :)

http://www.eurogroupforanimals.org/donate
and let me know

closed 2018-05-05 by negan07