negan07/ancistrus #30: MiniSNMP Config

adrianhewitt, 2018-05-14:
Just updated to the new monolithic image from my previous manually updated copy, turned on minisnmp through the webui and it doesn't seem to be reading any network interfaces.

previous conversation: https://github.com/negan07/ancistrus/issues/22

snmpwalk details:

```
snmpwalk -v2c -c public 192.168.0.1 .
SNMPv2-MIB::sysDescr.0 = STRING: router
SNMPv2-MIB::sysObjectID.0 = OID: SNMPv2-SMI::enterprises
DISMAN-EVENT-MIB::sysUpTimeInstance = Timeticks: (7884) 0:01:18.84
SNMPv2-MIB::sysContact.0 = STRING: -i
SNMPv2-MIB::sysName.0 = STRING: D7000
SNMPv2-MIB::sysLocation.0 = STRING: here
HOST-RESOURCES-MIB::hrSystemUptime.0 = Timeticks: (28841) 0:04:48.41
UCD-SNMP-MIB::memTotalReal.0 = INTEGER: 227428 kB
UCD-SNMP-MIB::memAvailReal.0 = INTEGER: 88820 kB
UCD-SNMP-MIB::memShared.0 = INTEGER: 0 kB
UCD-SNMP-MIB::memBuffer.0 = INTEGER: 0 kB
UCD-SNMP-MIB::memCached.0 = INTEGER: 26404 kB
UCD-SNMP-MIB::dskIndex.1 = INTEGER: 1
UCD-SNMP-MIB::dskPath.1 = STRING: /mnt/shares
UCD-SNMP-MIB::dskTotal.1 = INTEGER: 0
UCD-SNMP-MIB::dskAvail.1 = INTEGER: 0
UCD-SNMP-MIB::dskUsed.1 = INTEGER: 0
UCD-SNMP-MIB::dskPercent.1 = INTEGER: 0
UCD-SNMP-MIB::dskPercentNode.1 = INTEGER: 0
UCD-SNMP-MIB::laIndex.1 = INTEGER: 1
UCD-SNMP-MIB::laIndex.2 = INTEGER: 2
UCD-SNMP-MIB::laIndex.3 = INTEGER: 3
UCD-SNMP-MIB::laNames.1 = STRING: Load-1
UCD-SNMP-MIB::laNames.2 = STRING: Load-5
UCD-SNMP-MIB::laNames.3 = STRING: Load-15
UCD-SNMP-MIB::laLoad.1 = STRING: 0.16
UCD-SNMP-MIB::laLoad.2 = STRING: 0.48
UCD-SNMP-MIB::laLoad.3 = STRING: 0.26
UCD-SNMP-MIB::laConfig.1 = STRING: 1
UCD-SNMP-MIB::laConfig.2 = STRING: 5
UCD-SNMP-MIB::laConfig.3 = STRING: 15
UCD-SNMP-MIB::laLoadInt.1 = INTEGER: 16
UCD-SNMP-MIB::laLoadInt.2 = INTEGER: 48
UCD-SNMP-MIB::laLoadInt.3 = INTEGER: 26
UCD-SNMP-MIB::ssCpuRawUser.0 = Counter32: 437
UCD-SNMP-MIB::ssCpuRawNice.0 = Counter32: 22
UCD-SNMP-MIB::ssCpuRawSystem.0 = Counter32: 9847
UCD-SNMP-MIB::ssCpuRawIdle.0 = Counter32: 4294948724
UCD-SNMP-MIB::ssRawInterrupts.0 = Counter32: 0
UCD-SNMP-MIB::ssRawContexts.0 = Counter32: 0
UCD-SNMP-MIB::ssRawContexts.0 = No more variables left in this MIB View (It is past the end of the MIB tree)
```

negan07, 2018-05-14:
fill the snmp contact too in the gui otherwise for some reason it breaks the command at the parameter

didn't enter an email default string due to forbidden @ char used as special char for var begin in the cgi

easy to fix

in the meanwhile fill the contact with some string

adrianhewitt, 2018-05-14:
ohhhhh weird, but glad it's not a huge issue :) working now!