negan07/ancistrus #33: D7000 V2

dniasoff, 2018-06-04:
Was about to install and realised that I have a V2 router. I assume that it's very different.

Any chance you can work on that for a bounty. Please email me? My email address is daniel@redactus.co.uk.

negan07, 2018-06-05:
the v2 has in common the name only

it is a lantiq chipset with a brandized version of chaos chalmer 15.05

dniasoff, 2018-06-06:
Any chance we can coax you into working on the project. You seem to know you're way around netgear firmwares and doing it a 2nd time would be much quicker? V1 seems to be EOL so this would be a natural progession.

negan07, 2018-06-06:
both v1 and v2 have similar limitations

most work could be exported but device is essential to make tests and tunings

dniasoff, 2018-06-06:
No problem, give me your address, I will order 1 asap.

closed 2018-06-08 by negan07