negan07/ancistrus #40: Bufferbloat with latest update qos-sqm 2.0.0

pantelis86, 2018-12-01:


negan07, 2018-12-01:
?

negan07, 2018-12-01:
post the `/tmp/qos.state` content and
details & stats output of the gui

pantelis86, 2018-12-02:
cat: can't open '/tmp/var/lock/qos-sqm.lock': No such file or directory
cat: can't open '/tmp/var/lock/qos-sqm.lock': No such file or directory
cat: can't open '/tmp/var/lock/qos-sqm.lock': No such file or directory
cat: can't open '/tmp/var/lock/qos-sqm.lock': No such file or directory
cat: can't open '/tmp/var/lock/qos-sqm.lock': No such file or directory
cat: can't open '/tmp/var/lock/qos-sqm.lock': No such file or directory
cat: can't open '/tmp/var/lock/qos-sqm.lock': No such file or directory
cat: can't open '/tmp/var/lock/qos-sqm.lock': No such file or directory
cat: can't open '/tmp/var/lock/qos-sqm.lock': No such file or directory
cat: can't open '/tmp/var/lock/qos-sqm.lock': No such file or directory
ERROR: Giving up on getting lock after 10 attempts
ERROR: This is a bug: please issue report it

pantelis86, 2018-12-02:
http://www.dslreports.com/speedtest/42597790

pantelis86, 2018-12-02:
Version sqm 1.2.4 

---- qdisc ----
--- EGRESS ---
qdisc htb 1: root refcnt 2 r2q 10 default 41405 direct_packets_stat 0
 Sent 49279268 bytes 316879 pkt (dropped 0, overlimits 29118 requeues 0) 
 backlog 0b 0p requeues 0
qdisc fq_codel 110: parent 1:11 limit 1001p flows 1024 quantum 300 target 5.0ms interval 100.0ms ecn 
 Sent 49279268 bytes 316879 pkt (dropped 868, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
  maxpacket 1524 drop_overlimit 0 new_flow_count 36702 ecn_mark 0
  new_flows_len 1 old_flows_len 13
qdisc ingress ffff: parent ffff:fff1 ---------------- 
 Sent 859712511 bytes 596899 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
--- INGRESS ---
qdisc htb 1: root refcnt 2 r2q 10 default 41405 direct_packets_stat 0
 Sent 898751332 bytes 595898 pkt (dropped 0, overlimits 75767 requeues 0) 
 backlog 0b 0p requeues 0
qdisc fq_codel 110: parent 1:11 limit 1001p flows 1024 quantum 500 target 5.0ms interval 100.0ms ecn 
 Sent 720 bytes 4 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
  maxpacket 256 drop_overlimit 0 new_flow_count 1 ecn_mark 0
  new_flows_len 0 old_flows_len 1
qdisc fq_codel 120: parent 1:12 limit 1001p flows 1024 quantum 1500 target 5.0ms interval 100.0ms ecn 
 Sent 898750612 bytes 595894 pkt (dropped 1001, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
  maxpacket 1560 drop_overlimit 0 new_flow_count 27960 ecn_mark 0
  new_flows_len 1 old_flows_len 9
qdisc fq_codel 130: parent 1:13 limit 1001p flows 1024 quantum 300 target 5.0ms interval 100.0ms ecn 
 Sent 0 bytes 0 pkt (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
  maxpacket 256 drop_overlimit 0 new_flow_count 0 ecn_mark 0
  new_flows_len 0 old_flows_len 0

---- class ----
--- EGRESS ---
class htb 1:11 parent 1:1 leaf 110: prio 0 rate 5035Kbit ceil 5035Kbit burst 2.91309Kb cburst 2.91309Kb 
 Sent 50585024 bytes 317747 pkt (dropped 0, overlimits 0 requeues 0) 
 rate 52632bit 44ppsbacklog 0b 0p requeues 0
 lended: 316879 borrowed: 0 giants: 0
 tokens: 72297 ctokens: 72297

class htb 1:1 root rate 5035Kbit ceil 5035Kbit burst 2.91309Kb cburst 2.91309Kb 
 Sent 49279268 bytes 316879 pkt (dropped 0, overlimits 0 requeues 0) 
 rate 51464bit 44ppsbacklog 0b 0p requeues 0
 lended: 0 borrowed: 0 giants: 0
 tokens: 72297 ctokens: 72297

class fq_codel 110:2e parent 110: 
 (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
  deficit 136 count 0 lastcount 0 ldelay 3us
class fq_codel 110:85 parent 110: 
 (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
  deficit 228 count 0 lastcount 0 ldelay 3us
class fq_codel 110:b0 parent 110: 
 (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
  deficit 228 count 0 lastcount 0 ldelay 8us
class fq_codel 110:125 parent 110: 
 (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
  deficit 228 count 0 lastcount 0 ldelay 3us
class fq_codel 110:155 parent 110: 
 (dropped 0, overlimits 0 requeues 0) 
 backlog 0b 0p requeues 0
  deficit 228 count 0 lastcount 0 ldelay 3us
class fq_codel 110:1af parent 110: 
[... 137 lines cut ...]
filter parent 1: protocol arp pref 8 u32 fh 808: ht divisor 1 
filter parent 1: protocol arp pref 8 u32 fh 808::800 order 2048 key ht 808 bkt 0 flowid 1:11 
  match 00000000/00000000 at 0
filter parent 1: protocol all pref 999 u32 
filter parent 1: protocol all pref 999 u32 fh 800: ht divisor 1 
filter parent 1: protocol all pref 999 u32 fh 800::800 order 2048 key ht 800 bkt 0 flowid 1:12 
  match 00000000/00000000 at 8

--- MANGLE ---
Chain PREROUTING (policy ACCEPT 915K packets, 899M bytes)
 pkts bytes target     prot opt in     out     source               destination         
  148  116K DSCP       all  --  ppp1   *       0.0.0.0/0            0.0.0.0/0            DSCP match !0x00 DSCP set 0x00

Chain INPUT (policy ACCEPT 2039 packets, 419K bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain FORWARD (policy ACCEPT 913K packets, 899M bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain OUTPUT (policy ACCEPT 2029 packets, 424K bytes)
 pkts bytes target     prot opt in     out     source               destination         

Chain POSTROUTING (policy ACCEPT 915K packets, 899M bytes)
 pkts bytes target     prot opt in     out     source               destination         
  375 28520 DSCP       all  --  *      ppp1    0.0.0.0/0            0.0.0.0/0            DSCP match !0x00 DSCP set 0x00

pantelis86, 2018-12-02:
http://www.dslreports.com/speedtest/42600655

negan07, 2018-12-02:
upgrade busybox to fix

`opkg update && opkg install busybox`

closed 2018-12-04 by negan07