negan07/ancistrus #41: Problems with build #454

MrZeD81, 2018-12-07:
Hi.
I upgraded monolithic firmware from #380 to #454, but after the update the "Ancistrus" section into the router keep me noticing "New monolithic firmware avaiable" and the Build Revision Author Number is 380:
maybe you packed the wrong firmware into 454 package?
Regards
Dario

cryresta, 2018-12-07:
Hi Dario, you should try to access the router in a window in "**Incognito**" mode.
You will find out that the firmware has been updated correctly!
Unfortunately, browsing in normal mode cookies are used and probably the browser loads the old cookies of your router administration page.
I also encountered this problem and I solved it like this

Cristian

MrZeD81, 2018-12-07:
Tnx Cristian! I try and you're right! :)
Regards
Dario

negan07, 2018-12-09:
https://github.com/negan07/ancistrus/releases/latest

`Due to browser's cache retain main gui webpage may be mantained unaltered with the old version cached webpage: if this happen, refresh the internal frame of the webpage.`

closed 2018-12-09 by negan07