negan07/ancistrus #42: SSH Access Denied with the correct password (#454)

cryresta, 2018-12-07:
Hi everyone,
I installed build #454 a day ago and everything works perfectly except for ssh.
I state that it is the first time I try to connect in ssh always being connected in telnet and web interface.
And trying to connect I noticed that although I put the correct password (the same of the web interface) with the user 'root' but continues to return "Access Denied" (the problem is the same also with 'admin' user).
Has anyone already encountered this problem?

Thanks for collaboration.

negan07, 2018-12-07:
which ssh client in use ?

cryresta, 2018-12-08:
I use PuTTy 0.70

negan07, 2018-12-08:
follow the [tutorial](https://github.com/negan07/ancistrus/wiki/openssh-package)

ssh daemon must be turned on and active on GUI: check on core setup --> utilities -> process lists if sshd is running

on client setup must be done using port 22 with username: root

cryresta, 2018-12-10:
I had already activated the daemon and followed the tutorial before opening the issue, in fact I connected correctly with putty but when I entered the password I returned a constant: "Access denied" although the tutorial describes using the same password as the web interface .
I have the impression that it is:
or a key problem, or a configuration file problem (which is currently with default values).
Below are the logs of the ssh connection attempts

"**_[SSHD] Connection closed by authenticating root user 192.168.0.18 port 2115 [preauth] Monday, December 10,2018 09:17:25
[SSHD] Failed password for root from 192.168.0.18 port 2115 ssh2 Monday, December 10,2018 09:17:23_**"

I can not understand how it is possible that internally the ssh session is not called directly on port 22 but on another session port (2115 in this case).
I do not want the problem to be that .. I await updates.

negan07, 2018-12-10:
I tried to reset my own ssh keys then re run ssh daemon generating new keys with newer ssl 1.0.2q :
key was generated and login granted both on unix & putty as usual

the random port tcp 2115 is the internal socket port of the client, destination is common tcp 22

the debug log webpage on GUI is tiny so:
try to run from telnet console the sshd in foreground mode with:
`rc_ssh startforced debug`
leave the console open and then try connecting with putty and see the log output

did you change something in the `/usr/etc/sshd.conf` ?

cryresta, 2018-12-11:
Here is the complete log since the connection was attempted 

_

```
Connection from 192.168.0.15 port 1066 on 192.168.0.1 port 22
debug1: Client protocol version 2.0; client software version PuTTY_Release_0.70
debug1: no match: PuTTY_Release_0.70
debug1: Local version string SSH-2.0-OpenSSH_7.9
debug2: fd 3 setting O_NONBLOCK
debug3: ssh_sandbox_init: preparing rlimit sandbox
debug2: Network child is on pid 7084
debug3: preauth child monitor started
debug3: privsep user:group 0:0 [preauth]
debug1: permanently_set_uid: 0/0 [preauth]
debug1: list_hostkey_types: rsa-sha2-512,rsa-sha2-256,ssh-rsa,ecdsa-sha2-nistp25                                                                                        6,ssh-ed25519 [preauth]
debug3: send packet: type 20 [preauth]
debug1: SSH2_MSG_KEXINIT sent [preauth]
debug3: receive packet: type 20 [preauth]
debug1: SSH2_MSG_KEXINIT received [preauth]
debug2: local server KEXINIT proposal [preauth]
debug2: KEX algorithms: ecdh-sha2-nistp256,ecdh-sha2-nistp384,ecdh-sha2-nistp521                                                                                        ,diffie-hellman-group-exchange-sha256 [preauth]
debug2: host key algorithms: rsa-sha2-512,rsa-sha2-256,ssh-rsa,ecdsa-sha2-nistp2                                                                                        56,ssh-ed25519 [preauth]
debug2: ciphers ctos: aes128-ctr,aes192-ctr,aes256-ctr [preauth]
debug2: ciphers stoc: aes128-ctr,aes192-ctr,aes256-ctr [preauth]
debug2: MACs ctos: hmac-sha2-256,hmac-sha2-512 [preauth]
debug2: MACs stoc: hmac-sha2-256,hmac-sha2-512 [preauth]
debug2: compression ctos: none,zlib@openssh.com [preauth]
debug2: compression stoc: none,zlib@openssh.com [preauth]
debug2: languages ctos:  [preauth]
debug2: languages stoc:  [preauth]
debug2: first_kex_follows 0  [preauth]
debug2: reserved 0  [preauth]
debug2: peer client KEXINIT proposal [preauth]
debug2: KEX algorithms: curve25519-sha256@libssh.org,ecdh-sha2-nistp256,ecdh-sha                                                                                        2-nistp384,ecdh-sha2-nistp521,diffie-hellman-group-exchange-sha256,diffie-hellma                                                                                        n-group-exchange-sha1,diffie-hellman-group14-sha1,rsa2048-sha256,rsa1024-sha1,di                                                                                        ffie-hellman-group1-sha1 [preauth]
debug2: host key algorithms: ssh-ed25519,ecdsa-sha2-nistp256,ecdsa-sha2-nistp384                                                                                        ,ecdsa-sha2-nistp521,ssh-rsa,ssh-dss [preauth]
debug2: ciphers ctos: aes256-ctr,aes256-cbc,rijndael-cbc@lysator.liu.se,aes192-c                                                                                        tr,aes192-cbc,aes128-ctr,aes128-cbc,chacha20-poly1305@openssh.com,blowfish-ctr,b                                                                                        lowfish-cbc,3des-ctr,3des-cbc,arcfour256,arcfour128 [preauth]
debug2: ciphers stoc: aes256-ctr,aes256-cbc,rijndael-cbc@lysator.liu.se,aes192-c                                                                                        tr,aes192-cbc,aes128-ctr,aes128-cbc,chacha20-poly1305@openssh.com,blowfish-ctr,b                                                                                        lowfish-cbc,3des-ctr,3des-cbc,arcfour256,arcfour128 [preauth]
debug2: MACs ctos: hmac-sha2-256,hmac-sha1,hmac-sha1-96,hmac-md5,hmac-sha2-256-e                                                                                        tm@openssh.com,hmac-sha1-etm@openssh.com,hmac-sha1-96-etm@openssh.com,hmac-md5-e                                                                                        tm@openssh.com [preauth]
debug2: MACs stoc: hmac-sha2-256,hmac-sha1,hmac-sha1-96,hmac-md5,hmac-sha2-256-e                                                                                        tm@openssh.com,hmac-sha1-etm@openssh.com,hmac-sha1-96-etm@openssh.com,hmac-md5-e                                                                                        tm@openssh.com [preauth]
debug2: compression ctos: none,zlib [preauth]
debug2: compression stoc: none,zlib [preauth]
debug2: languages ctos:  [preauth]
debug2: languages stoc:  [preauth]
debug2: first_kex_follows 0  [preauth]
debug2: reserved 0  [preauth]
debug1: kex: algorithm: ecdh-sha2-nistp256 [preauth]
debug1: kex: host key algorithm: ssh-ed25519 [preauth]
debug1: kex: client->server cipher: aes256-ctr MAC: hmac-sha2-256 compression: n                                                                                        one [preauth]
debug1: kex: server->client cipher: aes256-ctr MAC: hmac-sha2-256 compression: n                                                                                        one [preauth]
debug1: expecting SSH2_MSG_KEX_ECDH_INIT [preauth]
debug3: receive packet: type 30 [preauth]
debug3: mm_sshkey_sign entering [preauth]
debug3: mm_request_send entering: type 6 [preauth]
debug3: mm_sshkey_sign: waiting for MONITOR_ANS_SIGN [preauth]
debug3: mm_request_receive_expect entering: type 7 [preauth]
debug3: mm_request_receive entering [preauth]
debug3: mm_request_receive entering
debug3: monitor_read: checking request 6
debug3: mm_answer_sign
debug3: mm_answer_sign: hostkey proof signature 0x3891c0(83)
debug3: mm_request_send entering: type 7
debug2: monitor_read: 6 used once, disabling now
[... cut ...]
debug3: ensure_minimum_time_since: elapsed 1.801ms, delaying 7.396ms (requested                                                                                         9.197ms) [preauth]
debug3: userauth_finish: failure partial=0 next methods="publickey,password,keyb                                                                                        oard-interactive,hostbased" [preauth]
debug3: send packet: type 51 [preauth]
debug3: receive packet: type 2 [preauth]
debug3: Received SSH2_MSG_IGNORE [preauth]
```

_

negan07, 2018-12-11:
logs is equal from begin to

```
debug3: mm_auth_password: waiting for MONITOR_ANS_AUTHPASSWORD [preauth]
debug3: mm_request_receive_expect entering: type 13 [preauth]
debug3: mm_request_receive entering [preauth]
debug3: mm_request_receive entering
debug3: monitor_read: checking request 12
debug3: mm_answer_authpassword: sending result 0
debug3: mm_request_send entering: type 13
Failed password for root from 192.168.0.15 port 1066 ssh2
debug1: Unable to open the btmp file /var/log/btmp: No such file or directory
```

```
debug3: mm_auth_password: waiting for MONITOR_ANS_AUTHPASSWORD [preauth]
debug3: mm_request_receive_expect entering: type 13 [preauth]
debug3: mm_request_receive entering [preauth]
debug3: mm_request_receive entering
debug3: monitor_read: checking request 12
debug3: mm_answer_authpassword: sending result 1
debug3: mm_request_send entering: type 13
Accepted password for root from 192.168.0.10 port 51624 ssh2
debug1: monitor_child_preauth: root has been authenticated by privileged process
```
WIth actual infos can't reproduce the issue, it seems to be an incompatibility of putty options

1. try to change temporary the http_password with an isalpha && isnum one (avoid some special chars for now), regenerate keys & try again
2. try to start from a whole clean version of putty without altering default settings (remove also stored keys and profiles if any)
3. try a different ssh client like termius or ssh to reproduce above issue

tried also another client like termius / shh without login problems
also automated login works with `ssh root@D7000`

negan07, 2018-12-14:
any news about this issue ?

cryresta, 2018-12-17:
Yes, unfortunately I was not able to make it work even following your last suggestions.
So it is not of vital use for me, but I would have liked it to work.

negan07, 2018-12-17:
which other clients did you try ?
have you tried to revert back to #380 ?

cryresta, 2019-01-01:
I have not tried to install the # 384 again because having already tested the problem of SSH with that version I thought it was useless.
The only other client I've tested apart from putty is "remoteNG"

negan07, 2019-01-07:
also a 7 level software firewall suite running on the host machine may block connection at login stage so make sure the ssh connection is granted

negan07, 2019-01-12:
any news about this stuff ?

closed 2019-01-18 by negan07