negan07/ancistrus #43: dnrd 'fails' regularly

jonwaland, 2018-12-12:
Not your problem, but a core issue with D7000 - the dns autoforwarder craps out after a large number of look ups - currently about a week here in my house.

So I want to setup a restart process say every night - do I just do a personal cronjob "rc_dnrd autostart" ?  Is there anything that gets kicked off by default on the hour (say 3am)? Do you know if there are any other flags to the rc_dnrd comand I need?

negan07, 2018-12-12:
`rc_dnrd autostart` doesn't exist

`rc_dnrd < stop | restart | start >`

but this is not a solution just a workaround: dnrd should not crash down in this way

better to investigate where's the problem with dnrd and eventually patch/upgrade it or replacing with something similar (dnsmasq ?)

any suggestion about another dns cache retainer/forwarder ?

jonwaland, 2018-12-12:
>how do/did u aware about failures ?
"dad - the internet is broken" - checked a bunch of devices and yes, it appeared to be dead.  reboots router.

couple weeks later, it happens again.
This time I did some investigation - ping to ip address works, ping to a locally cached address works. ping to a new address fails.... nslookup fails - but I have the router directing DNS look ups to an internal dnsmasq instance - I'm still using the DHCP on the router though as this is just easier.

SO investigating the DNS look ups - dns via the router is dead. DNS direct via the dnsmasq server is fine.  Don't know if there are any log messages on the router, but restarting dnrd fixes the issue.

(and yes - I meant restart not autostart - was badly multitasking)

Reading above, I'm assuming that the router initiated instance doesn't log - not knowing enough about the filesystem on this box, any suggestion on how and where to kick it off and have it log - then we can at least see if it reports any errors.

negan07, 2018-12-12:
upgrade busybox 

`opkg update && opkg install busybox
`
then logout/login and run 
`ps | grep dnrd`

copy the entire command you see
the cmd may be different depending on configurations
run `rc dnrd stop`
then run it in foreground pasting the same cmd adding `-d 9` at the end
look at the output msgs and paste them here

to log in background, `killall -9 dnrd` then replace `-d 9` above with `-l` but not sure if it will log something somewhere without code modification

jonwaland, 2018-12-14:
debug level 9 will kill filespace I suspect!!

```
admin@D7000:~$ /var/dnrd_1 -a 192.168.1.254 -m hosts -c off -r 0 -s 192.168.1.128 -d 9
Notice: caching turned off
Debug: initialising master DNS database
Debug: no blacklist: /etc/dnrd/blacklist
Warning: Using /etc/hosts will be removed in a future version. Please use only the /etc/dnrd/master file or use -m off.
Debug: initialising from /etc/hosts, domain= <none>
Debug: /etc/hosts: 7 records
Debug: added authority for 0.0.127.in-addr.arpa
Debug: added authority for 1.168.192.in-addr.arpa
Debug: 11 records in master DNS database
Debug: Received DNS query for "local"

- -- query
000 - B8 32 01 00 00 01 00 00 00 00 00 00 05 6C 6F 63  .2...........loc
010 - 61 6C 00 00 06 00 01                             al.....

id= 47154, q= 0, opc= 0, aa= 0, wr/ra= 1/0, trunc= 0, rcode= 0 [0100]
qd= 1
  name= local., type= 6, class= 1
ans= 0
ns= 0
ar= 0

Debug: Forwarding the query to DNS server 192.168.1.128
Debug: sending to: srv=192.168.1.128, client_time: 1544749597, ttl: 1, try_count: 0, sock 3, msg1dce8 ,len: 23
Debug: OK, let's wait for the response
Debug: Open sockets: 1, active: 1, count: 0, timeouts: 0
Debug: srv=192.168.1.128, myqid=37268, client_qid=12984
Debug: handling socket 3
Warning: RCODE was set. Ignoring reply from 192.168.1.128
Debug: check_reply failed
Debug: sending to: srv=192.168.1.128, client_time: 1544749597, ttl: 2, try_count: 1, sock 3, msg106f65c ,len: 23
Debug: OK, let's wait for the response
Debug: handling socket 3
Warning: RCODE was set. Ignoring reply from 192.168.1.128
Debug: check_reply failed
Debug: sending to: srv=192.168.1.128, client_time: 1544749597, ttl: 1, try_count: 2, sock 3, msg106f65c ,len: 23
Debug: OK, let's wait for the response
Debug: handling socket 3
Warning: RCODE was set. Ignoring reply from 192.168.1.128
Debug: check_reply failed
Debug: sending to: srv=192.168.1.128, client_time: 1544749597, ttl: 2, try_count: 3, sock 3, msg106f65c ,len: 23
Debug: OK, let's wait for the response
Debug: handling socket 3
Warning: RCODE was set. Ignoring reply from 192.168.1.128
Debug: check_reply failed
Debug: sending to: srv=192.168.1.128, client_time: 1544749597, ttl: 0, try_count: 4, sock 3, msg106f65c ,len: 23
Debug: OK, let's wait for the response
Debug: query_timeout: try_count: 5
Debug: sending to: srv=192.168.1.128, client_time: 1544749597, ttl: 0, try_count: 5, sock 3, msg106f65c ,len: 23
Debug: OK, let's wait for the response
Debug: handling socket 3
Warning: RCODE was set. Ignoring reply from 192.168.1.128
Debug: check_reply failed
Debug: Received DNS query for "local"

- -- query
000 - F7 48 01 00 00 01 00 00 00 00 00 00 05 6C 6F 63  .H...........loc
010 - 61 6C 00 00 06 00 01                             al.....

id= 63304, q= 0, opc= 0, aa= 0, wr/ra= 1/0, trunc= 0, rcode= 0 [0100]
qd= 1
  name= local., type= 6, class= 1
ans= 0
ns= 0
ar= 0

[... 147 lines cut ...]
ar= 0
```

debug =1 isn't great either

```
Notice: caching turned off
Debug: initialising master DNS database
Debug: no blacklist: /etc/dnrd/blacklist
Warning: Using /etc/hosts will be removed in a future version. Please use only the /etc/dnrd/master file or use -m off.
Debug: initialising from /etc/hosts, domain= <none>
Debug: /etc/hosts: 7 records
Debug: added authority for 0.0.127.in-addr.arpa
Debug: added authority for 1.168.192.in-addr.arpa
Debug: 11 records in master DNS database
Debug: Open sockets: 1, active: 1, count: 0, timeouts: 0
Warning: RCODE was set. Ignoring reply from 192.168.1.128
Debug: check_reply failed
Warning: RCODE was set. Ignoring reply from 192.168.1.128
Debug: check_reply failed
Warning: RCODE was set. Ignoring reply from 192.168.1.128
Debug: check_reply failed
```

I will check why I'm getting requests for 'local' though

negan07, 2018-12-14:
are you forwarding dns to dnsmasq at 192.168.1.128 ?
which dns servers did you setup ? manual or assigned from wan dhcp ? 
does the local dns server log something strange on dnsmasq ?

jonwaland, 2018-12-15:
yes - I have an intel nuc at 192.168.1.128 which I have dnsmaq running a local cache, plus doing requests from google (8.8.8.8) plus my ISP as a backup. 

but I'm using DHCP to allocate IP addresses around the network from the router (192.168.1.254), and if you use the router DHCP it offers itself as the only DNS server. I could move to using the NUC as DHCP, but it was easier to leave as it (40+ devices)  (there is no way in the Router GUI to force it to offer DHCP but with 192.168.1.128 as the default DNS server - this would be ideal. I should look at teh DHCP server options on the CLI I guess).

no errors show in the dnsmasq log. When dnrd fails, dns look ups to the router timeout - they never get to the NUC.

negan07, 2018-12-15:
dhcp is heavily customized and should be managed manually to forward something on internal lan group instead of wan side

but if I remember, dnsmasq acts also as dhcp server: if you don't need to use parental control genie app or xcloud or similar you can try to disable router one and use dnsmasq dhcp directly

jonwaland, 2018-12-17:
yes - dnsmasq does dhcp - I've just not had the will/time/effort to switch over - the D7000 is increasingly becoming an expensive adsl modem :-)

negan07, 2018-12-17:
[dnrd.zip](https://github.com/negan07/ancistrus/files/2688147/dnrd.zip)
I have compiled an updated version of dnrd called 2.21beta (2.20.4)

there are many code modifications I had to apply over because 2.19 from ng is very customized
and the new version includes some new stuff and options: some of them caused some issues I had to signal to the author.

Still to be tuned aligning ng cmd line option from `rc_dnrd` to new version, but seems stable for a try.

If you want to try it, do these cmds:
```
rc dnrd stop
cd /tmp
curl -k -O https://github.com/negan07/ancistrus/files/2688147/dnrd.zip
unzip dnrd.zip
chmod 755 dnrd
rm dnrd.zip
mv /usr/sbin/dnrd /usr/sbin/dnrd.orig
mv /tmp/dnrd /usr/sbin/
dnrd -a 192.168.1.254 -m /etc/hosts -c off -r 0 -s 192.168.1.128 -u admin
```

and to debug some stuffs, add -f -d [1-5] like this:
`dnrd -a 192.168.1.254 -m /etc/hosts -c off -r 0 -s 192.168.1.128 -u admin -f -d 5`

to revert back to the original dnrd:
```
dnrd -k || killall -9 dnrd || rc_dnrd stop
mv /usr/sbin/dnrd.orig /usr/sbin/dnrd
rc dnrd start
```

jonwaland, 2018-12-18:
```
admin@D7000:/tmp$ curl -k -O https://github.com/negan07/ancistrus/files/2688147/
dnrd.zip
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   559    0   559    0     0    535      0 --:--:--  0:00:01 --:--:--   535
admin@D7000:/tmp$ unzip dnrd.zip
Archive:  dnrd.zip
unzip: invalid zip magic 6D74683C
admin@D7000:/tmp$ cat dnrd.zip
<html><body>You are being <a href="https://github-production-repository-file-5c1aeb.s3.amazonaws.com/77776987/2688147?X-Amz-Algorithm=AWS4-HMAC-SHA256&amp;X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20181218%2Fus-east-1%2Fs3%2Faws4_request&amp;X-Amz-Date=20181218T051551Z&amp;X-Amz-Expires=300&amp;X-Amz-Signature=578da8464d4c9174ca418bc4921fbb736a96b399b5a7fd8bb89fe98716453b96&amp;X-Amz-SignedHeaders=host&amp;actor_id=0&amp;response-content-disposition=attachment%3Bfilename%3Ddnrd.zip&amp;response-content-type=application%2Fzip">redirected</a>.</body></html>admin@D7000:/tmp$
```

managed to get file onto a local server then curl it. 

```
admin@D7000:/tmp$ dnrd -a 192.168.1.254 -m /etc/hosts -c off -r 0 -s 192.168.1.1
28 -u admin -f -d 5
ERROR: tcpsock: Couldn't bind local address
Debug: Shutting down...
```

hmm.  retrying.

OK - got it running

So - it works. But if I want to get some useful debug, there is going to be a massive amount of data to trawl through

negan07, 2018-12-18:
yes the direct link maybe unavailable on issue threads so dl it locally then upload onto the router with `scp`

I would suggest you to try it for a while in daemon mode with:
`dnrd -a 192.168.1.254 -m /etc/hosts -c off -r 0 -s 192.168.1.128 -u admin`
and see if this version simply solves the bailing issue

otherwise if you want both to test & investigate on it you could try to launch it with a -f -d 3 debug level

jonwaland, 2018-12-18:
Its running now - has been for 18 hours thus far.

I'll leave it running and see if there is it fails - usually takes 1-2 weeks

negan07, 2018-12-18:
hoping this solves the issue

in the meanwhile will patch the program removing -u mandatory params & full path /etc/hosts to full match rc_dnrd then trying to send to the log events (the option -S does it but to write on gui messages it needs a tag) 
then the log may be sent to ip syslog server without bailing

jonwaland, 2018-12-19:
interesting point - I should set up syslog on another box.

negan07, 2018-12-19:
[dnrd.zip](https://github.com/negan07/ancistrus/files/2696879/dnrd.zip)
I made some modifications

now it should match main rc_dnrd behaviour so no need to type the cmd manually but simply:
`rc dnrd start` should start normally

I've added the syslog tags debug so if calling dnrd manually with debug -d 3 option it should write messages on the register router syslog on web gui
better not using more than -d 3 option to avoid flood hoping this is enough

then syslog can be sent though lan or a syslog local server with the register page option flags

hoping this could help identifying the bailing causes

jonwaland, 2018-12-21:
well, router just hard crashed. 

have just rebooted, and had to reinstall ancistrus-core, iproute2, and qos-sqm (although having installed qos-sqm, it correctly got the settings from nvram, which is good). I'm not running the monolithic image - is this correct? Also no longer seeing the package manager in the GUI?

For now Ive reverted to the firmware dnrd - wife acceptance factor etc. 

I need to get a remote syslog running elsewhere before I can restart this

negan07, 2018-12-21:
did dnrd run in debug mode ?

jonwaland, 2018-12-21:
when it crashed it was running the beta dnrd. But with no logging, as I dint' have a window open to it.

negan07, 2018-12-21:
I don't know why sercomm insists with this dns forwarder: all the embedded project have ignored it because there are many other like dnsmasq

I think that combination of your proper config through another dns forwarder has probably generated those bailings: the 2 versions are not so different and all the new features included are not in use

I suggest you to report as issue the recurrent bailing problem you encountered in the dnrd issue section: https://github.com/benjaminpetrin/dnrd/issues

I suggest you to include all the log output you have posted here all all the useful information you have to help the author to find the problem.

In the meanwhile my idea is to test dnrd with a similar situation redirecting dns queries from dhcp clients to the router dnrd and then logging it through its event register

jonwaland, 2018-12-24:
cool - alas, will all have to wait a week or two for now.

negan07, 2019-01-12:
any news about it ?

jonwaland, 2019-01-15:
OK - back on line. 

I've just set up a syslog server on my box that does everything (including dnsmasq) - so next step is to run either dnrd with debug logging enabled, or dnsmasq?

negan07, 2019-01-15:
one or another, not both, they can't socket the same port

I'd try dnsmasq because dnrd easily can give the same old results
(consider that dnsmasq doesn't include the hacks made by sercomm related to opendns, update url redir, netgear specs on recursion, and more but all there are all the fixes present in the openwrt package)

In second instance dnrd can be tried out

jonwaland, 2019-01-24:
sorry - off work so busy with "home stuff" - plus the DSL modem part of the D7000 has crapped out - now using an old billion 7800N as a DSL modem bridged into the D7000 as router/wifi.

still plan to get the newer dnrd running against my syslog server

jonwaland, 2019-01-31:
just run dnrd version 2.21_beta1 

```
dnrd -a 192.168.1.254 -m /etc/hosts -c off -r 0 -s 192.168.1.128 -u admin -f -d 3
Notice: caching turned off
Debug: initialising master DNS database
Debug: no blacklist: blacklist
Debug: /etc/hosts: 7 records
Debug: added authority for 0.0.127.in-addr.arpa
Debug: added authority for 1.168.192.in-addr.arpa
Debug: 11 records in master DNS database
getpwnam: No such file or directory
ERROR: Could not become "admin" user. Please create the user account or specify a valid user with  the -u option.
Debug: Shutting down...
```

restarted without the -u bit, and with -d 3

Running as -f

```
Notice: caching turned off
Debug: initialising master DNS database
Debug: no blacklist: blacklist
Debug: /etc/hosts: 7 records
Debug: added authority for 0.0.127.in-addr.arpa
Debug: added authority for 1.168.192.in-addr.arpa
Debug: 11 records in master DNS database
Debug: setting uid to 0
Debug: Received DNS query for "spclient.wg.spotify.com"
Debug: Forwarding the query to DNS server 192.168.1.128
Debug: sending to: srv=192.168.1.128, client_time: 1548911030, ttl: 1, try_count: 0, sock 3, msg1dc0c ,len: 41
Debug: OK, let's wait for the response
Debug: Received DNS query for "graph.facebook.com"
Debug: Forwarding the query to DNS server 192.168.1.128
Debug: sending to: srv=192.168.1.128, client_time: 1548911030, ttl: 1, try_count: 0, sock 6, msg1dc0c ,len: 36
Debug: OK, let's wait for the response
Debug: handling socket 3

```running without -f, from my syslog:
```
Jan 31 16:07:53 192.168.1.254 [DNRD] Forwarding the query to DNS server %s
Jan 31 16:07:53 192.168.1.254 [DNRD] sending to: srv=%s, client_time: %lu, ttl: %d, try_count: %d, sock %d, msg%x ,l
Jan 31 16:07:53 192.168.1.254 [DNRD] OK, let's wait for the response
Jan 31 16:07:54 192.168.1.254 [DNRD] handling socket %i
Jan 31 16:07:54 192.168.1.254 [DNRD] Received DNS reply for "%s"
Jan 31 16:07:54 192.168.1.254 [DNRD] Forwarding the reply to the host %s
Jan 31 16:07:54 192.168.1.254 [DNRD] Received DNS query for "%s"
Jan 31 16:07:54 192.168.1.254 [DNRD] Forwarding the query to DNS server %s
Jan 31 16:07:54 192.168.1.254 [DNRD] sending to: srv=%s, client_time: %lu, ttl: %d, try_count: %d, sock %d, msg%x ,l
Jan 31 16:07:54 192.168.1.254 [DNRD] OK, let's wait for the response
Jan 31 16:07:54 192.168.1.254 [DNRD] handling socket %i
Jan 31 16:07:54 192.168.1.254 [DNRD] Received DNS reply for "%s"
Jan 31 16:07:54 192.168.1.254 [DNRD] Forwarding the reply to the host %s
Jan 31 16:08:01 192.168.1.254 [DNRD] Received DNS query for "%s"
Jan 31 16:08:01 192.168.1.254 [DNRD] Forwarding the query to DNS server %s
Jan 31 16:08:01 192.168.1.254 [DNRD] sending to: srv=%s, client_time: %lu, ttl: %d, try_count: %d, sock %d, msg%x ,l
Jan 31 16:08:01 192.168.1.254 [DNRD] OK, let's wait for the response
```

regardless I'll leave it running and see if it dies.

negan07, 2019-01-31:
with the package version you can let it run directly transparently with `rc_dnrd restart` without any cmd line modifications needed apart  the debug -d 3 option as you did in the last try above

you should be able to read the dbg msgs on the log now
the package version is the 2.20.4 (same as 2.21b)

let's see if there are any issues

negan07, 2019-02-03:
[dnsblast.zip](https://github.com/negan07/ancistrus/files/2825719/dnsblast.zip)
I've found an easy dns forwarder stress test to do to accelerate and enhance the dnrd/dnsmasq stability test:
https://github.com/jedisct1/dnsblast

running from a linux machine host, with cmd like for ex:

`dnsblast dns_forwarder_ip 50000 100`

it sends 50000 queries with 100 queries/sec rate

contemporary, on the router console, with:

`top`

you can see dynamically the memory usage & cpu load (dnrd will run at top usage easily)

other examples are:

`How do I use it?
----------------

To send a shitload of queries to 127.0.0.1:

    dnsblast 127.0.0.1

To send 50,000 queries to 127.0.0.1:

    dnsblast 127.0.0.1 50000

To send 50,000 queries at a rate of 100 queries per second:

    dnsblast 127.0.0.1 50000 100

To send 50,000 queries at a rate of 100 qps to a non standard-port, like 5353:

    dnsblast 127.0.0.1 50000 100 5353

To send malformed packets, prepend "fuzz":

    dnsblast fuzz 127.0.0.1
    dnsblast fuzz 127.0.0.1 50000
    dnsblast fuzz 127.0.0.1 50000 100
    dnsblast fuzz 127.0.0.1 50000 100 5353
`
testing dnrd 2.20.4 with the cmd above with 50000 q at 100 q/sec the ratio is ~58% and cpu load is ~3%

with `dnsblast routerip` load goes to about 50% but ratio is less than 0.5%

with fuzz (malformed) queries all the queries are unresolved matching the expectations

dnrd it seems to remain stable, at least for a brief usage, cpu doesn't goes on saturation load remaining globally under 50% (75% considering the brcm internal buffer)

jonwaland, 2019-02-03:
installed the package dnrd just to be sure:
```
admin@D7000:~$ opkg install dnrd
Installing dnrd (2.20.4) on root.
Downloading https://raw.githubusercontent.com/negan07/ancistrus/gh-pages/ancistrus-arm-D7000/dnrd_2.20.4_armD7000.ipk.
Configuring dnrd.
```

restarted with -d 3

syslog is still faulty:

```
Feb  4 10:52:31 192.168.1.254 [DNRD] Received DNS query for "%s"
Feb  4 10:52:31 192.168.1.254 [DNRD] Forwarding the query to DNS server %s
Feb  4 10:52:31 192.168.1.254 [DNRD] sending to: srv=%s, client_time: %lu, ttl: %d, try_count: %d, sock %d, msg%x ,l
Feb  4 10:52:31 192.168.1.254 [DNRD] OK, let's wait for the response
Feb  4 10:52:31 192.168.1.254 [DNRD] handling socket %i
Feb  4 10:52:31 192.168.1.254 [DNRD] Received DNS reply for "%s"
Feb  4 10:52:31 192.168.1.254 [DNRD] Forwarding the reply to the host %s
```

will try see if I can hurt it though...

jonwaland, 2019-02-04:
lol
```
walaj@JonLabNUC:~/dnsblast/dnsblast-master$ ./dnsblast  192.168.1.254 50000 100
^Cnt: [5995] - Received: [2932] - Reply rate: [48 pps] - Ratio: [48.91%]
```

```
Feb  4 10:59:17 192.168.1.254 [DNRD] Socket limit reached. Dropping new queries
Feb  4 10:59:17 192.168.1.254 [DNRD] Received DNS query for "%s"
Feb  4 10:59:17 192.168.1.254 [DNRD] Forwarding the query to DNS server %s
Feb  4 10:59:17 192.168.1.254 [DNRD] Socket limit reached. Dropping new queries
Feb  4 10:59:17 192.168.1.254 [DNRD] Received DNS query for "%s"
Feb  4 10:59:17 192.168.1.254 [DNRD] Forwarding the query to DNS server %s
```

negan07, 2019-02-28:
anything new ?

jonwaland, 2019-02-28:
nothing new - not helped by various power outages, and my syslog server dying.

give me another couple weeks. or close - I've got enough to work with I think.

negan07, 2019-03-01:
all the time you need

does the server still crash sometimes ?

jonwaland, 2019-03-12:
I've had the router lock up - solid lights etc. hard reset required. Not sure if it dnrd though

negan07, 2019-03-12:
at this point inhibit dnrd service and try with dnsmasq

https://github.com/negan07/ancistrus/files/2706339/dnsmasq.zip

jonwaland, 2019-03-22:
ok - given up for now - has been up 15 days without a n issue, on your dnrd

I don't have time right now to migrate between my server dnsmasq and the embeded one.

Happy if you want to close this .

closed 2019-03-25 by negan07