negan07/ancistrus #44: DNRD problem when updating multiple hostnames

j27h, 2019-01-01:
I am using the DNS service provided by No-IP since two years now with a QNAP x86 NAS situated in another city. Yesterday I added a second hostname in my No-IP account and setted that DNS in my D7000 (build 454) in my actual home, 80 km away from the NAS, to set up an automatic backup routine using Rsync... and at this point I noticed a strange behaviour: everytime the D7000 updated the IP, both hostname's IP get updated in No-IP, changing also the one of the QNAP. Consequently, the NAS became inaccessible until QTS update his IP's hostname again.
Otherwise, when the public IP changes on the branded router connected to the QNAP, the NAS correctly updates only the IP connected to his specific hostname.

Seems like this is a know problem at Netgear, at least from firmware 1.0.1.60, see here:

[https://community.netgear.com/t5/DSL-Modems-Routers/D7000-Dynamic-DNS-feature-clobbers-all-No-IP-Hostnames/m-p/1683907#M25789](url)

Guess what... Netgear is ignoring the issue at the moment!

Is there a way to fix the DNRD so that only the correct hostname is updated when the D7000 public IP changes?

Thanks for all the work you keep doing!

Giorgio

negan07, 2019-01-01:
are you sure that dnrd is responsible of noip alias/ip association ?
this should be done by `noip2` client executable invoked by `rc_ddns`
if there's a bug inside `rc_apps` `rc_ddns` must be recreated

so there is a workaround:
create one dns with noip and the other with netgear dns service to skip the issue

anyway dnrd has been updated: don't think it will fix for the reason above
in any case there's also [dnsmasq](https://github.com/negan07/ancistrus/files/2706339/dnsmasq.zip) as an alternative to test

j27h, 2019-01-02:
No, I'm not sure that dnrd is the culprit. Even the issue is not that bothersome: the IP of the D7000 changes no more often tha one a week, and anyway within 30 minutes is overwritten with the correct one by the NAS.

I'm actually using the DNS provided by No-IP with the NAS, and a second one provided by Netgear (by the way, even this one is provided to Netgear by No-IP: "NETGEAR has partnered with NOIP to provide NETGEAR with our own host name domain") as a workaround. This way the two hostnames belong to two different non interfering No-IP account... same result obtainable opening directly two No-IP free account instead that adding a second hostname to a single account.

The latest version of DNRD is the 2.20.4 listed between the packages, or is there a more recent version (dnrd.zip) in the Issue 44 thread?

negan07, 2019-01-02:
the version released of dnrd is the last and includes all the hacks & modifications made by sercomm/netgear in the 2.19 old version

this won't fix the noip problem, it must be repaired acting on rc_apps (rc_ddns)

j27h, 2019-01-03:
I tied the new drnd 2.20.4 and nothing changed as you predicted.

Looking at the command you posted seem like the -Y option may be responsable for this issue. Just to better understand the results, I have to clarify that when you have more than one hostname registered in a single account, No-IP offer the option to create groups so that two or more hostnames could be updated with the same address by a single router/NAS. Trying to obtain the opposite, I created two groups, each one containing a single hostname, in the hope that this wolud led to and independent update fo the IPs.

This is the result of your command:

`noip2 -C -I $(anc nvram rget wan_ifname) -u "$(anc nvram rget ddns_noip_user_name)" -p "$(anc nvram rget ddns_noip_password)" -U 1440`

`Auto configuration for Linux client of no-ip.com.`

`2 groups are registered to this account.`
`Do you wish to have them all updated?[N] (y/N)  n`
`Do you wish to have group [GROUP1] updated?[N] (y/N) y`
`Do you wish to have group [GROUP2] updated?[N] (y/N) n`

`New configuration file '/etc/no-ip2.conf' created.`

As we can see, the two groups are seen, and the system ask if we want to update both, just the first one (GROUP1, D7000) or the latter (GROUP2, NAS). I specify that I wanted to update only GROUP1, and a new no-ip2.conf was created.

The D7000 keeps updating both hostnames anyway...

negan07, 2019-01-03:
it could be the schedule restart, embedded into the noip2 client with the netgear specs code modifications: this will invoke rc ddns which invokes noip2 with -Y option overriding and rewriting `/etc/no-ip2.conf`

[noip2.zip](https://github.com/negan07/ancistrus/files/2725506/noip2.zip)

this is the noip2 client compiled without netgear specs modifications to see if the issue was born into this piece of code
backup original noip2 rename it and replace it with
`scp noip2 root@D7000://usr/sbin/
`
note that the cron schedule should place a check every day at noon doing a `rc ddns restart`
to avoid this override, add `ddns` into core->schedule events->Type Original Cronjob Event to Disable here

then run noip2 manually as before try to see if the file is changed in the meanwhile

negan07, 2019-01-07:
any tryout made as above ?

j27h, 2019-01-07:
Yes, just some minutes ago: I have made a backup and replaced noip2 with the one you provided. I also disabled `ddns` chronjob.

Result: the router is unable to update the hostname. The error is the following: "_No update action. There is no IP address on the Internet port._"

[https://ibb.co/xDWW4Rx](url)

[https://ibb.co/6Z268Gn](url)

j27h, 2019-01-07:
I just noticed that, even if the router keeps displaying the error, it is still updating both hostnames (issue still present):

[https://ibb.co/FkpWSbv](url)

negan07, 2019-01-07:
The problem is mainly related to router wan interface associated to one wan ip: the client brings the wanif associated ip as the only one for both aliases

for it, this is logically correct

I think that the easiest way on your configuration is to have two independent aliases for local and remote host

the alternative is to disable noip on the router and override it manually with a query send&receive response using `curl `& creating an `/usr/etc/init.d/rc.post/ddns_post` script
every time rc_ddns is run it will send the query in the end

there are some other ddns update services, with larger providers support than those embedded on the router e.g. `inadyn` (but not sure if it supports an hack like that) or the `update_no-ip_com.sh` present on openwrt

closed 2019-10-24 by negan07