negan07/ancistrus #45: New Firmware v1.0.1.70 Released

lentis88, 2019-01-05:
http://www.downloads.netgear.com/files/GDC/D7000/D7000_V1.0.1.70_1.0.1.zip

negan07, 2019-01-07:
`https://www.downloads.netgear.com/files/GPL/R065-A-2018-0247_D7000v1 GPL.tgz`

link is wrong for sources, appears as
`https://www.downloads.netgear.com/files/GPL/R065-A-2018-0247_D7000v1 GPL.tgz`
with unwanted blank

tried many combinations, `R065-A-2018-0247_D7000v1` `GPL.tgz` `R065-A-2018-0247_D7000v1%20` %20GPL.tgz and so on unsuccessfully

MrZeD81, 2019-01-08:
D7000_V1.0.1.70_1.0.1.zip
https://www102.zippyshare.com/v/S7t70QlR/file.html

negan07, 2019-01-08:
the firmware image link is good, not the firmware sources https://kb.netgear.com/2649/NETGEAR-Open-Source-Code-for-Programmers-GPL

without firmware sources there's no way to identify differences on gpl source code
sent email to opensource support

negan07, 2019-01-15:
reply received: link should be fixed in a few days

Cranial, 2019-01-16:
[Temporary download link supplied to me by Netgear](https://ntgr-my.sharepoint.com/personal/ccook_netgear_com/_layouts/15/onedrive.aspx?id=%2Fpersonal%2Fccook_netgear_com%2FDocuments%2FGPL%2FR065-A-2018-0247_D7000v1%20GPL%2Etgz&parent=%2Fpersonal%2Fccook_netgear_com%2FDocuments%2FGPL&slrid=636eb69e-1059-0000-2600-b599dc2ff2eb)

negan07, 2019-01-16:
thanks but link is unreachable by guests

negan07, 2019-01-16:
[D7000_V1.0.1.68_apps_mini_httpd-1.17beta1-000-all.zip](https://github.com/negan07/ancistrus/files/2766204/D7000_V1.0.1.68_apps_mini_httpd-1.17beta1-000-all.zip)

Firmware modifications consist in a fix for multilogin sessions on web gui daemon.
Something else can concern closed code, if any.

negan07, 2019-01-17:
link fixed on netgear mirror & master moved to 1.0.1.70

closed 2019-01-21 by negan07

Cranial, 2019-02-04:
Not sure where else to put this. Just wanted to say thanks for this software. Having SNR Tweak available has reliably bumped my Downstream data rate from 7000 Kbps to just over 10000 Kbps.

negan07, 2019-02-05:
how did u obtain the source file from the bad link ?

Cranial, 2019-02-05:
I emailed Netgear directly and was supplied a temporary link by the open source team until they fixed the web link. I guess the wording of the email I used prompted them to supply the open source files as per the GPL.

negan07, 2019-02-05:
Same I did. I did it also on the community support. A new monolithic will be available soon: i want to add some stuff before.

Cranial, 2019-02-05:
Thank you @negan07 , excellent work.