negan07/ancistrus #47: Qos-sqm when using the D7000 as a router connected to a seperate ADSL modem

jonwaland, 2019-01-31:
Does Qos-sqm work correctly where the D7000 is connected to a seperate DSL modem running in bridge mode?

The settings GUI has correctly detected it is a pure ethernet connection rather than ADSL.

With an uplink of 1023 kbps and download of about 8000 kbps reported by the modem itself - what else do I set?

negan07, 2019-01-31:
the easiest way should be to connect the router directly

but sometimes this isn'y possible for isp limitations and moreover the sync of the vdsl routers in adsl mode is generally very conservative losing some bandwidth at the default snr condition

if the modem is in pure bridge mode the qos setup should reply the config you should obtain with the router connected directly

In this case it should be properly switch to adsl link-layer and then choose the fcp according to the adsl connection type depending on the isp (look at the modem config interface to discover it)

regarding bandwidth, to setup the limits it should be better to make a measurement before enabling qos with network-test assuring no other connections are present. Repeat the tests with different server locations in case of doubts or also give a try to UCLA upload only bandwidth test in the original GUI page.

the 1023kb of uplink is a limit value for ECN recommendation: if the real value is lower, but I suggest it in any case, it should be better to mantain the ECN uplink to off to avoid its overhead bandwidth usage.

jonwaland, 2019-02-01:
> the easiest way should be to connect the router directly 
yup - except the adsl modem subsystem no longer stays connected. even after factory reset and isolation test etc. But my old modem connects first time. everytime. but is a shit router

jonwaland, 2019-02-01:
> the 1023kb of uplink is a limit value for ECN recommendation: if the real value is lower, but I suggest it in any case, it should be better to mantain the ECN uplink to off to avoid its overhead bandwidth usage.

I find the uplink buffer management on the D7000 without the enhanced QoS is appalling.

negan07, 2019-02-01:
the main router should be set in pure bridge mode (modem only): and the routing traffic should be done by d7000 in ethernet mode

qos should be set to adsl link layer and, i suppose, pppoe-llc frame comp
if the limits are too high must be lowered of 5% at a first tryout, expecially if they are the bearer values (the real are easily lower)

jonwaland, 2019-02-04:
OK - did some more reading - the settings should match my ADSL connection - to ADSL PPPOE-LLC.

But with it set and downlink set to 85% of 8000kbps and uplink set to 85% of 1024kbps, the speedtest is 500kbps downlink! 

Does the qos-sqm package know which port to route traffic to when the DSL modem is not in use?

(ECN bit above - my mistake - I thought you meant QOS but realise you meant ECN as written)

jonwaland, 2019-02-04:
```
admin@D7000:~$ ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 16436 qdisc noqueue state UNKNOWN mode DEFAULT
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: ifb0: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN mode DEFAULT qlen 32
    link/ether 52:a2:f1:97:95:fe brd ff:ff:ff:ff:ff:ff
3: ifb1: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN mode DEFAULT qlen 32
    link/ether d6:00:b6:dc:68:98 brd ff:ff:ff:ff:ff:ff
4: sit0: <NOARP> mtu 1480 qdisc noop state DOWN mode DEFAULT
    link/sit 0.0.0.0 brd 0.0.0.0
5: ip6tnl0: <NOARP> mtu 1452 qdisc noop state DOWN mode DEFAULT
    link/tunnel6 :: brd ::
6: dsl0: <> mtu 0 qdisc noop state DOWN mode DEFAULT
    link/[29]
7: bcmsw: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 2048 qdisc noop state UNKNOWN mode DEFAULT qlen 1000
    link/ether 08:02:8e:dc:ae:18 brd ff:ff:ff:ff:ff:ff
8: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 2048 qdisc pfifo_fast state UP mode DEFAULT qlen 1000
    link/ether 08:02:8e:dc:ae:1c brd ff:ff:ff:ff:ff:ff
9: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast master group1 state DOWN mode DEFAULT qlen 1000
    link/ether 08:02:8e:dc:ae:18 brd ff:ff:ff:ff:ff:ff
10: eth2: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 2048 qdisc pfifo_fast master group1 state DOWN mode DEFAULT qlen 1000
    link/ether 08:02:8e:dc:ae:18 brd ff:ff:ff:ff:ff:ff
11: eth3: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast master group1 state DOWN mode DEFAULT qlen 1000
    link/ether 08:02:8e:dc:ae:18 brd ff:ff:ff:ff:ff:ff
12: eth4: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 2048 qdisc pfifo_fast master group1 state UP mode DEFAULT qlen 1000
    link/ether 08:02:8e:dc:ae:18 brd ff:ff:ff:ff:ff:ff
13: wl0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master group1 state UNKNOWN mode DEFAULT qlen 1000
    link/ether 08:02:8e:dc:ae:18 brd ff:ff:ff:ff:ff:ff
14: wl1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master group1 state UNKNOWN mode DEFAULT qlen 1000
    link/ether 08:02:8e:dc:ae:1a brd ff:ff:ff:ff:ff:ff
15: group1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT
    link/ether 08:02:8e:dc:ae:18 brd ff:ff:ff:ff:ff:ff
16: group2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UNKNOWN mode DEFAULT
    link/ether a2:77:19:d6:46:d1 brd ff:ff:ff:ff:ff:ff
17: wl0.1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master group1 state UNKNOWN mode DEFAULT qlen 1000
    link/ether 62:02:8e:dc:ae:19 brd ff:ff:ff:ff:ff:ff
18: wl0.2: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT qlen 1000
    link/ether 08:02:8e:dc:ae:18 brd ff:ff:ff:ff:ff:ff
19: wl0.3: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT qlen 1000
    link/ether 08:02:8e:dc:ae:18 brd ff:ff:ff:ff:ff:ff
20: wl1.1: <BROADCAST,MULTICAST> mtu 1500 qdisc pfifo_fast master group1 state DOWN mode DEFAULT qlen 1000
    link/ether 62:02:8e:dc:ae:1b brd ff:ff:ff:ff:ff:ff
21: wl1.2: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT qlen 1000
    link/ether 08:02:8e:dc:ae:1a brd ff:ff:ff:ff:ff:ff
22: wl1.3: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT qlen 1000
    link/ether 08:02:8e:dc:ae:1a brd ff:ff:ff:ff:ff:ff
33: eth0.1@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 2048 qdisc noqueue state UP mode DEFAULT
    link/ether 08:02:8e:dc:ae:19 brd ff:ff:ff:ff:ff:ff
34: ppp1: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1492 qdisc pfifo_fast state UNKNOWN mode DEFAULT qlen 255
    link/ppp
admin@D7000:~$ ip route
default via 150.101.32.41 dev ppp1
127.0.0.0/8 dev lo  scope link
150.101.32.41 dev ppp1  proto kernel  scope link  src 124.170.251.14
192.168.1.0/24 dev group1  proto kernel  scope link  src 192.168.1.254
239.0.0.0/8 dev group1  scope link
```

hmm - seems it uses ppp1 fine....

```
---- qdisc ----
--- EGRESS ---
qdisc htb 1: root refcnt 2 r2q 10 default 12 direct_packets_stat 0 ver 3.17
 linklayer atm overhead 40 mtu 2047 tsize 128 
qdisc fq_codel 110: parent 1:11 limit 1001p flows 1024 quantum 300 target 5.0ms interval 100.0ms 
qdisc fq_codel 120: parent 1:12 limit 1001p flows 1024 quantum 300 target 5.0ms interval 100.0ms 
qdisc fq_codel 130: parent 1:13 limit 1001p flows 1024 quantum 300 target 5.0ms interval 100.0ms 
qdisc ingress ffff: parent ffff:fff1 ---------------- 
--- INGRESS ---
[... 40 lines cut ...]

Chain INPUT (policy ACCEPT)
target     prot opt source               destination         

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination         

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination         
QOS_MARK_ppp1  all  --  0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xff
DSCP       all  --  0.0.0.0/0            0.0.0.0/0            DSCP match !0x00 DSCP set 0x00

Chain QOS_MARK_ppp1 (1 references)
target     prot opt source               destination         
MARK       all  --  0.0.0.0/0            0.0.0.0/0            MARK xset 0x2/0xff
MARK       all  --  0.0.0.0/0            0.0.0.0/0            DSCP match 0x08 MARK xset 0x3/0xff
MARK       all  --  0.0.0.0/0            0.0.0.0/0            DSCP match 0x30 MARK xset 0x1/0xff
MARK       all  --  0.0.0.0/0            0.0.0.0/0            DSCP match 0x38 MARK xset 0x1/0xff
MARK       all  --  0.0.0.0/0            0.0.0.0/0            DSCP match 0x2e MARK xset 0x1/0xff
MARK       all  --  0.0.0.0/0            0.0.0.0/0            DSCP match 0x2c MARK xset 0x1/0xff
MARK       all  --  0.0.0.0/0            0.0.0.0/0            tos match0x10/0x3f MARK xset 0x1/0xff
```

negan07, 2019-02-04:
> Does the qos-sqm package know which port to route traffic to when the DSL modem is not in use?

you mean the interface ?
`ppp1` is used in dsl mode, not ethernet
`eth0.1` is used in ethernet mode
the wan interface in use is set on nvram

`nvram get wan_ifname`

gives the wan intf name

is the measured speed is 500kbps maybe this is the real one, a trustable value for unbalanced atm dsl lines

to make a test you should connect with the other modem directly doing the same test and seeing if there are differences with the test made through the D7000

jonwaland, 2019-02-04:
> I have removed your own wan ip & gateway from the message above for security reason
oops! thanks

ok - all testing from 
without qos-sqm I can get **7.26 mbps** down/960kbps up.
With qos-sqm set to ADSL PPPOE-LLC (and 6800kbps/870kbps) I get 593kbps down/462kbps up
With qos-sqm set to TRUE Ethernet/Ethernet Link (and 6800kbps/870kbps) I get 791kbps down/420kbps up
with qos-sqm to to OTHER I get 488kbps down/375kbps up.

so I've turned it off. and will have to suffer poor uplink performance.

negan07, 2019-02-05:
this values with the router connected directly or through the other modem ?

which test did you try ?

jonwaland, 2019-02-06:
these values are all from a laptop connected via ethernet to the router, connected to the modem, testing via https://www.dslreports.com/speedtest

not sure how to see wan interface name is in the qos-sqm - I assumed ppp1 from the stats above

negan07, 2019-02-06:
is the router configured in wan or ethernet mode ?

ppp1 stands for wan mode = connected directly to the line

I suggest you to review connection & settings and to do a speed test directly from the router itself with qos disabled and all the other network peripherals turned off

jonwaland, 2019-02-07:
As the DSL modem is running in bridged mode, whilst I can access the device GUI if I connect directly to an ethernet port, I can't actually run a speed test or anything from it - is has no routing/DNS etc.

jonwaland, 2019-02-07:
there doesn't appear to be a qos_* variable for network port.

negan07, 2019-02-07:
> I still get ip route showing ppp1 as the internet facing port.

this shouldn't happen
did you connect properly the 2 routers through the lan 5 red port ?
the router can work in ethernet mode it has been developed for this

what happen if connecting router directly ?

jonwaland, 2019-02-08:
>did you connect properly the 2 routers through the lan 5 red port ? 

yup - otherwise I would have no internet access - the phone line is connected to the modem, not D7000

>what happen if connecting router directly ?
If you mean why am I not using the DSL modem in the D7000? Because it appears to be broken (connects for approx 5 seconds, then drops sync and retrains). Netgear support on this device is shit and I can't afford to replace it

negan07, 2019-02-08:
save your nvram settings (with the gui utilities, not the original one)

do an hard reset then connect the D7000 to the modem on the red port only
assure that the modem is connected properly to the adsl line
do the connection wizard choosing ethernet profile conf
don't select access point mode: this is another config mode turning off d7000 routing
then the wan port should result eth0.1

> Because it appears to be broken (connects for approx 5 seconds, then drops sync and retrains). Netgear support on this device is shit and I can't afford to replace it

this looks bad, if there aren't known incompatibilities between ISP and broadcom chipset (should look very unusual) the router should connect properly

if the wan port has been damaged it will be replaced with V2 different and out of this purpose

jonwaland, 2019-02-11:
> this looks bad, if there aren't known incompatibilities between ISP and broadcom chipset (should look very unusual) the router should connect properly

yup- it used to work. then failed. I guess I should check it again in case it was transitory.

Away from home this week.

negan07, 2019-02-28:
anything new about this ?

jonwaland, 2019-02-28:
no progress

negan07, 2019-03-21:
nothing new ?

a broken router cannot be useful: if in time, I suggest you to negotiate a substitution of it
in this case, it will be replaced with V2 intel lantiq different from this

otherwise, try the access point mode using one of the other port for wan

if the switch lan ports are still working the ethernet mode could be used also without the red port with some software modifications

jonwaland, 2019-03-22:
as posted elsewhere, not getting time to look into this right now.

I can't afford he downtime to RMA the router - it works as a router, just not as a DSL modem.

Next month I should be moving to HFC/cable supplied highspeed broadband - so I suspect I'll revist then. 

close if needed, or I'll be back end of April ish

closed 2019-03-25 by negan07