negan07/ancistrus #48: Web interface Access Problem after ReadySHARE USB activating

cryresta, 2019-02-02:
After activating the ReadyUSB SHARE feature of the router (which I had not tested yet)
There was an unusual problem: I can no longer access the router's web interface.
Before encountering the problem in the "ReadyUSB SHARE" section I had activated the access via network share (\\ folder) and access via HTTP ... I believe that the latter has generated some conflict of ports with the webUI that now returns this error:

[image]

Now I'm quite confused because I do not know if there is a way to disable the ReadySHARE USB (http access) function via CLI from telnet or there is a need to make a factory reset from the button ...
In the hypothesis of the factory reset from the button on the router there is no risk of bricking the router not being the stock firmware?

cryresta, 2019-02-02:
Obviously I've already rebooted it more than one times,  but the problem persists.. What do you advise me to do?

cryresta, 2019-02-02:
The Firmware version is # 454

cryresta, 2019-02-02:
After some verification I noticed that the problem occurs only on Google Chrome and only with the latest version (Version 72.0.3626.81 - 64 bit), so it is not clear if it is a too excessive restriction of chrome or a lack of implementation on ancistrus.

I await updates, thanks in advance

negan07, 2019-02-04:
activating the usb web service at http;//192.168.0.1/shares no access problem here or both the web gui with firefox gecko ESR on debian

there's no kind of know issues doing a reset of the settings with any kind of firmware

to override the setting involved without the web gui, enter the console with telnet/ssh and type:
```
nvram set http_server_enable=0
rc usb_service reconfig
```

closed 2019-02-05 by negan07