negan07/ancistrus #50: D7000v1 - Google Chrome can not access Admin Page 192.168.0.1 due to Nulls in Header

m0lz, 2019-03-03:
Anyone using Chromium (Google Chrome, or other derivatives) version 72 (or above), can not access a D7000 Admin page. If you type 192.168.0.1 in the URL Bar of Chrome, you will get an error page which reads "ERR_INVALID_HTTP_RESPONSE"

Firmware version installed = D7000_V1.0.1.70_1.0.1

See the following topics for reference ..

https://bugs.chromium.org/p/chromium/issues/detail?id=929170

https://bugs.chromium.org/p/chromium/issues/detail?id=927364

https://community.netgear.com/t5/DSL-Modems-Routers/GOOGLE-CHROME-ERROR-PAGE-DUE-TO-NULLS-IN-HEADER-OF-ROUTER/m-p/1701839#M26062

Example of NULLs being returned causing the issue ..

`HTTP/1.1 401 Unauthorized
Set-Cookie: <not including this here, though it's just a session token>
Server: 
Date: Thu, 07 Feb 2019 04:56:14 GMT
WWW-Authenticate: Basic realm="NETGEAR D7000"
Content-Type: text/html
Content-Language: en-US,en;q=0.9
X-Frame-Options: SAMEORIGIN
<NULL>
9
X-XSS-Protection: 1;mode=block
<NULL>
X-Content-Type-Options: nosniff
<NULL>Connection: close
`

Because the NULL characters are none Spec, Google Chromium project developers will not fix the issue (which would necessitate reverting Chrome to accepting NULLs), because from Googles point of view they are not only none spec, they are also a potential security risk.

Netgear have now end of life on D7000v1 and will also not fix the issue with the official firmware.

Its possible also that other browsers (Edge, Firefox) will also follow suit in not allowing NULLs to be accepted, so in future the D7000 can only be supported by the Netgear Genie, which is not Linux compatible (I think anyway), and woefully inadequate for advanced use.

Could you please take a look at this issue and possibly fix it ?

See comments #26 and #42 for pointers on what to fix, here 
https://bugs.chromium.org/p/chromium/issues/detail?id=927364

And see comment #44 closes the issue on Chromium as a "Wont Fix".

negan07, 2019-03-03:
the problem resides on `mini_httpd` headers string creation code

https://acme.com/software/mini_httpd/

all the router with this web server daemon are affected I think

the problem is that netgear uses a very old and customized version of it named 1.17b1, the latest is the 1.30

in the latest version the `add_headers()` is different, the piece of code involved above has been removed but this doesn't mean it fixes the issues

bumping the version, will break all the readycloud readyshare parental control & genie support without the necessary patches

m0lz, 2019-03-04:
I sent an email to someone at the mini_httpd project, and the response was ..

"Thanks for the report! I was not aware there were NULs in the headers,
that definitely sounds like a bug. I'll take a look soon."

So keep an eye out for updates to mini_httpd I reckon :)

negan07, 2019-03-04:
this is a good news, even if I was hoping the latest version didn't involve the issues

anyway

> Example of NULLs being returned causing the issue ..
>
> HTTP/1.1 401 Unauthorized Set-Cookie: <not including this here, though it's just a session token> Server: Date: Thu, 07 Feb 2019 04:56:14 GMT [...]

- from which chrome version is the problem involving ?
- where did you find the example of header code above ?
- is the string above complete ?

negan07, 2019-03-04:
maybe found the bug...

this executable seems to fix the problem

[mini_httpd.zip](https://github.com/negan07/ancistrus/files/2928295/mini_httpd.zip)

I've simply removed this piece of code:

[mini_httpd_null_headers_malformed_fix.zip](https://github.com/negan07/ancistrus/files/2928341/mini_httpd_null_headers_malformed_fix.zip)

what is the impact of this removal in the normal gui usage is to be known

m0lz, 2019-03-05:
I'm using the official 1.0.1.70_1.0.1 firmware image from Netgear available here https://www.netgear.com/support/product/D7000.aspx#Firmware%20Version%201.0.1.70

The patch you have linked in a zip, I do not know how to use it, but if the fix comes out in a firmware image I am willing to test - If its a case of you have too much on though and want to do other things before providing any firmware image I will keep popping in to check periodically for updates.

Good to see the label "Fixed" though :) .. Thank you very much for looking into it, dedicated programmers with the knowledge to solve the issues left behind ought to be paid by Netgear, a full time wage.

negan07, 2019-03-05:
fix confirmed also on chromium ipk mobile

the zip must be extracted on `/tmp` and moved overwriting the original `mini_httpd`

`rc http stop && mv mini_httpd /usr/sbin/`

the gui will be restarted then

blanchg, 2019-03-06:
Thank you! I can confirm this works, I had to manually kill the mini_httpd process as rc http stop didn't work

negan07, 2019-03-06:
sorry

`rc httpd stop`

anyway package available as `mini_httpd-1.17beta1-072`

`opkg update && opkg install mini_httpd`

Andreash67, 2019-03-07:
Apologies for the newbie question, but would someone explain the process of applying the patch to the firmware for the D7000, please?

negan07, 2019-03-07:
if not done yet, enter the console and download then execute `startup.sh`

https://negan07.github.io/ancistrus/

then install the fixed package with:

`opkg update && opkg install mini_httpd`

penlanone, 2019-03-07:
Can anyone advise on the following:

Installing essential packages: zlib openssl curl ancistrus-core utelnetd opkg
Downloading https://raw.githubusercontent.com/negan07/ancistrus/gh-pages/ancistrus-arm-D7000/Packages.gz. /etc/opkg: can't resolve symbol 'SSL_CTX_set_next_proto_select_cb'
Collected errors:  * opkg_prepare_url_for_install: Couldn't find anything to satisfy 'zlib'.
zlib installation failed: check repository urls on /etc/opkg.conf

many thanks.

negan07, 2019-03-07:
it's an openssl version mismatch

which fw version are you starting with ?

penlanone, 2019-03-07:
V1.0.1.70_1.0.1

negan07, 2019-03-08:
retry again from a clean reinstalled fw version

penlanone, 2019-03-08:
Perfect, thanks! Yes, ssl was hanging on my first try. I reinstalled a fresh fw, the script completed without error then "opkg update && opkg install mini_httpd " Excellent - no more Chrome errors! ;)

negan07, 2019-03-08:
the first try you made failed because the script was inadequately adapted to partition info/status creation

it was a long time bug discovered now

now startup.sh has been fixed

m0lz, 2019-03-09:
negan .. Huuuuge thank you 👍 

I have just installed the Firmware image build #552
( https://github.com/negan07/ancistrus/releases )

Had to use MS Edge to update the Firmware .. But now I have no problems accessing 192.168.0.1 with Chromium Version 72.0.3626.121 (Official Build) (64-bit).

Screenshot - https://i.imgur.com/4yoRkWH.png

And so many new options I have no clue about - I think I need to restrain myself from clicking too many things until I have some idea about all the capabilities therein :)

One final question : On the Ancistrus / Core setup / Service Inhibitions - Is the Telnet Daemon on by default on these routers (its something I have never had access to before, and noticed it is ticked enabled by default after installing this firmware, so wonder if that has always been the case even before installing your firmware) ?. I have turned that off for now, as its something I will not use personally and I have a feeling that makes the router more secure if its disabled, which is desirable in my case.

negan07, 2019-03-10:
> negan .. Huuuuge thank you 👍
>
> I have just installed the Firmware image build #552
> ( https://github.com/negan07/ancistrus/releases )
>
> Had to use MS Edge to update the Firmware .. But now I have no [...]

Have a read to the [wiki board](https://github.com/negan07/ancistrus/wiki) to understand all the options and packages before usage

> One final question : On the Ancistrus / Core setup / Service Inhibitions - Is the Telnet Daemon on by default on these routers (its something I have never had access to before, and noticed it is [...]

It can be disabled if you plan to use web gui only or an ssh client as console terminal.

The choice was to enable telnet console as default because initially there wasn't a web GUI project addition and the only way to install/remove/interact with the new modifications was the terminal console only.

Now it's possible to use telnet or ssh/scp with openssh package and the more intuitive web gui.

Anyway for security reason telnet remote access is disabled because the external login would be unencrypted and exposed to traffic loggers while it's possible to setup a more secure ssh remote console login.

> Edit: PS that closes the issue as far as I am concerned - Thank you very much for your help giving this superb router a few more years life. Will be checking in here and using any future builds and hope to offer more useful info for the project.
> 
> Edit 2: The Release file Build #552 was named D7000-V1.0.1.68_1.0.1-ancistrus_negan07-build-552.zip
> Its no problem as the content of the zip is correct, but ought to be named D7000-V1.0.1.70_1.0.1-ancistrus_negan07-build-552.zip

Sorry my mistake

Also the previous monolithic version zip pack was badly named, now both corrected, thanks for the notification.

m0lz, 2019-03-11:
:) Thanks again, now that I know what the inhibits do they are a very welcome addition, some nice options there. For now I have inhibited Netgears xCloud because I will never use it.

closed 2019-03-18 by negan07

m0lz, 2019-04-05:
@negan07 I know this is closed and you may never read it again, but if you do I would just like to say I have now updated again to #592 and continue to be amazed at the superb unpaid work you are doing here.

Also note, with the official netgear firmware my family used to occasionally experience what seemed to be the router locking up. It used to happen maybe once every couple of weeks, or sometimes it would be a few month before it happened, but when it happened we needed to restart the router to clear the issue.

We have not experienced that issue at all since starting using your firmware updates, and regularly we can have up to 16 devices connected by WIFI plus a WIFI extender upstairs.

You have clearly fixed a lot more underlying issues with this firmware, that we would not have received from Netgear.

Many thanks again, its very much appreciated :).

negan07, 2019-04-08:
> I know this is closed and you may never read it again
notifications are still active for this but it seems the problem to be fixed

i don't know if the other problems mentioned are firmware related but the monolithic has a deflated size and resource usage in the average and will help adding more features in the future