negan07/ancistrus #51: Genie App

lentis88, 2019-03-04:
With the latest 526 firmware the Genie app does not work properly anymore, always reports "wrong password".
I tried to reset it several times and reinstall the official 1.0.1.70 version and the app works correctly.
Someone I encountered the same problem?

negan07, 2019-03-04:
did this involved also #454 ?

lentis88, 2019-03-05:
Nope, but apparently it's a problem of the app.
On the Netgear> Genie forum, many people have the same problem.

negan07, 2019-03-05:
did some test

build #526 with latest package updates (included openssl)
genie app from windows: all good
genie app from ipk android: all good
genie app from ios: not good

in this case, got the login after some tries on the genie gui: tick/untick save password and/or setup router ip and/or pwd visibile/not visible and then after some tries it did it

I agree with genie app mobile problem occurred on the latest update

negan07, 2019-03-06:
In this case the fix must arrive from netgear devs because the genie source code is not available

anyway it seems the bug ticket has been dispatched as confirmed in the discussion above

negan07, 2019-03-28:
Maybe found the bug.

Not a specific genie app issue but a firmware regeneration one present on the source code.

Every time a monolithic or original firmware was made, the file `/www.eng/currentsetting.htm` was recreated but in an unappropriate way with wrong variables.

The original firmware wasn't affected by this problem because its  `/www.eng/currentsetting.htm` file was right and different from the one obtainable after making the firmware from sources.

This is the correct one
[currentsetting.zip](https://github.com/negan07/ancistrus/files/3018247/currentsetting.zip)

For those IOS people users wanting to fix the local login Genie app problem quickly, simply log the console and:

`echo "LoginMethod=2.0" >>/www.eng/currentsetting.htm`

should be satisfying
Package core update will come.

In the meanwhile it should be safer to compare each orig firmware file with those present on sources to see if there are other unwanted mismatches.

closed 2019-03-31 by negan07