negan07/ancistrus #52: OpenVPN: TLS Key negotiation failed / connectivity

ComputaBloke, 2019-05-21:
I've checked prior OpenVPN ticket #5 and sources on Netgear and OpenVPN and can't identify the correct solution for this, so I hope you might be able to help!

Error in client logs (smart phone):
`TLS Error: TLS key negotiation failed to occur within 60 seconds (check your network connectivity)`
I can also see **no** connection attempt on the server-side OpenVPN logs.

I'm using latest D7000 Ancistrus monolithic image V1.0.1.70_1.0.1, where you have enabled versions: OpenVPN 2.4.7; OpenSSL 1.0.2r; 26 Feb 2019
Through D7000 WebGUI (Ancistrus > Management > OPENVPN), I've configured OpenVPN TAP/TUN server (server.conf & server_phone.conf), mostly defaults with _two server options_: force UDP4 (ignore IP6 to try to keep config simple to pre 2.4 levels), and to show all parameters to help me debug through logs:
- proto udp4
- mute 256

Also enabled your OpenSSH package, and through PuTTY terminal, have checked `ps` and `netstat -a` confirms both `/usr/sbin/openvpn` processes are running with corresponding TAP/TUN bridge config files, and that there are mappings for default configured TAP/TUN ports.

The only info now in server logs is: `NOTE: when bridging your LAN adapter with the TAP adapter, note that the new bridge adapter will often take on its own IP address that is different from what the LAN adapter was previously set to`
UDPv4 link local (bound): [AF_INET] x.x.1.254:12973
/bin/ip addr add dev tun0 x.x.2.1/24 broadcast x.x.2.255

I have exported and built monolithic *.ovpn client files (with embedded ca, cert and key) and successfully imported these into two clients for:
- Windows: OpenVPN GUI (OpenVPN 2.4.7 x86_64-w64-mingw32; OpenSSL 1.1.0j; 20 Nov 2018); and
- Android: OpenVPN for Android (OpenVPN 2.5-icsopenvpn/v0.7.8.0g; OpenSSL 1.1.1b; Feb 22 2019)

Both clients timeout, can't get past TLS negotiation, with or without LZ compression.

The remote IP is correct and statically assigned. Client/Server versions all seem close enough to work.
Not sure if I have to configure something to use the `tun0` x.x.2.0 subnet, but at this stage think not?
I've not played with remote management, port forwarding or iptables at this point since other posts have been conflicting as to whether changes to forwarding/firewalls might be required when the VPN server is running on the gateway device itself.

Happy to share some more detail logs if it would help.
Can you offer me any tips!?

ComputaBloke, 2019-05-21:
Oh, I just noticed while there is no attempts in the VPN logs, the Server log does show VPN attempts:
`[OpenVPN, connection drop]from remote IP address: [undef] Tuesday, May 21,2019 11:52:55 `

negan07, 2019-10-10:
unfortunately can't reproduce the situation

any news about this ?

closed 2019-10-24 by negan07