negan07/ancistrus #53: OpenVPN to NordVPN (provider VPN)

cryresta, 2019-05-21:
Hello,
I was wondering if it was possible to create a vpn tunnel to a vpn provider (like NordVpn) using OpenVPN.
I already configured the classic VPN client to the netgear and it works fine.
But instead it is possible to connect the router as a client to one of the NordVPN servers (having the ovpn file obviously importable) in order to crypt all the outgoing traffic?

Sorry if I opened this request as an "issue" but I didn't know how else to contact you

Thanks in advance.

negan07, 2019-10-10:
don't know that vpn prov and its configuration, the internal router unix vpn client maybe needs a reconfig

have a look at:
https://downloads.nordcdn.com/configs/archives/servers/ovpn.zip

in fact many settings look different than built in ones

negan07, 2019-10-24:
any improvement about it ?

tommaso82, 2019-10-26:
I'm also trying to connect to a vpn as a client ... the connection actually happens but there is no navigation the packets are routed badly and there is a dns problem

it would be great if it worked

tommaso82, 2019-10-26:
I also tried to set the router as an access point (to solve the routing problem) but in that mode the tunnel device is missing and there was no way to create it

negan07, 2019-10-26:
I don't understand if it'a nordvpn connection attempt failure or something general purpose related.
Please specify.

tommaso82, 2019-10-26:
I was speaking in general, as far as my personal vpn server is concerned, if I launch from the router openvpn --config myconfig.ovh the connection as a client happens but there is no way to set the correct routing to use it or at least I am not up to it
and it would be very useful for me to be able to use this modem as a vpn client because it has
pretty powerful cpu that allows decent speeds in vpn

negan07, 2019-10-26:
>  it would be very useful for me to be able to use this modem as a vpn client because it has
> pretty powerful cpu that allows decent speeds in vpn
not the purpose of the package

the `openvpn` package option, like in the original firmware, acts as a vpn server router, while phones and other devices are clients connecting to it

to act as a client, the router needs a different configuration working mode: can be setup and executed manually

tommaso82, 2019-10-27:
I know it is not the purpose of the package, but at software level to make it working as a client nothing is missing. I suppose it's something in IPTABLES that prevents it from working correctly, I'm trying to make it work and I hope with your help to succeed, maybe a once it works we can think of a web interface.

negan07, 2019-10-27:
this could be a good cue to improve openvpn config functionality opening to personal config or templates

some client examples are present in the openvpn git section: https://github.com/OpenVPN/openvpn/blob/master/sample/sample-config-files/client.conf
https://github.com/OpenVPN/openvpn/blob/master/sample/sample-config-files
https://nordvpn.com/it/tutorials/linux/openvpn/

iptables rules should't need to be involved because as a default policy valid related & established outbound connections are always enabled

tommaso82, 2019-10-27:
i have a vps (virtual private server) where i run some stuff included openvpn (server)
i have tried to connect with the d7000 as client using correct configuration
the connection happens but then there is no way to go on line and i dont know whi.

I guess should be something related routing/iptables rules
but i can't figure out easly i m working on it

negan07, 2019-10-28:
are the 2 devices connected to different network geographic or wan kind different ?

post here the openvpn log ( `/bin/cat /tmp/openvpn_log` )
both for vps server too if possible may help

tommaso82, 2019-10-28:
`root@D7000:/usr/openvpn$ insmod /lib/modules/tun.ko
root@D7000:/usr/openvpn$ openvpn --mktun --dev tun0
Mon Oct 28 21:50:12 2019 TUN/TAP device tun0 opened
Mon Oct 28 21:50:12 2019 Persist state set to: ON

root@D7000:/tmp$ route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         192.168.100.1   0.0.0.0         UG    0      0        0 ppp1
127.0.0.0       *               255.0.0.0       U     0      0        0 lo
192.168.0.0     *               255.255.255.0   U     0      0        0 group1
192.168.100.1   *               255.255.255.255 UH    0      0        0 ppp1

openvpn --config myconfig.ovpn

Mon Oct 28 21:54:41 2019 SENT CONTROL [server_KnEYeLYIke90EDPT]: 'PUSH_REQUEST' (status=1)
Mon Oct 28 21:54:41 2019 PUSH: Received control message: 'PUSH_REPLY,dhcp-option DNS 1.0.0.1,dhcp-option DNS 1.1.1.1,redirect-gateway def1 bypass-dhcp,route-gateway 10.9.0.1,topology subnet,ping 10,ping-restart 120,ifconfig 10.9.0.98 255.255.255.0,peer-id 0,cipher AES-128-GCM'
Mon Oct 28 21:54:41 2019 OPTIONS IMPORT: timers and/or timeouts modified
Mon Oct 28 21:54:41 2019 OPTIONS IMPORT: --ifconfig/up options modified
Mon Oct 28 21:54:41 2019 OPTIONS IMPORT: route options modified
Mon Oct 28 21:54:41 2019 OPTIONS IMPORT: route-related options modified
Mon Oct 28 21:54:41 2019 OPTIONS IMPORT: --ip-win32 and/or --dhcp-option options modified
Mon Oct 28 21:54:41 2019 OPTIONS IMPORT: peer-id set
Mon Oct 28 21:54:41 2019 OPTIONS IMPORT: adjusting link_mtu to 1624
Mon Oct 28 21:54:41 2019 OPTIONS IMPORT: data channel crypto options modified
Mon Oct 28 21:54:41 2019 Data Channel: using negotiated cipher 'AES-128-GCM'
Mon Oct 28 21:54:41 2019 Outgoing Data Channel: Cipher 'AES-128-GCM' initialized with 128 bit key
Mon Oct 28 21:54:41 2019 Incoming Data Channel: Cipher 'AES-128-GCM' initialized with 128 bit key
Mon Oct 28 21:54:41 2019 TUN/TAP device tun1 opened
Mon Oct 28 21:54:41 2019 TUN/TAP TX queue length set to 100
Mon Oct 28 21:54:41 2019 /bin/ip link set dev tun1 up mtu 1500
Mon Oct 28 21:54:41 2019 /bin/ip addr add dev tun1 10.9.0.98/24 broadcast 10.9.0.255
Mon Oct 28 21:54:41 2019 /bin/ip route add 164.132.111.80/32 via 192.168.100.1
Mon Oct 28 21:54:41 2019 /bin/ip route add 0.0.0.0/1 via 10.9.0.1
Mon Oct 28 21:54:41 2019 /bin/ip route add 128.0.0.0/1 via 10.9.0.1
Mon Oct 28 21:54:41 2019 Initialization Sequence Completed

not working

openvpn --config myconfig.ovpn --redirect-gateway

Mon Oct 28 22:03:45 2019 SENT CONTROL [server_KnEYeLYIke90EDPT]: 'PUSH_REQUEST' (status=1)
Mon Oct 28 22:03:45 2019 PUSH: Received control message: 'PUSH_REPLY,dhcp-option DNS 1.0.0.1,dhcp-option DNS 1.1.1.1,redirect-gateway def1 bypass-dhcp,route-gateway 10.9.0.1,topology subnet,ping 10,ping-restart 120,ifconfig 10.9.0.98 255.255.255.0,peer-id 1,cipher AES-128-GCM'
Mon Oct 28 22:03:45 2019 OPTIONS IMPORT: timers and/or timeouts modified
Mon Oct 28 22:03:45 2019 OPTIONS IMPORT: --ifconfig/up options modified
Mon Oct 28 22:03:45 2019 OPTIONS IMPORT: route options modified
Mon Oct 28 22:03:45 2019 OPTIONS IMPORT: route-related options modified
Mon Oct 28 22:03:45 2019 OPTIONS IMPORT: --ip-win32 and/or --dhcp-option options modified
Mon Oct 28 22:03:45 2019 OPTIONS IMPORT: peer-id set
Mon Oct 28 22:03:45 2019 OPTIONS IMPORT: adjusting link_mtu to 1624
Mon Oct 28 22:03:45 2019 OPTIONS IMPORT: data channel crypto options modified
Mon Oct 28 22:03:45 2019 Data Channel: using negotiated cipher 'AES-128-GCM'
Mon Oct 28 22:03:45 2019 Outgoing Data Channel: Cipher 'AES-128-GCM' initialized with 128 bit key
Mon Oct 28 22:03:45 2019 Incoming Data Channel: Cipher 'AES-128-GCM' initialized with 128 bit key
Mon Oct 28 22:03:45 2019 TUN/TAP device tun1 opened
Mon Oct 28 22:03:45 2019 TUN/TAP TX queue length set to 100
Mon Oct 28 22:03:45 2019 /bin/ip link set dev tun1 up mtu 1500
Mon Oct 28 22:03:45 2019 /bin/ip addr add dev tun1 10.9.0.98/24 broadcast 10.9.0.255
Mon Oct 28 22:03:45 2019 /bin/ip route add 164.132.111.80/32 via 192.168.100.1
Mon Oct 28 22:03:45 2019 /bin/ip route add 0.0.0.0/1 via 10.9.0.1
Mon Oct 28 22:03:45 2019 /bin/ip route add 128.0.0.0/1 via 10.9.0.1
Mon Oct 28 22:03:45 2019 Initialization Sequence Completed
^CMon Oct 28 22:04:26 2019 event_wait : Interrupted system call (code=4)
Mon Oct 28 22:04:26 2019 /bin/ip route del 164.132.111.80/32
Mon Oct 28 22:04:26 2019 /bin/ip route del 0.0.0.0/1
Mon Oct 28 22:04:26 2019 /bin/ip route del 128.0.0.0/1
Mon Oct 28 22:04:26 2019 Closing TUN/TAP interface
Mon Oct 28 22:04:26 2019 /bin/ip addr del dev tun1 10.9.0.98/24
Mon Oct 28 22:04:26 2019 SIGINT[hard,] received, process exiting

[... 7 lines cut ...]
default         10.9.0.1        128.0.0.0       UG    0      0        0 tun1
default         192.168.100.1   0.0.0.0         UG    0      0        0 ppp1
10.9.0.0        *               255.255.255.0   U     0      0        0 tun1
127.0.0.0       *               255.0.0.0       U     0      0        0 lo
128.0.0.0       10.9.0.1        128.0.0.0       UG    0      0        0 tun1
164.132.111.80  192.168.100.1   255.255.255.255 UGH   0      0        0 ppp1
192.168.0.0     *               255.255.255.0   U     0      0        0 group1
192.168.100.1   *               255.255.255.255 UH    0      0        0 ppp1

tried also 
openvpn --config glnetita.ovpn --pull-filter ignore redirect-gateway
i got this while running

root@D7000:/tmp$ route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         192.168.100.1   0.0.0.0         UG    0      0        0 ppp1
10.9.0.0        *               255.255.255.0   U     0      0        0 tun1
127.0.0.0       *               255.0.0.0       U     0      0        0 lo
192.168.0.0     *               255.255.255.0   U     0      0        0 group1
192.168.100.1   *               255.255.255.255 UH    0      0        0 ppp1

now i m online but with my isp ip

`

tommaso82, 2019-10-28:
I did another test, I configured the d7000 from the advanced menu in wireless access point mode.
When i launch openvpn as client i get connected and if i check my ip from d7000 itself i get my vps ip
root@D7000:~$ curl http://api.ipify.org/?format=plain -w "\n"
my.vps.ip

negan07, 2019-10-29:
to test iptables role on it, with safe care, try to switch the INPUT chain default policy from DROP to ACCEPT

`iptables  -P INPUT ACCEPT`

or either

`iptables -P FORWARD ACCEPT`

but consider it temporarly because in this mode the router is very exposed
revert back to:

`iptables  -P INPUT DROP`

`iptables -P FORWARD DROP`

as soon as possible

best should be making test between 2 ethernet router under a local subnet

tommaso82, 2019-10-29:
I tried what you said and makes no difference. 
The connection on the device is working (not for dns packet udp:53 get filtred) if I try to check my ip I get the one of the remote endpoint 
But from the outside (wifi or ethernet client does not go) I am attaching some test logs made with tcpdump

`vpn connected, pinging 1.1.1.1 from d7000` 
` root@D7000:~$ ping 1.1.1.1` 
` PING 1.1.1.1 (1.1.1.1): 56 data bytes` 
` 64 bytes from 1.1.1.1: seq=0 ttl=55 time=36.562 ms` 
` 64 bytes from 1.1.1.1: seq=1 ttl=55 time=35.736 ms` 

` root@D7000:~$ tcpdump -nn -i tun0 host 1.1.1.1` 
` tcpdump: verbose output suppressed, use -v or -vv for full protocol decode` 
` listening on tun0, link-type RAW (Raw IP), capture size 65535 bytes` 
` 00:33:12.068148 IP 10.9.0.98 > 1.1.1.1: ICMP echo request, id 62991, seq 17, length 64` 
` 00:33:12.103735 IP 1.1.1.1 > 10.9.0.98: ICMP echo reply, id 62991, seq 17, length 64` 

no traffic on ppp1, not even on group1

vpn link is working traffic is correctly routed and we are on-line using remote endpoint ip but dns are not working

next test:

vpn connected, pinging from windows machine connected wifi to d7000

` C:\Users\tom>ping 1.1.1.1` 

` Esecuzione di Ping 1.1.1.1 con 32 byte di dati:` 
` Risposta da 1.1.1.1: byte=32 durata=30ms TTL=55` 
` Risposta da 1.1.1.1: byte=32 durata=29ms TTL=55` 

` root@D7000:~$ tcpdump -nn -i ppp1 host 1.1.1.1` 
` tcpdump: verbose output suppressed, use -v or -vv for full protocol decode` 
` listening on ppp1, link-type LINUX_SLL (Linux cooked), capture size 65535 bytes` 
` 00:38:43.410733 IP MY.PUBLIC.ISP.IP > 1.1.1.1: ICMP echo request, id 1, seq 30970, length 40` 
` 00:38:43.440010 IP 1.1.1.1 > MY.PUBLIC.ISP.IP: ICMP echo reply, id 1, seq 30970, length 40` 
` 00:38:44.415699 IP MY.PUBLIC.ISP.IP.56.17.248 > 1.1.1.1: ICMP echo request, id 1, seq 30972, length 40` 
` 00:38:44.445180 IP 1.1.1.1 > MY.PUBLIC.ISP.IP: ICMP echo reply, id 1, seq 30972, length 40` 

` root@D7000:~$ tcpdump -nn -i group1 host 1.1.1.1` 
` tcpdump: verbose output suppressed, use -v or -vv for full protocol decode` 
` listening on group1, link-type EN10MB (Ethernet), capture size 65535 bytes` 
` 00:40:10.671473 IP 192.168.0.2 > 1.1.1.1: ICMP echo request, id 1, seq 31056, length 40` 
` 00:40:10.700548 IP 1.1.1.1 > 192.168.0.2: ICMP echo reply, id 1, seq 31056, length 40` 
` 00:40:11.685495 IP 192.168.0.2 > 1.1.1.1: ICMP echo request, id 1, seq 31058, length 40` 
` 00:40:11.715584 IP 1.1.1.1 > 192.168.0.2: ICMP echo reply, id 1, seq 31058, length 40` 

no traffic on tun0 device

tommaso82, 2019-10-31:
I'm trying to do some iptables debugging but it seems that there are no raw tables 
I thought I would do it using rules like these
 
`iptables -t raw -A OUTPUT -p icmp -j TRACE`
 `iptables -t raw -A PREROUTING -p icmp -j TRACE`

but can't add it... solutions?

negan07, 2019-11-01:
disable built in openvpn daemon
```
nvram set openvpn_enable=0
rc openvpn stop
```
then run manually the procedure to enable openvpn as client with personal config
```
insmod /lib/modules/tun.ko
openvpn --mktun --dev tap0
ifconfig tap0 0.0.0.0 promisc up mtu <mtuval>
brctl addif group1 tap0
openvpn --config  /etc/myconfig.ovpn &
ip route add to <subnet>/24 table 101 dev tun0
```

if needed, `anc fw add ls udp OVPN port port`

tommaso82, 2019-11-03:
I finally managed to be on-line via the d7000 using the on-board vpn client connection. My current configuration : d7000 set as access point, (cascading) 192.168.1.1 is my isp modem/router ip and 192.168.1.155 is the d7000 ip.

**_All that was missing was a simple a masquerade rule:_**

`iptables -t nat -A POSTROUTING -s 192.168.1.1/24 -j MASQUERADE`

The only big problem now is that I can't use dhcp to connect otherwise I get (from the isp / router modem of my isp) 192.168.1.1 as default gw instead of 192.168.1.155 which is the ip of the d700 that I need to have as a gateway to take advantage of the on-board client vpn connection.

to be online with the vpn you have to configure the wifi/ethernet client by hand specifying ip / netmask / defaultgw and dns (my case: 192.168.1.x 255.255.255.0 192.168.1.155 1.1.1.1)

The only solution that comes to me is a dhcp server to push the correct gateway to the clients but at this point we will also have to use a different subnet (right now the d7000 is in the same subnet as the isp modem router)... or not?

However, this is the simple script I use for VPN client connection when the router is started using runlevel startup command in personal istruction from the ancistrus menu

```
#!/bin/sh
/usr/sbin/nvram set openvpn_enable=0
/usr/sbin/rc openvpn stop
/usr/sbin/insmod /lib/modules/tun.ko
/usr/sbin/openvpn --mktun --dev tun0
/usr/sbin/ifconfig tun0 0.0.0.0 promisc up
/usr/sbin/iptables -t nat -A POSTROUTING -s 192.168.1.1/24 -j MASQUERADE
/usr/sbin/sleep 55
/usr/sbin/openvpn --config /usr/openvpn/aruba.ovpn --dev tun0 --log /tmp/vpnx.log &

```

negan07, 2019-11-04:
good

`iptables -t nat -A POSTROUTING -s 192.168.1.1/24 -j MASQUERADE`

you're in ap_mode so the same thing should be done creating a virtual server rule on the wan router to D7000 local ip ?

for the dhcp relay client you should check ovpn client options, there should be something related

tommaso82, 2019-11-05:
i m in this situation
https://i.ibb.co/d6byXL3/RETE.jpg
I believe that the solution to get  dhcp server to work on the d7000 is to create a new natted subnet  (like 192.168.2.x) right?

As for the dhcp serer in particularI i tried to start it launching /usr/sbin/rc_app/rc_dhcpd (some kind of wrapper for /usr/sbin/udhcpd ?) but I can't change the configuration every time I modify it is overwritten...

negan07, 2019-11-05:
yes in fact you need to write it manually and run udhcpd manually out of rc script

otherwise create the `rc_dhcpd_post` script running another instance of udhcpd with the modified nat config file as arguments

negan07, 2019-11-05:
if I remember the ap_mode=1 completely disable internal dhcp so it can be run manually modifying parameters with own preferences

so trying to setup nvram variables like this:
```
anc nvram set lan_ipaddr 192.168.2.1 dhcp_start_ip 192.168.2.2 dhcp_end_ip 192.168.2.254
nvram commit
rc dhcpd start
```
should make it run but another interface should be created with 192.168.2.1 as local ip

tommaso82, 2019-11-06:
` ifconfig group1:0 192.168.2.1`
`iptables -t nat -A POSTROUTING -s 192.168.2.1/24 -j MASQUERADE`

then 

`anc nvram set lan_ipaddr 192.168.2.1 dhcp_start_ip 192.168.2.2 dhcp_end_ip 192.168.2.254`
`nvram commit`
`rc dhcpd start`

Now we have a fully functional system that assigns ip clients on the 192.168.2.x subnet

negan07, 2019-11-06:
hard path to follow I think

wifi clients should connect the specific D7000 ssid and then receiving correct dhcp assignement to join vpn
moving to isp ssid to avoid vpn

the same with ethernet devices connected to D7000

dhcp lease should be renewed at the 1st time attempt

tommaso82, 2019-11-06:
we need to have 2 separate subnets
to fix things we need to activate the wan port (I assume it is eth0?) and assign the group1 bridge a class of ip different from the wan ones, and then the routing must be fixed ...

> ` ifconfig group1:0 192.168.2.1`
> `iptables -t nat -A POSTROUTING -s 192.168.2.1/24 -j MASQUERADE`
> 
> then
> 
> `anc nvram set lan_ipaddr 192.168.2.1 dhcp_start_ip 192.168.2.2 dhcp_end_ip 192.168.2.254`
> `nvram commit`
> `rc dhcpd start`
> 
> we have a fully functional system that assigns ip to clients through dhcp on the 192.168.2.x subnet
> 
> but only if on the 192.168.1.1 subnet there is no dhcp server
> 
> we need to have 2 separate subnets
> to fix things we need to activate the wan port (I assume it is eth0?) and assign to the group1 bridge a class of ip different from the wan ones, and then the routing must be fixed ...

tommaso82, 2019-11-07:
FINALLY we did it, then to get it all working properly as I have put d7000 in modem / router mode and then

```
/usr/sbin/nvram set openvpn_enable=0
/usr/sbin/rc openvpn stop
/usr/sbin/insmod /lib/modules/tun.ko
/usr/sbin/openvpn --mktun --dev tun0
/usr/sbin/ifconfig tun0 0.0.0.0 promisc up
/usr/sbin/sleep (to be defined)
/usr/sbin/openvpn --config /usr/openvpn/config.ovpn --dev tun0 --log /tmp/vpnx.log &
/usr/sbin/sleep (to be defined)
/usr/sbin/echo "2 vpn" >> /etc/iproute2/rt_tables
/usr/sbin/ip route add default via 10.9.0.1 dev tun0 table vpn
/usr/sbin/ip route add 192.168.0.0/24 dev group1 table vpn
/usr/sbin/ip rule add from 192.168.0.0/24 lookup vpn prio 1000
/usr/sbin/iptables -t nat -A POSTROUTING -s 192.168.0.0/24 -o tun0 -j MASQUERADE

```
assuming that the d7000 lan ip is 192.168.0.1 and vpn gw is 10.9.0.1

I put all the commands in a bash script (to use with ancistrus personal istruction menu) but I'm having problems with the final part (relative to the correct routing of the 192.168.0.0 subnet)

manually entering commands works perfectly

negan07, 2019-11-07:
first of all good job

upload speed may be limited by the different iptables traffic rules or routing a direct test should be needed to see real effects, unfortunately, not possible

it should be interesting to see what happen connecting D7000 directly to wan

dos protection can be disabled as well from advanced wan menu

tommaso82, 2019-11-07:
i erased all iptables rules but upload problem still persist...
next test is to get this working in ap mode only...
and then when you have some time i can give you a test account on my vpn to let you make 
some test...maybe a web interface to manage this stuff would be great

negan07, 2019-11-07:
the idea is to create an `rc_openvpn_client` script including your own shell code with the nvram settings addition and other parameters, then a webgui sub-page like readyshare webpage where parameters can be setup & enable

openvpn as client and orig server one can co-exist but the best way should be to make it run from wan conn directly

tommaso82, 2019-11-13:
the only problem currently is the upload bandwidth locked to less than a mega, do you have any way to check doing some tests?

negan07, 2019-11-13:
directly, not
but there's a variable, the isp router, can't be checked
so a test could be with `D7000<---->WAN<------>VPNSERVER`

minematte, 2020-03-27:
Hi, 
I tried to follow your discussion but when I try to connect to the VPN server (router is acting like a client), I can't authenticate. Is there any solution? Thanks.

tommaso82, 2020-03-27:
if router is acting like client you dont need windows client
explain yourself better

minematte, 2020-03-27:
> if router is acting like client you dont need windows client
> explain yourself better

I want my router tunneling all traffic through VPN, I've added the VPN server address but I can't authenticate

tommaso82, 2020-03-27:
post your logs
openvpn --log /tmp/vpn.log

minematte, 2020-03-28:
`
Fri Mar 27 18:46:53 2020 us=21285 Current Parameter Settings:

Fri Mar 27 18:46:53 2020 us=21425   config = '/etc/server.conf'
Fri Mar 27 18:46:53 2020 us=21480   mode = 1
Fri Mar 27 18:46:53 2020 us=21535   persist_config = DISABLED
Fri Mar 27 18:46:53 2020 us=21688   persist_mode = 1
Fri Mar 27 18:46:53 2020 us=21743 NOTE: --mute triggered...
Fri Mar 27 18:46:53 2020 us=21821 229 variation(s) on previous 5 message(s) suppressed by --mute
Fri Mar 27 18:46:53 2020 us=21878 OpenVPN 2.4.8 arm-unknown-linux-uclibcgnueabi [SSL (OpenSSL)] [LZO] [EPOLL] [MH/PKTINFO] [AEAD]
Fri Mar 27 18:46:53 2020 us=21972 library versions: OpenSSL 1.0.2t  10 Sep 2019, LZO 2.10
Fri Mar 27 18:46:53 2020 us=22255 NOTE: when bridging your LAN adapter with the TAP adapter, note that the new bridge adapter will often take on its own IP address that is different from what the LAN adapter was previously set to
Fri Mar 27 18:46:53 2020 us=24053 Diffie-Hellman initialized with 4096 bit key
Fri Mar 27 18:46:53 2020 us=25194 Current Parameter Settings:
Fri Mar 27 18:46:53 2020 us=25337   config = '/etc/server_phone.conf'
Fri Mar 27 18:46:53 2020 us=25392   mode = 1
Fri Mar 27 18:46:53 2020 us=25442   persist_config = DISABLED
Fri Mar 27 18:46:53 2020 us=25498   persist_mode = 1
Fri Mar 27 18:46:53 2020 us=25553 NOTE: --mute triggered...
Fri Mar 27 18:46:53 2020 us=25833 230 variation(s) on previous 5 message(s) suppressed by --mute
Fri Mar 27 18:46:53 2020 us=25900 OpenVPN 2.4.8 arm-unknown-linux-uclibcgnueabi [SSL (OpenSSL)] [LZO] [EPOLL] [MH/PKTINFO] [AEAD]
Fri Mar 27 18:46:53 2020 us=25991 library versions: OpenSSL 1.0.2t  10 Sep 2019, LZO 2.10
Fri Mar 27 18:46:53 2020 us=26659 TLS-Auth MTU parms [ L:1654 D:1212 EF:38 EB:0 ET:0 EL:3 ]
Fri Mar 27 18:46:53 2020 us=27132 TUN/TAP device tap0 opened
Fri Mar 27 18:46:53 2020 us=27226 TUN/TAP TX queue length set to 100
Fri Mar 27 18:46:53 2020 us=27413 Data Channel MTU parms [ L:1654 D:1450 EF:122 EB:411 ET:32 EL:3 AF:3/1 ]
Fri Mar 27 18:46:53 2020 us=27481 Could not determine IPv4/IPv6 protocol. Using AF_INET6
Fri Mar 27 18:46:53 2020 us=27564 Socket Buffers: R=[188416->188416] S=[188416->188416]
Fri Mar 27 18:46:53 2020 us=27656 setsockopt(IPV6_V6ONLY=0)
Fri Mar 27 18:46:53 2020 us=27778 UDPv6 link local (bound): [AF_INET6][undef]:12974
Fri Mar 27 18:46:53 2020 us=27844 UDPv6 link remote: [AF_UNSPEC]
Fri Mar 27 18:46:53 2020 us=27866 Diffie-Hellman initialized with 4096 bit key
Fri Mar 27 18:46:53 2020 us=27914 MULTI: multi_init called, r=256 v=256
Fri Mar 27 18:46:53 2020 us=28049 Initialization Sequence Completed
Fri Mar 27 18:46:53 2020 us=30151 TLS-Auth MTU parms [ L:1622 D:1212 EF:38 EB:0 ET:0 EL:3 ]
Fri Mar 27 18:46:53 2020 us=31057 TUN/TAP device tun0 opened
Fri Mar 27 18:46:53 2020 us=31151 TUN/TAP TX queue length set to 100
Fri Mar 27 18:46:53 2020 us=31240 do_ifconfig, tt->did_ifconfig_ipv6_setup=0
Fri Mar 27 18:46:53 2020 us=31338 /bin/ip link set dev tun0 up mtu 1500
Fri Mar 27 18:46:53 2020 us=35006 /bin/ip addr add dev tun0 192.168.2.1/24 broadcast 192.168.2.255
Fri Mar 27 18:46:53 2020 us=38667 Data Channel MTU parms [ L:1622 D:1450 EF:122 EB:406 ET:0 EL:3 AF:3/1 ]
Fri Mar 27 18:46:53 2020 us=38774 Could not determine IPv4/IPv6 protocol. Using AF_INET6
Fri Mar 27 18:46:53 2020 us=38862 Socket Buffers: R=[188416->188416] S=[188416->188416]
Fri Mar 27 18:46:53 2020 us=38908 setsockopt(IPV6_V6ONLY=0)
Fri Mar 27 18:46:53 2020 us=39021 UDPv6 link local (bound): [AF_INET6][undef]:12973
Fri Mar 27 18:46:53 2020 us=39076 UDPv6 link remote: [AF_UNSPEC]
Fri Mar 27 18:46:53 2020 us=39138 MULTI: multi_init called, r=256 v=256
Fri Mar 27 18:46:53 2020 us=39263 IFCONFIG POOL: base=192.168.2.2 size=252, ipv6=0
Fri Mar 27 18:46:53 2020 us=39386 Initialization Sequence Completed
Fri Mar 27 18:48:52 2020 us=93897 event_wait : Interrupted system call (code=4)
Fri Mar 27 18:48:52 2020 us=94003 event_wait : Interrupted system call (code=4)
Fri Mar 27 18:48:52 2020 us=94636 TCP/UDP: Closing socket
Fri Mar 27 18:48:52 2020 us=94743 TCP/UDP: Closing socket
Fri Mar 27 18:48:52 2020 us=94792 Closing TUN/TAP interface
Fri Mar 27 18:48:52 2020 us=94869 Closing TUN/TAP interface
Fri Mar 27 18:48:52 2020 us=94983 /bin/ip addr del dev tun0 192.168.2.1/24
Fri Mar 27 18:48:52 2020 us=95563 SIGTERM[hard,] received, process exiting
Fri Mar 27 18:48:52 2020 us=109229 SIGTERM[hard,] received, process exiting
Options error: Unrecognized option or missing or extra parameter(s) in /etc/server.conf:24: it66.nordvpn.com (2.4.8)
Use --help for more information.
Options error: Unrecognized option or missing or extra parameter(s) in /etc/server_phone.conf:24: it66.nordvpn.com (2.4.8)
Use --help for more information.
Options error: Unrecognized option or missing or extra parameter(s) in /etc/server.conf:24: it66.nordvpn.com (2.4.8)
Use --help for more information.
Options error: Unrecognized option or missing or extra parameter(s) in /etc/server_phone.conf:24: it66.nordvpn.com (2.4.8)
Use --help for more information.
Options error: Unrecognized option or missing or extra parameter(s) in /etc/server.conf:24: it66.nordvpn.com (2.4.8)
Use --help for more information.
Options error: Unrecognized option or missing or extra parameter(s) in /etc/server_phone.conf:24: it66.nordvpn.com (2.4.8)
Use --help for more information.
`

negan07, 2020-12-05:
any news about this ?

closed 2021-05-20 by negan07

tommaso82, 2021-06-25:
vpn is working... next step is including wireguard... client/server...