negan07/ancistrus #54: D7000 SMB v2 support for USB devices

tomtomek9, 2019-06-21:
WOuld it be possible to have a look into enabling SMBv2 support for readyshare USB storage option for the Netgear D7000?

negan07, 2019-10-10:
the samba version used is very customized 3.0.22
isn't it already enabled as default ?

tomtomek9, 2019-10-10:
Can you check if you can successfully share an usb storage?

Regards,
Tom

negan07, 2019-10-11:
I actually have 2 storage usb devices in the 2 usb ports locally shared between win ios android unix systems without any problems

smbv2 should be already enabled (not the same for smbv3 maybe due to the old version)

you could try to enforce the min version editing `/etc/samba/smb.conf` adding under `[global]` stage the option:
`min version SMBV2`

then save the conf kill and re run manually smb daemon:
```
killall -9 smbd
killall -9 nmbd
smbd -D -P 19
nmbd -D -P 19
```

tomtomek9, 2019-10-13:
I might be doing something wrong but with Telnet connection I cannot see
samba under /etc. When trying to connect via WinSCP I am getting access
denied. Maybe I am just doing something 'lame' but please assist if
possible.

negan07, 2019-10-13:
can't telnet with scp, just ssh
samba conf is under `/tmp/etc/samba/smb.conf`
`/etc` is a symlink of `/tmp/etc` mounted as ramdisk

tomtomek9, 2019-10-14:
When I telnet with putty there is nothing under samba folder. I cannot
connect via SSH (putty or WinSCP) as it is giving me access denied.

tomtomek9, 2019-10-14:
Found it under /etc/samba.conf/smb.conf

negan07, 2019-10-14:
`/etc/samba.conf/smb.conf` is the default one

the onecreated from the readyshare usb webpage options is under `/tmp/etc/samba/smb.conf`
samba service must be enabled before

negan07, 2019-10-17:
any news about ?

tomtomek9, 2019-10-25:
I have made the changes to the config. It still does not work with the suggested amendments. Thing is, my systems have SMBv1 disabled by default. Any other suggestions?

negan07, 2019-10-25:
router's samba should work with v2 without problems

try to setup on router advanced menu readyshare and look if the disk is mounted properly with related shared dirs

besides the first method try also to enable http access and see if it's visible

tomtomek9, 2019-10-29:
I have set this up on the advanced menu and I can browse to the USB through the router interface, when selecting folders to share.
I can also access the share via http.

Still, with system with disabled SMBv1 I cannot browse to it.

tomtomek9, 2019-10-29:
I believe on newer routers the SMB v2 is an option to be enabled form the advanced menu. Nevertheless, the issue seems to be with Netbios and Network Discovery. More details here:
https://community.netgear.com/t5/Nighthawk-WiFi-Routers/SMB2-Support-for-R7300-Nighthawk-Readyshare/m-p/1659948/highlight/true

negan07, 2019-10-29:
upgrading/changing samba inside the router will bring non desirable effects (web gui, genie, cloud, etc strictly depend on it)

the safest thing to do is: running smbd & nmbd manually with different configuration like `min version SMBV2`

or trying temporary to run an updated recompiled version to test the differences

tomtomek9, 2019-10-30:
Min version does not resolve this issue, as I have already stated earlier.  Would there be a chance to have the image built with samba code of another router that has had it updated as this is visible here:
[image]
https://community.netgear.com/t5/General-WiFi-Routers-Non/ReadyShare-SMB1-Error/m-p/1484250/highlight/true

negan07, 2019-11-01:
R9000 has samba branch version 4 installed while D7000 has branch version 3 this is probably the difference

last fw source are available if it shares the same genie interface maybe samba could have a chance to be exported without losing compatibility, but I don't believe it..

can try to compile a 3.6.25 (lede patch fixes) and running manually to see if it fixes the issues related
the same for 4.9.13 from openwrt

tomtomek9, 2019-11-02:
If you could implement the newer samba version for testing that would be great!

Regards,
Tom

negan07, 2019-11-04:
to mantain retrocompatibility with sercomm/netgear devs specifics the newest possible version is 3.6.25

moving to 4 means to break this compatibility 
[samba-3.0.22-ng-diff.zip](https://github.com/negan07/ancistrus/files/3806355/samba-3.0.22-ng-diff.zip)

here's the diff code between orig samba 3.0.22 and netgear 3.0.22

tomtomek9, 2019-11-05:
According to release notes, 3.6.25 supports SMB v2 so that should be sufficient. Do you think you will be able to compile the image with the new version?

negan07, 2019-11-05:
yes it requires some work & test but it should

strange is that 3.0.22 actually doesn't..

negan07, 2019-11-06:
here's a 3.6.25 `smbd `& `nmbd` binary recomplied

[smb3.6.25.zip](https://github.com/negan07/ancistrus/files/3817084/smb3.6.25.zip)

they're untouched without patches & not tiny in size so test them carefully from the `/tmp` dir with simplest storage configuration possible without making config changes meanwhile

```
killall -9 nmbd
killall -9 smbd
/tmp/nmbd -D -P 19
/tmp/smbd -D -P 19

```

revert back to original with
```
killall -9 smbd
killall -9 nmbd
rc smb restart
```

tomtomek9, 2019-11-07:
I am getting permission denied on trying to run from /tmp. Anything I need to do to allow it? Logged in as root via ssh.

negan07, 2019-11-07:
`chmod 755 /tmp/*mbd`

tomtomek9, 2019-11-07:
Thanks. Now I am getting the below:
/tmp/nmbd: can't load library 'libtalloc.so.2'

negan07, 2019-11-07:
[sambaso.zip](https://github.com/negan07/ancistrus/files/3821999/sambaso.zip)
here some shared libraries

put them on /tmp/lib then
`export LD_LIBRARY_PATH=/tmp/lib`

tomtomek9, 2019-11-07:
Now I am getting:
/tmp/smbd: '/tmp/lib/libtalloc.so.2' is not an ELF file
/tmp/smbd: can't load library 'libtalloc.so.2'

negan07, 2019-11-07:
does `nmbd` updated run ?

in this case try to run
`smbd -D -P 19`

tomtomek9, 2019-11-07:
nmbd produces same error
Let me check with changing permisions

negan07, 2019-11-07:
no way so it needs reconfig

negan07, 2019-11-15:
[smb3625.zip](https://github.com/negan07/ancistrus/files/3853482/smb3625.zip)
I've recompiled nmbd & smbd 3.6.25 with lede patches

not tiny in size so test them carefully from the `/tmp` dir with simplest storage configuration possible without making config changes meanwhile

/etc/samba/smb.conf must be edited setting up:
`dos charset = cp866
`in place of `dos charset = utf8`

now running:
```
killall -9 nmbd
killall -9 smbd
/tmp/nmbd -D -P 19
/tmp/smbd -D -P 19

```

revert back to original with
```
killall -9 smbd
killall -9 nmbd
rc smb restart
```

negan07, 2019-11-16:
with the updated samba version 3.6.25 & setting up `protocol = SMB2` on smb config samba Dialect has moved from 1.5 of original 3.0.22 to 2.0.2 satisfying the upgrade request

So the problem seems to be fixed: now `smbd` code needs required adaptations to match netgear's specs like lower cpu priority, passwd & directories, ...

Moreover a way must be found to reduce very huge binary code size (10-12Mb)...

tomtomek9, 2019-11-19:
Do you think you will be able to compile it with the firmware?

negan07, 2019-11-19:
with the actual sstripped size and removing libntpass & old binaries, there are up to 12mb more

as monolithic, rootfs partition space could be enough but it won't give more chances of package upgrades or modifications

anyway give a try to this temporary solution to test its stability (performances may improve because processes are not reniced but at the expense of resources)

tomtomek9, 2020-03-05:
Hi negan07. Do you think you will be able to compile a new firmware with the updated samba or will that be a manual process of replacing the old version?

negan07, 2020-12-05:
the idea was to create the .ipk package leaving it outside of monolithic due to space reasons as described

it can be installed manually later after firmware upgrade

negan07, 2020-12-16:
package created
see above

closed 2020-12-21 by negan07