negan07/ancistrus #55: Reboot schedule

Cams77, 2019-07-14:
Hi,
what commands should i insert in crontab to reboot router at specific time?
I tried with 2 different commands but none works on this router while on another openwrt router does.
First one is:
30 4 * * * root shutdown -r now
Second one is:
30 4 * * * sleep 70 && touch /etc/banner && reboot
Thanks for support.
P.S: using latest firmware my bufferbloat is 1ms both as vdsl modem and router only (other bridge modem)!!! Great work!!! Thanks.

Cams77, 2019-07-24:
Solved:
Solved: added cronjob
30 4 * * * root reboot
this works fine.

NOTES:
Using telnet i found out that shutdown command is not installed/recognised while reboot command is.
I have also tried without success
30 4 * * * root sleep 70 && touch /tmp/etc/version && reboot
even though sleep, touch and reboot commands are working fine.

negan07, 2019-10-10:
shutdown is not included into compiled bb (usually not needed)

closed 2019-10-11 by negan07