negan07/ancistrus #56: New Firmware 1.0.1.74

lentis1, 2019-07-22:
http://www.downloads.netgear.com/files/GDC/D7000/D7000_V1.0.1.74_1.0.1.zip

Cranial, 2019-07-24:
No source code available yet. Same as last update, might take a few days for Netgear to post it.

Cranial, 2019-07-24:
Contacted Netgear, the GPL Source is now up on: [https://kb.netgear.com/2649/NETGEAR-Open-Source-Code-for-Programmers-GPL](url)

or direct download link: [https://www.downloads.netgear.com/files/GPL/D7000-V1.0.1.74_1.0.1.tgz](url)

giga89, 2019-07-30:
I think the project is not followed anymore ...

lentis1, 2019-07-30:
> I think the project is not followed anymore ...

Even the past years during the summer there were no updates.
Usually resumes around October.

giga89, 2019-07-30:
> > I think the project is not followed anymore ...
> 
> Even the past years during the summer there were no updates.
> Usually resumes around October.

I hope that :)

m0lz, 2019-07-31:
@negan07 does your Ancistrus **need** rebasing on the latest official release ?

NetGear have not posted any public release notes, but apparently it fixes the old NULLs issue https://github.com/negan07/ancistrus/issues/50

.. But if thats all it does, we are better off using your builds which as far as I can tell due to NetGears lack of public information, your builds address more issues than the current official builds of the firmware.

joelstucki, 2019-10-10:
Any update on this? Is this project dead? I have upgraded to 1.0.1.74 and now my router crashes twice a day. I am now in the market for an alternative firmware build.

negan07, 2019-10-10:
> > I think the project is not followed anymore ...
> 
> Even the past years during the summer there were no updates.
> Usually resumes around October.

> @negan07 does your Ancistrus **need** rebasing on the latest official release ?
>
> NetGear have not posted any public release notes, but apparently it fixes the old NULLs issue #50
>
> .. But if thats [...]

[diff-V1.0.1.70-D7000_V1.0.1.74.zip](https://github.com/negan07/ancistrus/files/3714387/diff-V1.0.1.70-D7000_V1.0.1.74.zip)

fixes are security enhancements:

- web gui password check, recovery & reset
- buffer enlargements (hostnames & dhcp client length)
- mini_httpd url path check enhancements
- mini_httpd 002 patch "null_headers_malformed_fix" included (mini_httpd_1.17beta1-073_armD7000.ipk)

closed 2019-10-24 by negan07