negan07/ancistrus #57: email failed 500 unrecognized command

tominwi, 2019-07-26:
New to ancistrus having installed to my D7000 on 9-July, and every night since at 0100a when logs get emailed, the log shows many dozens of these:

[email failed] 500 unrecognized command Tuesday, July 09,2019 01:00:07

The log does get sent-out, so this is only a nuisance but I wonder if anyone here knows what about the ancistrus firmware might cause this. Of course, the Netgear firmware did not...

negan07, 2019-10-10:
what is the smtp config of the working email and the other not-working ?

tominwi, 2019-10-11:
The one email I have that works is a Torguard address with SMTP of mail.private-mail.com and port 465. But as I said I get dozens of those error messages in the log.

I've not been able to get any of my other email accounts to work with the Netgear, including all of Apple/icloud.com, smtp.mail.yahoo.com Port 465,  smtp-mail.outlook.com Port 587, smtp.aol.com Port 587, nor mobile.charter.net or mail.charter.net which use Port 587.

negan07, 2019-10-11:
just tried a gmail account with `smtp.gmail.com` port 465 ssl auth enabled full working without any error message

report all the log webpage config, webpage mail config and the kind of wan connection of the router

tominwi, 2019-10-14:
I thought it would be easier to just create & use a gmail account but that does not work at all for me. It seems my problem might have to do with my ISP but in any case since I am the only one having this problem, and the errors are only a nuisance anyway, I am going to leave well enough alone. Sorry for the distractio negan07.

negan07, 2019-10-15:
forgot to ask the firmware version even if it could be involved..

using `smtpc` from command line:

```
==========================================================
Usage: ./smtpc [m:s:f:r:h:p:U:P:cv] < files		
	-m mime type						
	-s subject						
	-f from addr (if NULL use recipient)			
	-r recipient						
	-h mail server							
	-p mail port (default=25)				
	-U user name (ESMTP)					
	-P password  (ESMTP)					
	-c Clear syslog					
	-v verbose   (DEBUG)					
========================================================
```

it should be possible to test the smtp email sender quickly and deeply to understand if it could be an ssl issue as it could be

tominwi, 2019-10-24:
As your prior post was Greek to me, I defer to my earlier statement "Sorry for the distraction negan07". Do not hold-up a new release over this.

closed 2019-10-24 by negan07