negan07/ancistrus #58: Check NTFS dick at the start

giga89, 2019-07-30:
How can I scan the usb-connected disk when the modem is rebooted?

negan07, 2019-10-10:
https://github.com/negan07/ancistrus/wiki/Personal-instructions#s-runlevel

- core setup
- personal instr
- Type Personal RunLevel Boot Stage Startup Command here: `/usr/sbin/rc_app/rc_e2fsck start /dev/sda1`
- add runlevel startup

supposing `/dev/sda1` the usb disk device

negan07, 2019-10-11:
forgot to remember that the rc_e2fsck doesn't fork to the background (even if this should not affect the machine boot stage effectively) and that a restart of the ntfs disk device could be recommended at the end so a usb device reconfig cmd should be added (see above)

giga89, 2019-10-13:
Thank you! You are super!

negan07, 2019-10-26:
reopened this issue because It was wrongly related to an ntfs disk device while `e2fsck` is ext2/3/4 fs related only

to monitor an ntfs disk health enable `smartd` (smartmontools)
otherwise to check an ntfs partition there is `ntfs-3g` ( https://www.tuxera.com/community/open-source-ntfs-3g/ )

closed 2019-10-27 by negan07