negan07/ancistrus #59: Can we have some screen shots please

MikeWilliams-UK, 2019-09-26:
It would be really nice if you could supply some screen shots in the GitHub WiKi

negan07, 2019-10-10:
In which wiki page(s) do you need screenshot usages ?

MikeWilliams-UK, 2019-10-10:
I think there ought to be a separate wiki page showing the web UI pages.

I have seen a demo of the DD-WRT pages at http://www.informatione.gmxhome.de/DDWRT/Standard/V23final/index-2.html

I appreciate that a full simulation might be out of the question, but the basics would be useful.

negan07, 2019-10-10:
the same thing is already present in a mostly simplified way

each web gui option is symlinked to the related wiki board chapter/paragraph simply clicking it you're reminded to the related paragraph

integrating it in the internal router bottom help frame creates some disadvantages like avoiding dynamic updating/fixing and making firmware web pages heavier

MikeWilliams-UK, 2019-10-10:
Maybe I am being think here, but I can find any screen shots in the wiki, all I can find is lots of command line code examples.

negan07, 2019-10-11:
some wiki pages were created at the time of cmd line only stuffs: the integration is simple following the linked paragraph

if something specific not clear, please specify the related wiki page