negan07/ancistrus #60: VLAN Tagging Wifi

Cranial, 2019-10-11:
Hi Negan, looking to see if you can help.

I'm trying to figure out if I can vlan tag the Guest WiFi on the D7000. Currently it's setup as an Access Point, connected to my Ubiquiti USG LAN 1. I want to tag Guest network so I can pass that to the guest portal on my controller.

Looking on the D7000, I can see a utility **vlanctl**, but haven't found any available documentation online.

```
root@D7000:~$ vlanctl --help

VLAN Control Utility:

::: Usage:

vlanctl

        --if <if_name> Sets the target Interface of a composite vlanctl command to <if_name>.

        --rx Sets the direction of a composite vlanctl command to RECEIVE

        --tx Sets the direction of a composite vlanctl command to TRANSMIT

        --tags <nbr_of_tags> Sets the number of tags of a composite vlanctl command to <nbr_of_tags>

        --if-create <real_if_name> <if_index> Creates a new VOPI named <real_if_name>.v<if_index> and attaches it to the real device
         <real_if_name>. For instance, if this command were executed for the eth0 real interface and the VOPI interface index were
         set to 3, the resulting interface would have been named eth0.v3.

        --if-create-name <real_if_name> <vlan_if_name> Creates a new VOPI named <vlan_if_name> and attaches it to the real device

        --if-delete <vlan_if_name> Destroy the VOPI named <vlan_if_name>.

        --rule-append Inserts a new Tagging Rule as the last rule of the specified Tagging Rule Table. Dependencies: --if, --rx or
         --tx, and --tags.

        --rule-insert-before <rule-id> Inserts a new Tagging Rule before the Tagging Rule whose identifier matches <rule-id> in the
        specified Tagging Rule Table. Dependencies: --if, --rx or --tx, and --tags.

        --rule-insert-after <rule-id> Inserts a new Tagging Rule after the Tagging Rule whose identifier matches <rule-id> in the
        specified Tagging Rule Table. Dependencies: --if, --rx or --tx, and --tags.

        --rule-remove <rule-id> Removes the Tagging Rule that matches <rule-id> from the specified Tagging Rule Table. Dependencies:
        --if, --rx or --tx, and --tags.

        --rule-remove-all <real_if_name> <vlan_if_name> Removes all the Tagging Rules for the vlan device. 

        --show-table Lists all Tagging Rules stored in the specified Tagging Rule Table. Dependencies: --if, --rx or --tx, and
        --tags.

        --default-tpid <tpid> Sets the default TPID value of a tagging rule table to <tpid>. When a table is created, its default
        TPID value is set to 0x8100. Dependencies: --if, --rx or --tx, and --tags.

        --default-pbits <pbits> Sets the default PBITS value of a tagging rule table to <pbits>. When a table is created, its 
        default PBITS value is set to 0. Dependencies: --if, --rx or --tx, and --tags.

        --default-cfi <cfi> Sets the default CFI value of a tagging rule table to <cfi>. When a table is created, its default CFI 
        value is set to 0. Dependencies: --if, --rx or --tx, and --tags.

        --default-vid <vid> Sets the default VID value of a tagging rule table to <vid>. When a table is created, its default VID 
        value is set to 1 (as per IEEE 802.1Q). Dependencies: --if, --rx or --tx, and --tags.

        --cfg-dscp2pbits <dscp> <pbits> Programs the entry number <dscp> of the DSCP-TO-PBITS translation table of a Real Device to
        the value specified by <pbits>. When a tagging rule table is created, the default values of the DSCP-TO-PBITS table are 
        set by copying the lowest 3 bits of each DSCP value as the PBITS value, for each entry in the table. For instance, the 
        following entries are programmed by default: DSCP=5:PBITS=5, DSCP=15:PBITS=7, etc. The DSCP-TO-PBITS translation table 
        has 64 entries. Dependencies: --if.

        --show-dscp2pbits Lists the values programmed in the DSCP-TO-PBITS table of the specified Real Device. Dependencies: 
        --if.

        --cfg-tpid <tpid0> <tpid1> <tpid2> <tpid3> Configures the TPID Table entries of a given Real Interface. The configured 
[... 203 lines cut ...]
          RX packets:29833381 errors:0 dropped:15 overruns:0 frame:33465316
          TX packets:29911759 errors:4514 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:4168385661 (3.8 GiB)  TX bytes:353953371 (337.5 MiB)
          Interrupt:92 

wl0.1     Link encap:Ethernet  HWaddr 62:5D:82:D8:24:C9  
          inet6 addr: fe80::605d:82ff:X:X/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:33465316
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

wl1       Link encap:Ethernet  HWaddr 40:5D:82:D8:24:CA  
          inet6 addr: fe80::425d:82ff:X:X/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:127636241 errors:0 dropped:1 overruns:0 frame:31205843
          TX packets:324202192 errors:51244 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:671185127 (640.0 MiB)  TX bytes:1335835440 (1.2 GiB)
          Interrupt:93
```

I'm not sure if this is possible at all, with the native vlanctl package or with an external package, but thought you would know more about this than me.

negan07, 2019-10-11:
have a look at https://github.com/RMerl/asuswrt-merlin.ng/blob/master/release/src/router/vlanctl/unitTest/vlantest should share the same broadcom (?) utility

have also a look at VLAN/IPTV config hidden web page under advanced/install menu

Cranial, 2019-10-11:
>have a look at https://github.com/RMerl/asuswrt-merlin.ng/blob/master/release/src/router/vlanctl/unitTest/vlantest should share the same broadcom (?) utility ...

@negan07 thanks Negan, I’ll check it out. If I have any success I’ll update here for anyone else it might help. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

negan07, 2019-10-24:
any other improvement ?

Cranial, 2019-10-25:
No chance to work on it yet, been travelling for work again. Will update when I can.

closed 2020-12-05 by negan07

prophecy4you, 2023-12-13:
The hidden iptv page did the trick to enable the VLAN (and trunking), just had to manually add the guest network to the br created in the hidden page 👍