negan07/ancistrus #62: Ancistrus #605 - Occasional lock up requiring router reboot

m0lz, 2019-11-24:
This has happened a couple of times since installing #605

The previous version #592 was always stable / reliable

Symptom : Router locks up no internet access, solid lights.

The only solution is to reboot the router .. Then it behaves well for a few weeks, and then it happens again seemingly randomly.

Anyone else getting this with #605 ?

negan07, 2019-11-24:
updates from #592 to #54 605 are few, only package upgrades

the same result is obtained starting from #592 and upgrading all the packages

have settings & configurations being changed in the meantime ?

tominwi, 2019-11-24:
m0lz I updated to #605 on Oct 28 which was close to the day it was released and while I have had an ISP outage and a DNS outage since, I do not recall having to reboot due to lockup. I did before with #592 have a couple of lockups, maybe two in the 2-1/2 months that I had it installed.

m0lz, 2019-11-25:
> have settings, configurations & packages being changed/installed/renewed in the meantime ?

Hi @negan07 

None of those have been changed.
My settings remain the same as were used in the previous build.
And I have not manually changed or updated anything else.

The Ancistrus Menu with advanced options to disable elements are the same as default

m0lz, 2019-11-26:
I cant find anything historically that would tie in as an ISP outage in my area, there was an outage about fifty miles away yesterday but the time of that outage does not coincide.

Will try to get more information from the router logs if it happens again, technically I am not too knowledgeable with these matters but will try to get more clues as to what may be going wrong over the coming weeks.

m0lz, 2019-12-22:
Hi negan, please close this issue, I have not been able to reproduce it since my initial report, and your firmware has been performing admirably without any issues. Sorry to take up anyones time unnecessarily.

closed 2019-12-22 by m0lz