negan07/ancistrus #63: OpenVPN issues - No server certification method has been enabled

cryresta, 2019-12-02:
Hello Negan,
I am writing to you because i'm encountering an issue with my remote access VPN since I updated the last version of ancistrus #605.

After the update to the last firmware version I have tried many times to connect leaving the default configuration provided by the router, but I'm getting this error code: 

>  _WARNING: No server certificate verification method has been enabled._

It keeps the "MANAGEMENT:> STATE" and After 60 seconds I get:

>  _Mon Dec 02 22:30:28 2019 TLS Error: TLS key negotiation failed to occur within 60 seconds (check your network connectivity)
Mon Dec 02 22:30:28 2019 TLS Error: TLS handshake failed_

Obviously my dynDNS is updated and registered, I'm trying from different remote network but it seems there is nothing to do. What do you suggest? Before the update was working fine... All packages are updated and I have followed the usual instructions as "how to rename the TUN / TAP network card" and swapping the old configuration with the new one downloaded by the VPN section of the router etc.

I have read a lot of forums but I do not understand the root cause of the issue, because teorically the certificate is present on the configuration... I need your help What is missing?
 
My configuration:

_client
dev tap
proto udp
dev-node NETGEAR-VPN
remote example.com 12974
resolv-retry infinite
nobind
persist-key
persist-tun
ca ca.crt
cert client.crt
key client.key
cipher AES-256-CBC
comp-lzo
verb 5_

alemol93, 2020-04-07:
add "remote-cert-tls server"

cryresta, 2020-04-13:
Hello, I tried to add what you suggest and I managed to remove the error string. But unfortunately VPN is still not working, the only way with which it runs it's when I try to connect through the same internal network to the VPN.
All other attempts from external network keep this string : "MANAGEMENT_STATE> x.x.x.x.x CONNECT".
I tried also with TCP protocol (thinking about some ISP restriction of UDP) but nothing to do, it is always on "MANAGEMENT STATE > ...". When I had the previous versions of Negan the VPN alsways worked perfectly, since I have upgraded to the #605 i was no longer able to make it work.

Any suggestion would be very helpful for me ;)

alemol93, 2020-04-13:
Hi, glad to hear you solved some errors! I suggest you to check also the server config file named as "PC server" under "show config section" in ANCISTRUS/Management/openvpn.
Heading back to the client config file: "remote example.com 12974" this is why it works in local and not in remote. If you are using Dynamic DNS change "example.com" with your personal DDNS host address, if not then change it with IP address given by your Internet Service Provider.

cryresta, 2020-04-20:
Hello alemol, thanks for the follow up. 
Obviously I set up DDNS with the correct addressing (not "example.com" :D) and I noticed that from the Internal network it works because the router resolves my ddns locally (it has an entry at which he responds locally based on the DDNS account responses).
While if I launch OpenVPN from an external network (for example connected by hotspot to my mobile's LTE network it seems to be unable to establish the connection with the IP, even if resolves correctly the dynamic DNS.
I have tried with both the UDP and TCP versions but there seems to be nothing to do. He gets stuck on "MANAGEMENT_STATE" and after the timer expires he makes another attempt.
I am sorry because before the update to # 605 it worked perfectly ... I have no idea what could have changed.

I also checked the "Pc Server" configuration but it seems to be correct ... I don't know what to do anymore ...

I attach the "Pc Server" config (currently set on TCP mode):

[image]

Thanks in advance for any other suggestion.

alemol93, 2020-04-21:
Hello cryresta!
  Unfortunately i'm stuck at this point too. Idk how it was with previous builds because #605 is my first ancistrus test but it used to work a month ago (local only) and now everything is broken (no local neither remote).
For what i know it seems to be a client-server connection problem:
https://openvpn.net/community-resources/how-to/#starting-up-the-vpn-and-testing-for-initial-connectivity

closed 2020-12-05 by negan07