negan07/ancistrus #64: Ancistrus #605

MinoMuratori, 2019-12-04:
Hi all, Hi Negan,
first of all I want to apologize for my bad English
I've been using monilitich firmware for years, today I updated to version 605 and I found a couple of small problems:
1 - the old version still appears on the ancistrus page;
2 - I updated the packages too, but continues to offer them for update.
I have attached a couple of screenshots to be clearer.

Thanks for the support and the wonderful job you do!

[image]
[image]

pantelis86, 2019-12-04:
clear your history of the browser

MinoMuratori, 2019-12-05:
> clear your history of the browser

It worked!
Thank you so much for the quick reply, and sorry for the stupid question.

closed 2020-12-05 by negan07