negan07/ancistrus #66: Question: QoS

netgear603, 2020-04-09:
First - apologies for being a total novice.

If the NetGear is acting as wireless router only (DSL connection is provided by separate router hooked to NetGear 10/100 WAN port), is QoS applicable? Can it be set for the Ethernet connection only?

When I do a "rc_qos details", I get a "Cannot Find Device "ppp1"", and any attempt to set QoS service to "True Ethernet" results in the QOS SQM screen reverting to the ADSL (atm) setting.

Basically - have very low speed DSL (max 3 Mps down/.9 Mbps up), and want to do buffering at the WAN (eth1?) interface of the Netgear D7000 (which is hooked to upstream DSL modem/router ... (an Actiontec wn784 - for what it's worth). 

Thanks, and please excuse ignorance.

netgear603, 2020-04-09:
Sorry - throttling (not buffering) at WAN interface.

netgear603, 2020-04-09:
Note - I see this is similar to a closed issue ... but I am not sure if configuration is the same. I have my DSL modem set with WiFi off (so essentially acting as a bridge), then hooked up from Eth on the DSL modem to the NetGear. However, as far as I know, the DSL modem is NOT set to RFC1483 mode (or whatever bridge mode is). It has set up the PPPoE link, and passing traffic I assume via normal Ethernet to the NetGear - which is acting as the wireless hub. The netgear- as I said - does not see any PPP1 connection...

negan07, 2020-12-05:
qos acts on output interface only
router must be used as primary otherwise as secondary wan ethernet router

closed 2020-12-16 by negan07