negan07/ancistrus #68: New Firmware 1.0.1.78

mcrs969, 2020-04-17:
Hello Negan07,
 
I follow you since the beginning of this project and I have to say your FW made my D7000 an even better device!

Herewith an issue as NETGEAR released the FW 1.0.1.78 for D7000. 

Find here download link:
http://www.downloads.netgear.com/files/GDC/D7000/D7000_FW_V1.0.1.78_1.0.1.zip
and here release notes:
https://kb.netgear.com/000061832/D7000-Firmware-Version-1-0-1-78

from such notes it seems mainly FW security enhancements rather than performance optimizations. 
Do you think your version would be compatible?

Matt-Mf, 2020-04-20:
New GPL Source from NETGEAR are up: https://www.downloads.netgear.com/files/GPL/D7000v1_V1.0.1.78_1.0.1.tgz

tominwi, 2020-05-08:
It seems to me it's been said in the past that negan07 only works this project aperiodically at best. @negan07 can you give us any indication of when you might update your ancistrus builds with this (seemingly extensive) new release from Netgear? Thank you.

Matt-Mf, 2020-05-26:
#72

negan07, 2020-12-05:
[diff-D7000_V1.0.1.74_78.zip](https://github.com/negan07/ancistrus/files/5647133/diff-D7000_V1.0.1.74_78.zip)

```
    [Security Fixes] PSV-2018-0318 Weak Password Policy

    [Security Fixes] PSV-2019-0010 Cross-Site Scripting in "block site" Configuration

    [Security Fixes] PSV-2019-0012 /www/adv_index.htm Exposed Unauthenticated

    [Security Fixes] PSV-2019-0013 Cross-Site Scripting in /www/adv_index.htm

    [Security Fixes] PSV-2019-0014 Command Injection in PPPOE Functionality

    [Security Fixes] PSV-2019-0015 Cross-Site Scripting in "remote management" Configuration

    [Security Fixes] PSV-2019-0016/0018 Setup Actions Permitted Unauthenticated

    [Security Fixes] PSV-2019-0109 mini_httpd Authentication Bypass Vulnerability

    [Security Fixes] PSV-2019-0110/0184 Password Storage Information

    [Security Fixes] PSV-2019-0113 Password change & debug mode

    [Security Fixes] PSV-2019-0021 "Zombie POODLE" and "GOLDENDOODLE"

    [Security Fixes] PSV-2019-0022 Remote code execution

    [Security Fixes] PSV-2019-0192/0244 Remote code execution

    [Security Fixes] PSV-2019-0193/0245 Cross Site Scripting (XSS) in IPv6 Autoconfig settings

    [Security Fixes] PSV-2019-0170 Invalid CPE Certificate for Remote Access

    [Security Fixes] PSV-2019-0155 httpd lan_ipaddr stack overflow vulnerability

    [Security Fixes] PSV-2019-0145 httpd friendly_name stack overflow vulnerability

    [Security Fixes] PSV-2019-0124 NVRAM configuration injection caused by "SetNTP" parameter of SOAP "DeviceConfig-Set

    [Security Fixes] PSV-2019-0140 NVRAM configuration injection caused by "New5GCTSRTSThreshold" parameter of SOAP "WLANConfiguration-SetAdvancedW

    [Security Fixes] PSV-2019-0141 NVRAM configuration injection caused by "NewCTSRTSThreshold" parameter of SOAP "WLAN 
```

updated openssl to 1.0.2h, still old but fresher than 1.0.0, but regressed label to 0.9.8 generating issues with packages compiled against it on the newer version

updated bb login with failsafe httpd measure

updated wlan pw policy

note: sources compiled from another machine system (ubuntu against debian) creating some compatibility issues when compiling (see more)

mcrs969, 2020-12-05:
Great to see you here. Thanks Negan!

closed 2020-12-18 by negan07