negan07/ancistrus #69: How to Install

ocelotrainbow, 2020-04-24:
Hi, can someone advise me how to install this custom firmware?

Can you also highlight what benifits it brings?

Thanks

Matt-Mf, 2020-04-24:
Install:
1. Go to releases section
2. From assets download latest D7000_ _ _ .zip
3 Extract .img
4 Then follow this using the extracted .img: [https://kb.netgear.com/23960/How-do-I-manually-update-the-firmware-on-my-NETGEAR-router](url)
5 Done!

Check the wiki section for finding more info about the project:[https://github.com/negan07/ancistrus/wiki/ancistrus-core-package](url)

Personally the drastic improvements regarding bufferbloat worth everything.

ocelotrainbow, 2020-04-24:
Thanks Matt!

How can i confirm i have D7000v1?   I checked on back of router and don't see any V2 written on it, so does that mean it's V1 yeah?

ocelotrainbow, 2020-04-24:
Thats amazing about the bufferbloat thats the main reason i was looking into custom firmware.  Do i need to enable QoS to get the reduced bufferbloat?

ocelotrainbow, 2020-04-24:
Thanks I just flashed the firmware.

Look at the difference!  Thanks so much!

Stock Firmware
[image]

Ancistrus with the new QOS-SQM turned on
[image]

CarlosC1975, 2020-05-07:
Hi ocelotrainbow. I just installed the firmware also. Looks good. 
Just wondering what settings you changes to get the the improvement. Can you share that file.

My router was working good before updating to 
78 original. Updated to this custom firmware and Its much better. But still need to tweak it a bit.

ocelotrainbow, 2020-05-08:
Hi Carlos

All i did was go into the *ANCISTRUS* tab then Networking > QOS SQM.
Turned QOS Service on and limited my uplink and downlink speeds by about
5-10% of their full speed.

Hope this helps!

Thanks

tominwi, 2020-05-23:
FWIW I downgraded from Netgear's official v78 to v74ancistrus and struggled rather mightily to get my settings to "take". I would click "Turn QOS Service On" checkbox but then Apply would just clear it again. I tried numerous times to no avail, and sometime later with another Firefox launch I finally got my settings to "stick". Dunno by what magic; maybe old v78 settings were interfering somehow, altho I was NOT using Netgear's QOS.

The good news is my DSLR speedtest's Bufferbloat went from an F grade to an A, and Overall I'm now A+.

closed 2020-12-05 by negan07