negan07/ancistrus #70: Port transmission-daemon from DDWRT to torrent directly into Readyshare USB devices

mkexc, 2020-05-04:
I'm writing this here as a suggestion for a possible new function to implement to this fantastic firmware.
I've searched the web but couldn't find the .ipk and try to install it by myself.
Thank you

negan07, 2020-12-05:
transmission can be added but with very low priority, because pretty invasive for the firmware structure, for the usb driver and also well supported by storage dedicated machines like synology

mkexc, 2020-12-23:
I'm a developer myself, I can try but I don't know where to start looking for developing for this platform. If you can give me any directions, I will look into it!

negan07, 2020-12-23:
start cloning the git and prepare machine for development (debian is recommended/mandatory)
follow the instructions on the wiki
https://github.com/negan07/ancistrus/wiki/Prepare-&-compile-sources
https://github.com/negan07/ancistrus/wiki/How-sources-are-organized
https://github.com/negan07/ancistrus/wiki/Create-a-new-work-app
https://github.com/negan07/ancistrus/wiki/Build-a-package

mkexc, 2021-01-20:
Thank you, I will start working on it after college's exams session.

negan07, 2021-03-28:
`Transmission` package is on the way

`opkg update && opkg install transmission`

Make sure to upgrade curl & core package before for best effort.

Linden1, 2021-03-29:
Thank you so much @negan07 Easy install and setup, working great.
Thank you, thank you, thank you!

closed 2021-03-31 by negan07