negan07/ancistrus #71: Web GUI doesn't load after install and router refresh

BigLep, 2020-05-13:
I had the latest Netgear firmware (1.0.1.78) installed and then ran the startup.sh script (https://github.com/negan07/ancistrus/wiki/startup.sh ).  At that point, the web GUI still loaded and I saw the new tab.  After rebooting the router though, I can't access the web UI.

I can still telnet into the router and fortunately the router/modem still works, but nothing is listening on port 192.168.0.1:80.  I don't see any http process when I do a ```ps | grep http```.

I didn't see any logs, but am not sure where best to look.

Does anyone know how to either:
1. Debug why the HTTP server isn't starting OR
2. Install the monolithic firmware .img via command line (to see if that works)?

I have spent a couple hours tonight poking around and looking at previous GitHub issues, and am not sure how to get unblocked.  

Thanks a lot
Steve

BigLep, 2020-05-15:
Are there any suggestions on this?  Does anyone have info on how mini_http is configured/started on the D7000 router?

I tried using tftp to flash my router (https://kb.netgear.com/000059634/How-to-upload-firmware-to-a-NETGEAR-router-using-Windows-TFTP ), but when I put it into the reset mode, I could never ping it (despite being connected with ethernet and manually configuring the IP address on my computer).  

Any pointers are welcome.  At this point I have a router with no web GUI.  I was working on this for a friend I have lived with the last two months to hopefully improve the reliability of his internet connectivity.  I don't feel good though leaving him with a router he can't administer without a GUI.

BigLep, 2020-05-15:
When I would manually attempt to invoke the httpd process (```/usr/sbin/mini_httpd```), I was getting a "couldn't find openssl.so" message about a specific version (I didn't write down the exact error message).  

I used opkg to install mini_httpd: ```opkg update && opkg install mini_httpd``` (per discussion in https://github.com/negan07/ancistrus/issues/50 ).

The httpd process is now running again!

The only issue I'm seeing now is that nothing renders under the ANCISTRUS > Management|Networking|Storage|General navigation links on the left-hand side.  This happens on latest Chrome and Firefox for me:

[image]

Anyways, I will attempt to just install the monolithic firmware (rather than running startup.sh on the 1.0.1.78 Netgear firmware and see if that presents all the options in the Web GUI.  I'll report back after.

BigLep, 2020-05-16:
I installed the monolithic firmware via the web GUI and I now see the items in the left nav loading.
[image]
I'll close and resolve the issue.  The key takeaway is that folks should install using the monolithic image rather than startup.sh.  I wish I had done that to start :)

closed 2020-05-16 by BigLep

negan07, 2020-12-05:
this is due to the regress to openssl 0.9.8 label (1.0.2 installed) from the producer developers
for this reason mini_httpd wasn't able to find the lib because missing the libssl.0.9.8 symlink

to solve, create links manually
```
cd /lib
ln -s libcrypto.so.1.0.0 libcrypto.so.0.9.8
ln -s libssl.so.1.0.0 libssl.so.0.9.8

```
or as a workaround:
`opkg install mini_httpd
`

anyway new packages fix the problem