negan07/ancistrus #72: Compiling new 1.0.1.78 Firmware

Matt-Mf, 2020-05-26:
Hi guys, i'm trying to compile ancistrus form the the new GPL sources to create a new .img at least for testing. 
I've setup a Kubuntu Virtuabox container and installed all the dependencies needed, but i got stuck to this point.

`freadahead.c: In function 'freadahead': 
freadahead.c:83:3: error: #error "Please port gnulib freadahead.c to your platform! Look at the definition of fflush, fread, ungetc on your system, then report this to bug-gnulib."`

[image]

Helps and ideas are appreciated because i'm still a noob in this field.

Matt-Mf, 2020-05-26:
After doing some investigation, i think that new GLIBC version (>2.24) is the problem.

Matt-Mf, 2020-05-26:
I've gone further but stuck with a LOCALE problem.
`extra/locale/gen_wc8bit.c: In function 'main':
extra/locale/gen_wc8bit.c:375:28: warning: array subscript is above array bounds
extra/locale/gen_wc8bit.c:434:29: warning: array subscript is above array bounds
extra/locale/gen_wc8bit.c:500:27: warning: array subscript is above array bounds`

Error is similar to these:

http://uclibc.10924.n7.nabble.com/not-x86-host-not-glibc-host-build-error-bootstrapping-uclibc-with-locale-support-is-it-possible-at-a-td5027.html

https://forum.archive.openwrt.org/viewtopic.php?id=53171

fcolasuonno, 2020-06-29:
I started to have a look at his one as well. I tried building V1.0.1.74 first on Ubuntu 20.04 LTS first
First `scripts/hostprepare.sh` to install some necessary packages.
For the "Please port gnulib freadahead.c to your platform" error replace `diffs/crosstools/D7000__crosstools-013-m4_various_fixes.diff` with 
```
diff -urN crosstools-gcc-4.6-linux-3.4-uclibc-0.9.32-binutils-2.21-sources/src/buildroot-2011.11/dl/m4-1.4.15/lib/stdio.in.h crosstools-gcc-4.6-linux-3.4-uclibc-0.9.32-binutils-2.21-sources/src/buildroot-2011.11/dl/m4-1.4.15/lib/stdio.in.h
--- crosstools-gcc-4.6-linux-3.4-uclibc-0.9.32-binutils-2.21-sources/src/buildroot-2011.11/dl/m4-1.4.15/lib/stdio.in.h  2010-07-29 15:21:37.000000000 +0200
+++ crosstools-gcc-4.6-linux-3.4-uclibc-0.9.32-binutils-2.21-sources/src/buildroot-2011.11/dl/m4-1.4.15/lib/stdio.in.h  2016-12-26 22:47:44.000000000 +0100
@@ -141,7 +141,15 @@
    so any use of gets warrants an unconditional warning.  Assume it is
    always declared, since it is required by C89.  */
 #undef gets
+#if defined(__GLIBC__) && !defined(__UCLIBC__) && !__GLIBC_PREREQ(2, 16)
 _GL_WARN_ON_USE (gets, "gets is a security hole - use fgets instead");
+#endif
+#if !defined _IO_ftrylockfile && defined _IO_EOF_SEEN
+# define _IO_ftrylockfile _IO_EOF_SEEN
+#endif
+#if !defined _IO_IN_BACKUP && defined _IO_EOF_SEEN
+# define _IO_IN_BACKUP 0x100
+#endif
 
 #if @GNULIB_FOPEN@
 # if @REPLACE_FOPEN@
```
That should allow building the crosstools, then the kernel.
When compiling the source, uClibc will complain that it cannot find libncursesw.so.5, so in `/lib32` I ran `sudo ln -s libncursesw.so.6 libncursesw.so.5` to try to link with libncursesw6.
Now I have an error compiling `library/libogg-1.3.0/src` where libtool cannot find the compiler.

Matt-Mf, 2020-06-29:
Thanks @fcolasuonno.
The only difference here is that i'm trying to build from the new V1.0.1.78 sources and not the old V1.0.1.74.
I've manually edited Settings.mak to pull the new sources in.

The last time i tried i was able to build succesfully crosstools and i've already resolved freadahead block, but then i got stuck building the kernel from the crosstools with the above _locale issue_.
Probably it's a bug or i'm missing something :/

Now i'm busy with other projects, but i will take another look asap.

gianx89, 2020-08-04:
@Matt-Mf any updates?

gianx89, 2020-11-08:
@Matt-Mf @fcolasuonno @negan07 any updates?

fcolasuonno, 2020-11-08:
Sorry, switched to an openwrt router

negan07, 2020-12-05:
https://github.com/negan07/ancistrus/wiki/Prepare-&-compile-sources

starting platform is wrong, need debian

closed 2020-12-17 by negan07

Matt-Mf, 2020-12-30:
Sorry @gianx89 but I also switched to another router. In the past i've tried Debian, but the results were the same. So not much resolved.
If i have fresh news i'll report back.