negan07/ancistrus #74: Ookla Built-in Speedtest functionality

negan07, 2020-12-06:
Has anyone tried to use the OOKLA speedtest present into the original netgear's qos config webpage gui recently ?
Same as from console:

`speedtest
`

```
root@D7000:/usr/etc$ speedtest
Prepareing test data, it will take some time...
Getting the five nearest servers
download flag = 0
can not get client info, quit now

```
It seems to be down or unavailable

Linden1, 2020-12-08:
Same here too.

negan07, 2020-12-08:
With the network-test package servers sometimes unstable too, it's safer to add an alternative.

The ookla speedtest website furnishes a speedtest tool compiled for arm arch, tested and working as standalone, but with the expense of about 1,5mb of space

negan07, 2020-12-24:
Made a simple automate script for gui. Working from ramfs to save space.

closed 2020-12-26 by negan07