negan07/ancistrus #76: JDownloader

melidron, 2020-12-26:
Hello from greece and merry Xmas
Is there a way to make a working script or any other way so the D7000  renew the ip for continuous downloading through JDownloader software?
in the software's setting it recognise the router but there is any working method to make router renew its ip except manual way
Thanks a lot

negan07, 2020-12-26:
```
rc wan adslmode|atmqos 
rc wan auto devid
rc wan start|stop|restart [wanid] 
rc wan up|down wanid
```
`wanid` is mostly 1

`rc wan restart 1` should be the cmd
even if not always the dhcp lease is also renewed: there is always the possibility to obtain the same ip again
so some delay may help
```
rc wan stop 1
sleep 30
rc wan start 1
```
or
```
rc wan down 1
sleep 30
rc wan up 1
```
increasing delay if needed

Merry Christmas !

melidron, 2020-12-27:
Hello again and thank u for your quick respont
where do I paste all these commands? and what parts ?
inside jdownloader ? or is it a batch? or something else?
excuse my ignorance and thank u again my friend

negan07, 2020-12-27:
they are console command, usually used into telnet or ssh cmd line session

melidron, 2020-12-27:
how can i use them on JDownloader?
is it easy to explain more?

negan07, 2020-12-27:
the procedure to use console cmd is:

`telnet 192.168.0.1`
root
pw

`rc wan restart 1`
`exit` (ctrl+D)

I don't know the automation procedures of the software, but if the router is recognized the easiest way is through cmd line

melidron, 2020-12-27:
the reconection methods that JDownloader2 use in settings is !. External Batch Reconnect 2. External tool Reconnect 3. Liveheader,,,
Can u tell me if there is a way to make a working batch with the commands u told me? what should I paste inside this Batch file?
thank u again

negan07, 2020-12-27:
is the program able to client telnet ? or ssh ?
if not you should call it from windows side then proceed from it

melidron, 2020-12-28:
i have tried many times the method u have pasted and it can only disconnect router but never connect it again...the progmam does not offer a client telnet or ssh but if i knew how to make a batch file that open telnet DOS window and then connect to routter with admin and passworc then run the command u told me that would solve my problem...

negan07, 2020-12-28:
there isn't a properly "reconnect button" on web gui but reboot one so proceed manually inserting local router ip, user/pw then create a new script
a macro operation record will start: enter the credential go to adv tab then router info then reboot (upleft)
at the end save the script

melidron, 2020-12-29:
yes there is reconnect button in advanced mode... but its not that the point... Anyway I managed to do what I wanted with vbs script where send key to run telnet and your amazing command yout wrote that renew my ip adress in only 16 - 21 seconds... Thanks a lot for your support And happy new year my friend :) :)

negan07, 2020-12-29:
happy new year too

closed 2020-12-29 by negan07