negan07/ancistrus #77: How to change DHCP lease time

giga89, 2021-01-29:
Hi to all!
Is there a parameter to change this value?
Thx a lot

negan07, 2021-01-30:
the router uses a modified version of `udhcp-0.9.7`
options are generated into:
`/etc/udhcpd.conf[1-8]` where [1-8] are the vlan id if any (typically 1 is the number)

```
root@D7000:/$ cat /etc/udhcpd.conf1
server		10.0.0.1
start		10.0.0.11
end		10.0.0.50
interface	group1
group_id	1
option	subnet	255.255.255.0
option	router	10.0.0.1
option	dns	10.0.0.1
option	lease	86400
```

the lease related option is the last 
`option       lease       86400
`
configured with a fixed value of 24hours

Considering a single vlan (id 1), a possible workaround to modify it could be (and it must be tested):

```
rc_dhcpd stop
rc_dhcpd create 1
ln -sf /usr/sbin/udhcpd /var/udhcpd_1
sed -i "s|86400|13333|" /etc/udhcpd.conf1
/var/udhcpd_1 /etc/udhcpd.conf1 &
```

giga89, 2021-02-01:
Hi Negan, 
first of all: you are awesome.
I solved with your commands
Thank u, happy life

closed 2021-02-01 by giga89

negan07, 2021-02-01:
thank you too

In truth this is a dirty workaround, to be reapplied at every code restart, so take care expecially on cronjobs or wan reconnections