negan07/ancistrus #79: Request - Default setting

m0lz, 2021-02-10:
Hi Negan

I am not currently using ancistrus (had to use the Netgear firmware for a while due to being more up to date for security fixes), but will be using your next release .. So currently I do not remember what the default for the following setting in ancistrus is ..

On "Advanced, Setup, WAN Setup" - Disable SIP ALG

Disable SIP ALG on the Netgear original firmware is by default not disabled

Currently with NAT Slipstreaming exploit I believe this setting is best set to disabled, along with having up to date Browsers to protect against this new exploit.

Apologies if you already do set this as disabled by default, but if you do not, then please could you set it as disabled for the next release of Ancistrus.

negan07, 2021-02-10:
Don't understand the problem.

Security is assured at this time also with 1.0.1.74 based with latest packages. 

Openssl package version is newer than orig one.
minihttpd (security fixes are mainly pointed to this) is also updated with cookie samesite attribute compliant to actual browsing standards.
Openvpn removed lzo lib embedded, source of leaks and improved ciphers & digest algorithm robustness.
And so on (zlib, curl, ...)

m0lz, 2021-02-11:
OK well if you believe that the updates you mention mitigate NAT Slipstreaming ..
https://samy.pl/slipstream/
.. Then there is no issue to be concerned with, and this can be closed.

negan07, 2021-02-15:
Cannot find a conjunction reference between security enhancements and this stuff.

The article refers to R7000 router, quite different from D7000.
It seems to be related to voip protocols: this router missis direct voip ports.

An eventual attack should redirect itself to the voip gateway: this should be a problem of the voip terminal gateway or router machine with voip port interfaces.

Anyway, changing a default settings won't improve security because if already set, option value remain chosen even if changing the default value (simple to do editing `/usr/etc/default` with an hex editor).
Nothing will change also removing the option from wan setting: moreover it should be harder to turn the option off..

The only way should be to upgrade the related code module with something better, if any.

Try to:

`cat /proc/sipalg_enable`

and see if there's something.

It appears that enable/disable `sip_alg` simply related to:

```
echo 0 >/proc/sipalg_enable
echo 1 >/proc/sipalg_enable
```

negan07, 2021-02-20:
There seems to be no `/proc/sipalg_enable` file so this wan option looks to be orphaned because its action produces no effects.

This would fix the problem involved in any case.

closed 2021-02-20 by negan07